MarlinFirmware/Marlin #10206: UBL and PROBE_OFFSET

alexxy, 2018-03-24:
UBL should play nice with Z_PROBE_OFFSET

For example i have dondolo like extruder and using BL-Touch as Z_PROBE (in that case it has large offsets (10,-42,-0.8). 

So UBL net should be corrected to Z_PROBE_OFFSET (some points cant be reached by Z-Probe)

brookstalley, 2018-03-24:
What branch are you using? I find that it does work this way in bugfix-1.1.x. 

My BLTouch is at offset 58,0 from my primary extruder. After doing a G29 P1, the mesh looks like:

```
READ: Bed Topography Report:
READ: (0,6)                                           (6,6)
READ: (8,246)                                        (196,246)
READ:      .          .      -0.236   -0.205   -0.161   -0.217   -0.183  
READ:      .          .      -0.168   -0.131   -0.140   -0.216   -0.207  
READ:      .          .      -0.111    0.130   -0.150   -0.235   -0.280  
READ:      .          .      -0.131   -0.133   -0.047   -0.304   -0.347  
READ:      .          .      -0.096   -0.122   -0.158   -0.307   -0.305  
READ:      .          .      -0.122   -0.165   -0.195   -0.287   -0.248  
READ: [    .   ]      .      -0.141   -0.208   -0.235   -0.255   -0.228  
READ: (8,8)                                            (196,8)
READ: (0,0)                                             (6,0)
```

...where the empty spots are places the probe can't reach.

Roxy-3D, 2018-03-24:
Do a G29 P3 twice to fill in the unreachable columns...    And then do a G26 to see how well (or not) the filled in mesh points match reality.

If some of them are not perfect...   move the nozzle over them and do a G29 P4 R to edit them up or down.

> So UBL net should be corrected to Z_PROBE_OFFSET (some points cant be reached by Z-Probe)

That can be done by simply specifying a MESH_INSET.   But...   You will give up a lot of your print envelope doing that.   The current approach of saying some mesh points can not be auto probed still allows people to use their full print envelope.

alexxy, 2018-03-25:
I use bugfix-2.0.x. I can set mesh inset to e.g. 50 so effective probbing space will be 100 x 100 instead of ~180 x 150 for 200x200 bed (since i have large probe offset).

Also  does UBL code shifts probe points to PROBE_OFFSET? Since effectively if you probe at 

(X, Y) for extruder coordinats probe is at (X - PROBE_OFFSET_X, Y - PROBE_OFFSET_Y)

Roxy-3D, 2018-03-25:
Yes, the probe philosophy is different between UBL and Bi-Linear.  They both use the same lower level probing system.   But because of the mesh editing features built into UBL, the ?_PROBE_OFFSET_FROM_EXTRUDER is handled by the UBL logic.    UBL knows how big the mesh is and where it is positioned on the bed.   And if a mesh point is unreachable by the automatic probing, it is denoted with a '.' and can be filled in later by what ever means the user chooses.

teb1934, 2018-03-25:
Interestingly enough I was going to write up a similar issue to OP's.  I'm using the 1.1.8BUGFIX from a couple of weeks back with an 18MM Inductive Probe on a Wanhao Duplicator i3 with MKS Base v1.5 board.  Even with precisely measured z probe offsets (x -18, y - 48), when I run a G29 P1 it still tries to probe areas where the probe is off the bed and if I don't manually cut the power it will crash the nozzle into the bed.  For not I have 60mm Mesh Inset defined and have tuned my mesh perfectly within that area - but that equates to a ~25% usable areas of my bed.

I would expect the results that brookstalley received, where it probes the bed where there is a valid probe point and skips over the results where there was not a valid point for the probe to trigger.

Roxy-3D, 2018-03-25:
> I would expect the results that brookstalley received, where it probes the bed where there is a valid probe point and skips over the results where there was not a valid point for the probe to trigger.

Yes.  That is what is supposed to happen.   If I had a system where the problem happened, it would be easy to find the root cause and correct it.    I would really like to encourage you to see if you can find the issue.  It will help a lot of people.   Probably it makes sense to turn on the DEBUG_LEVELING because that tells you (very verbosely) what is happening at each point in the process.

thinkyhead, 2018-03-25:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

boelle, 2019-02-20:
@alexxy still having an issue with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-11 by shitcreek