MarlinFirmware/Marlin #10208: M18 E, M17 while Auto Home & Bed Level?

Decstasy, 2018-03-24:
Hello,

I have posted the following text in a huge FB group without a single answer :( I need some help.

I've installed a new probing sensor (LJC18A3-H-Z/BX with optocoupler) incl. mount on my Anet A6 and it's very close to my extruder stepper. Is there a function to disable the extruder stepper like the heatbed and fans while homing and leveling?

As soon as the Z axis goes down to the bed and the sensor gets triggered, it keeps the signal when the Z axis moves up again; after disabling all steppers, the probing sensor loses the high status. Sometimes the sensor sends a signal as soon the extruder gets enabled.

I've read some source and I think, that this might be simple to implement just some configuration option and function which is similar to thermalManager.pause called with QUIET_PROBING in Marlin_main.cpp.
Like:
```
stepper.synchronize(); // Wait for buffered moves to complete
// Only enable on boards that have separate ENABLE_PINS
#if E0_ENABLE_PIN != X_ENABLE_PIN && E1_ENABLE_PIN != Y_ENABLE_PIN
disable_e_steppers();
safe_delay(100);
#endif
```
And reenable extruder stepper if it has been disabled:
```
enable_E0();
enable_E1();
enable_E2();
enable_E3();
enable_E4();
```

I thought also about "M18 E" and "M17" after homing and leveling but it gets activates as soon it starts leveling.

What do you think? Can someone help me?

Thank you and best regards,
Dennis

thinkyhead, 2018-03-29:
The E axis _should_ remain off until the first E move. This is handled at the planner level. And then it should time out when there are no additional E moves for the period set by `M18 E`, if the corresponding options are set in the configuration.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@Decstasy did you as thinky suggested?

Decstasy, 2019-02-21:
I patched it myself and everything was fine. After about a month I tried the newest bugfix 1.1.x branch without my patch and the problem was gone. So with the newest bugfix branch you should be fine.

boelle, 2019-02-21:
@Decstasy use bugfix 2.0, that is where things happen, and its both 8 bit and 32 bit

but please hit the green close button below

closed 2019-02-21 by Decstasy