MarlinFirmware/Marlin #10209: Compilation error if TMMC2130 and EEPROM are active at the same time

fsadak, 2018-03-24:
As mentioned in the title, when the TMC2130 and EEPROM are activated at the same time, I received a lot of compilation errors similar to the following.

**sketch\configuration_store.cpp: In static member function 'static bool MarlinSettings::save()':
configuration_store.cpp:835: error: expected '}' before numeric constant
           0
           ^
configuration_store.cpp:835: error: expected ',' or ';' before numeric constant****

When I checked the configuration_store.cpp file, I noticed that there was no comma after the second and third zeros in the following lines, and that the last line should be 0, 0, 0. When I corrected it in this way, it was compiled without problems.

int16_t tmc_sgt[XYZ] = {
      #if ENABLED(SENSORLESS_HOMING)
        #if defined(X_HOMING_SENSITIVITY) && (ENABLED(X_IS_TMC2130) || ENABLED(IS_TRAMS))
          stepperX.sgt(),
        #else
          0,
        #endif
        #if defined(Y_HOMING_SENSITIVITY) && (ENABLED(Y_IS_TMC2130) || ENABLED(IS_TRAMS))
          stepperY.sgt(),
        #else
          0 **<== No comma**
        #endif
        #if defined(Z_HOMING_SENSITIVITY) && (ENABLED(Z_IS_TMC2130) || ENABLED(IS_TRAMS))
          stepperZ.sgt()
        #else
          0 **<== No comma**
        #endif
      #else
        0 **<== Should be 0, 0, 0**
      #endif
    };

Bob-the-Kuhn, 2018-03-25:
It looks like you are correct.  Both bugfix-1.1.x and bugfix-2.0.x have this.

Please ZIP & attach your Configuration.h, Configuration_adv.h & pins_YOUR-BOARD.h files.  This'll make it a lot easier to verify the fix.

fsadak, 2018-03-25:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1845160/Configuration.zip)
Here are the files. Please note that i made some pin assignment changes for software SPI.

thinkyhead, 2018-03-28:
Thanks for the report! This is now patched.

commit 47cd0120 (2018-03-28): Fix missing comma in tmc_sgt

Fixes #10209
commit e848bebb (2018-03-28): Fix missing comma in tmc_sgt

Fixes #10209
commit 2bc7f416 (2018-03-28): Fix missing comma in tmc_sgt

Fixes #10209
closed 2018-03-28 by thinkyhead