MarlinFirmware/Marlin #10212: Can't compile with Zero Extruders!

Maglin, 2018-03-25:
### Description

Wanting to use Marlin on a CNC machine that has two Y and two X steppers and one Z stepper.  I want to use discrete drivers for each stepper motor.  When I set `#define EXTRUDERS 0` and only that the FW will not compile.  I can make the desired setup in Repitier but I don't like Repitier as a FW.  Thank you for any help you can provide.

### Steps to Reproduce

1. Change `#define EXTRUDERS 1` to `#define EXTRUDERS 0` in configuration.h
2. Click Verify

**Expected behavior:** 
For the FW to compile.

Since CNC has been added I expected to be able to use all 5 available steppers for axis drives.  I initially configured with duel X and duel Y enabled and get an error that there isn't enough available extruder drivers let.  Then when I set the number of extruders to 0 I get an array error.

**Actual behavior:**  
Array error during compiling.

#### Additional Information
Here is the Error Message from the IDE.

>  Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jerem\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jerem\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jerem\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0242 -ide-version=10805 -build-path C:\Users\jerem\AppData\Local\Temp\arduino_build_962237 -warnings=none -build-cache C:\Users\jerem\AppData\Local\Temp\arduino_cache_221681 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose T:\Downloads\3D Printer\MPCNC\firmware\Marlin\Marlin\Marlin.ino
> C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jerem\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jerem\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jerem\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0242 -ide-version=10805 -build-path C:\Users\jerem\AppData\Local\Temp\arduino_build_962237 -warnings=none -build-cache C:\Users\jerem\AppData\Local\Temp\arduino_cache_221681 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose T:\Downloads\3D Printer\MPCNC\firmware\Marlin\Marlin\Marlin.ino
> Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
> Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
> Detecting libraries used...
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\Marlin.ino.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\G26_Mesh_Validation_Tool.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\I2CPositionEncoder.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\M100_Free_Mem_Chk.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\MarlinSerial.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\Marlin_main.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\Max7219_Debug_LEDs.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\Sd2Card.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\SdBaseFile.cpp" -o "nul"
> "C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Ar [... cut ...]
> C:\Users\jerem\AppData\Local\Temp\arduino_build_962237\sketch\planner.cpp:97:38: note: in expansion of macro 'ARRAY_BY_EXTRUDERS1'
> 
>  float Planner::e_factor[EXTRUDERS] = ARRAY_BY_EXTRUDERS1(1.0); // The flow percentage and volumetric multiplier combine to scale E movement
> 
>                                       ^
> 
> planner.cpp:162: error: too many initializers for 'uint8_t [0] {aka unsigned char [0]}'
> 
>    uint8_t Planner::g_uc_extruder_last_move[EXTRUDERS] = { 0 };
> 
>                                                              ^
> 
> exit status 1
> 'ARRAY_0' was not declared in this 
> 
> > scope
>

Bob-the-Kuhn, 2018-03-25:
Here's a quick & dirty if you want to do something quickly.

Personally I just leave it at 1, set TEMP_SENSOR_0 to 998 and ignore all the other extruder specific options.

If you want to use move than 3 stepper **drivers** for X, Y & Z then you'll have to do some playing.   The easiest option is to just not stuff anything in the E0 socket.  If you must use E0 because your controller only has the exact number of sockets needed then there are tricks that can be played with the pins_YOUR_BOARD.h file.

Bob-the-Kuhn, 2018-03-25:
Here's a quick & not quite as dirty.  It takes care of the number of extruders, the temp sensor and more than 3 steppers.

Just add **#define NO_EXTRUDERS** to your **Configuration.h** file and replace your **Conditionals_LCD.h** file with one of these:
[Conditionals_LCD 1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/1845022/Conditionals_LCD.1.1.x.zip)
[Conditionals_LCD 2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/1845020/Conditionals_LCD.2.0.x.zip)

There's a lot more that needs to be done to make the LCD display consistent with no extruders.

Maglin, 2018-03-25:
Oh that worked to get it to compile with both duel X and Duel Y enabled..  I'll test it out tomorrow.  Thank you for your help.

Maglin, 2018-03-25:
Another problem.  The E0 driver only goes one direction no matter what.  The E1 driver will move in both directions though.  This is control via the LCD...  I didn't try moving it via serial.  I can post my config if needed but it's pretty basic other than the duel x and y axis's...

Bob-the-Kuhn, 2018-03-25:
Try this image.  It has the direction issue fixed .  It also has the fan and temperature stuff stripped from the LCD status display.

https://www.dropbox.com/s/yio71g4b9xollq1/Marlin-Bob-2.zip?dl=0

Maglin, 2018-03-25:
Flashing now...  I'll update shortly.

Maglin, 2018-03-25:
Update.  I'm getting a different error with the new 3 files added.  I got a compile error with the modified Conditionals_LCD.h and the one from GitHub.  Here is an error log from the one from GitHub.

[MarlinCNC Compile Error.txt](https://github.com/MarlinFirmware/Marlin/files/1845739/MarlinCNC.Compile.Error.txt)

Bob-the-Kuhn, 2018-03-25:
Please ZIP and attach Configuration.h, Configuration_adv.h & pins_YOUR-BOARD.h

I should be able to reproduce the errors using those.

I expect there are configuration options selected that need not be.

Maglin, 2018-03-25:
I'm not using any special board so standard pins_RAMPS.h is being used.  Here are all my requested files.

[MarlinCNC.zip](https://github.com/MarlinFirmware/Marlin/files/1845763/MarlinCNC.zip)

Bob-the-Kuhn, 2018-03-26:
It compiles for me.  Here's the result.  
https://www.dropbox.com/s/qf25ktj63v90f8x/Marlin-Bob-2%20compiles.zip?dl=0

Try deleting the **.piolibdeps** directory.  That needs to be done when switching between 1.1.x and 2.0.x and sometimes just because the sky is blue.

Maglin, 2018-03-26:
I didn't realize you have just moved files.  I know what I didn't compile because I was using it wrong.  I'm going to flash this after I fully config it and test it out tomorrow.  I've run out of time today.  I'll report all my finding and give feedback on potential needed changes.  Thank you again.

Maglin, 2018-03-26:
So I was getting the file name too long compile errors.  did some searching and uninstalled 1.8.x and grabbed 1.9.0 beta IDE.  Using the build you said you could compile I can't compile.  I've moved it to the root of my C drive almost.  Moved the sketchbook location as well.  Updated all libraries and installed U8Glib from the repo instead of through the Libary manager.  I'm at wit's  end here.  Here is my error message currently.

[Compile for board error.txt](https://github.com/MarlinFirmware/Marlin/files/1846469/Compile.for.board.error.txt)

Bob-the-Kuhn, 2018-03-26:
I've been compiling with PlatformIO.  

1.9 beta has a known bug.  It hates comments after pin definitions.

Here's a couple of files with the comments stripped out.  It compiled for me on 1.9 with the config you posted earlier.

[1.9 files.zip](https://github.com/MarlinFirmware/Marlin/files/1848495/1.9.files.zip)

Maglin, 2018-03-27:
I've installed PlatformIO and I must say I've wasted a lot of time use the IDE to compile.  It compiles now no problem.  I'm not going to get to test it out until tomorrow.  Just wanted to update.

Maglin, 2018-03-30:
Here is my testing.  X/Y working partially.  I say partially because I have to initialize the EEPROM before I can move any axis in a neg move.  You can go 10mm positive but can't go back to 0.  This is via the LCD menu.  Once you initialize the EEPROM you can move in a negative direction.  That is about it.  I have G2/G3 modes working nicely.

thinkyhead, 2018-03-30:
> Once you initialize the EEPROM

Note that `M502` by itself doesn't initialize the EEPROM. It simply applies the default settings in SRAM. If `M500` is then done, it writes those defaults to the EEPROM.

Maglin, 2018-03-30:
@thinkyhead if I do nothing other than choosing to initialize EEPROM from the LCD menu I can move other than positive.  I can do a M502, M500, M501 to accomplish the same thing..  which I created a Macro for so it's not so buried in the lcd menu.

If I don't do this after a reset I can go positive 10 but not return.  This behavior is in any axis.  That is what I mean by a negative move.  Not a move smaller than the origin.

thinkyhead, 2018-03-30:
>  I can go positive 10 but not return. This behavior is in any axis.

Bizarre. What does `M114` report at the point before you try to move? What does `M119` report?

> M502, M500, M501

The `M501` in that sequence is redundant and can be left out (if you are using the latest code).

thinkyhead, 2018-04-05:
Any additional information you can provide will help us to solve this.

Maglin, 2018-04-10:
Sorry I was TDY for a week.  I'm going to reattack this in a few days when time allows me to work on the machine again.

thinkyhead, 2018-09-28:
> Is there an easy fix for this?

Please file a new issue.

boelle, 2019-02-20:
@bowzee close this one

boelle, 2019-02-20:
@Maglin is this one still relavant?

fmuntean, 2019-06-17:
I like Marlin for my 3D printer. Would be nice if configuration would allow for easy CNC conversion so I can use the same firmware for both 3D printing and CNC.
Would be nice to be able to disable extruders,  temperature sensors and any PID controls.
Currently we have to keep the Extruder0 and one temperature sensor as type 999.

kpter, 2020-03-09:
Can I also votes for this?
I successfully using the Marlin for my CNC. And I also use **double X** + **double Y** + **Z** (it is **all 5** drivers  of RAMPS).
It is a lot of places to configure (I appoint fake pins in pins_RAMPS.h)
It may be a one directive in the configuration, for example:
`    #define CNC`
and in this section disable all extruders, temperature sensors, overheating checks, LCD information about extruders and fans, PIDs).

thinkyhead, 2020-03-11:
@kpter — Zero extruders has been possible for a long while now. If compilation ever breaks with `EXTRUDERS` set to 0 please report that as a bug and we'll aim to get it patched quickly.

closed 2020-03-11 by thinkyhead