MarlinFirmware/Marlin #10218: UBL minimum distance between probing points...?!

rooiejoris, 2018-03-25:
### Description

Is there maybe a minimal distance between the probing points when using UBL...?!
I am doing a small experiment and abuse the UBL to compensate for a very unflat and small surface.
The first test went pretty well, but then the surface was more or less flat, now the surface is not flat and at some points the z moves very very far down or up. [Like in way to far, have to block the z axis by hand and switch of the machine and after a while when I turn it on it is still rotating. When switching of the machine the arduino stays powered over USB.]

[I have some graphic design, vector drawing experience and have the feeling that the 'handles' of the mesh point are to strong, but on the other hand, I have no idea how the mesh is calculated]

I see problems with this bumpy small surface
Bed Topography Report:
(0,7)                   (3,7)
(1,199)                (33,199)
 -0.097    0.006   -0.573   -3.986
  0.074    1.698    0.844   -3.523
 -0.117    1.720    0.813   -3.024
 -0.196    1.649    0.668   -3.117
 -0.327    1.463    0.430   -2.187
 -0.327    1.375    0.252   -3.239
 -0.157    1.563    0.468   -4.871
[-0.206]  -0.444   -0.742   -3.639
(1,1)                    (33,1)
(0,0)                     (3,0)

but not with these mesh points [sorry didn't know of the G92 T command yet...]
G29 I 999
M421 I 0 J 0 Z -0.702500 ; X 1.00, Y 5.00
M421 I 0 J 1 Z -0.436250 ; X 1.00, Y 54.50
M421 I 0 J 2 Z -0.181250 ; X 1.00, Y 104.00
M421 I 0 J 3 Z 0.088750 ; X 1.00, Y 153.50
M421 I 0 J 4 Z 0.152500 ; X 1.00, Y 203.00
M421 I 1 J 0 Z -0.372500 ; X 17.50, Y 5.00
M421 I 1 J 1 Z -0.113750 ; X 17.50, Y 54.50
M421 I 1 J 2 Z -0.003125 ; X 17.50, Y 104.00
M421 I 1 J 3 Z 0.272500 ; X 17.50, Y 153.50
M421 I 1 J 4 Z 0.400000 ; X 17.50, Y 203.00
M421 I 2 J 0 Z -0.010625 ; X 34.00, Y 5.00
M421 I 2 J 1 Z 0.055000 ; X 34.00, Y 54.50
M421 I 2 J 2 Z 0.073750 ; X 34.00, Y 104.00
M421 I 2 J 3 Z 0.377500 ; X 34.00, Y 153.50
M421 I 2 J 4 Z 0.525625 ; X 34.00, Y 203.00
M421 I 3 J 0 Z 0.223750 ; X 50.50, Y 5.00
M421 I 3 J 1 Z 0.216250 ; X 50.50, Y 54.50
M421 I 3 J 2 Z 0.139375 ; X 50.50, Y 104.00
M421 I 3 J 3 Z 0.465625 ; X 50.50, Y 153.50
M421 I 3 J 4 Z 0.739375 ; X 50.50, Y 203.00
M421 I 4 J 0 Z 0.379375 ; X 67.00, Y 5.00
M421 I 4 J 1 Z 0.311875 ; X 67.00, Y 54.50
M421 I 4 J 2 Z 0.205000 ; X 67.00, Y 104.00
M421 I 4 J 3 Z 0.437500 ; X 67.00, Y 153.50
M421 I 4 J 4 Z 0.865000 ; X 67.00, Y 203.00

Marlin v1.1.8, downloaded last week

### Steps to Reproduce

G28
G29 P1 ; does automatic probing all points
G29 A ;(to activate the UBL System)
G29 S1 ;(to save the Mesh in slot 1)
M500
G29 S0
M500
G29 S-1 ;show mesh coordinates
G29 V4 T

**Expected behavior:** [What you expect to happen]

It should follow the uneven surface

**Actual behavior:** [What actually happens]

at some points the z moves are very large
and just Z, without any X or Y moves

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1845658/configurations.zip)

Roxy-3D, 2018-03-25:
That is a seriously warped bed!    One way to get confidence the 'right thing' is happening is put the nozzle 1mm or 2mm above the bed.   And then move it (slowly!!!) around the bed.   If UBL is activated, you should see the nozzle track the contour of the bed.   (Or at least the mesh points you have probed.)

Most people need more than a 4x4 mesh.   And given how warped your bed is, it would be wise to go much higher like 10x10.

rooiejoris, 2018-03-25:
Well its more a bended surface where I want to print on...: )
I made some extra additions with extra info, but you also noted that the shape is bed size [in X] is very small...? I thought maybe thats could be the problem...?!

Roxy-3D, 2018-03-25:
Let's 
- bump up the mesh size...   10x12 probably makes sense
- do G29 P1 to build a mesh.   
- do a G29 T and we probably will see a column (or two) of dots...  And a row (or two) of dots.
- fill those unprobed locations with G29 P3 T and a follow up G29 P3 T
- do a G29 A to activate UBL
- Do a G29 S1 to save the mesh
- Do a M500 to get the EEPROM storage updated with the machine's current state
- lower the nozzle to 2 mm above the bed.
- carefully move the nozzle around on the bed.   Does it track accurately?   Be especially careful in the bed areas that were unreachable by the Z-Probe.   We may need to edit those areas to reflect reality.

Please do that and report back on any issues.

rooiejoris, 2018-03-25:
Now doing a full Y move at X22 and only get a 'strange' large but not extreme z-move around Y47
SENDING:G29 T
Bed Topography Report:
(0,14)                                                                   (9,14)
(1,199)                                                                (33,199)
 -1.432   -1.316   -1.227   -0.807   -0.759   -0.644   -1.156   -1.689   -2.551   -3.772

 -1.717   -0.500    0.473    0.453    1.208    0.895    0.175   -0.817   -2.613   -4.109

 -0.620    0.528    0.730    0.961    0.989    0.674   -0.037   -1.162   -2.926   -3.301

 -0.937   -0.108    0.366    0.597    0.839    0.576   -0.161   -1.317   -3.239   -6.425

 -1.079   -0.206    0.372    0.674    0.687    0.378   -0.316   -1.542   -2.975   -4.957

 -1.147   -0.273    0.288   -1.231    1.161    0.858    0.149   -1.062   -3.179   -9.230

 -0.802    0.053    0.644    0.959    1.043    0.713    0.004   -1.276   -3.344   -4.100

 -0.892   -0.022    0.556    0.844    0.871    0.558   -0.170   -1.346   -2.934   -2.804

 -0.941   -0.061    0.479    0.753    0.674    0.349   -0.350   -1.520   -3.204   -4.194

 -0.969   -0.106    0.441    0.681    0.625    0.231   -0.507   -1.646   -3.059   -4.636

 -1.016   -0.104    0.449    0.689    0.627    0.306   -0.483   -1.689   -3.507   -5.600

 -0.811    0.025    0.683    0.953    0.936    0.486  [-0.303]  -1.492   -3.294   -5.114

 -0.761    0.089    0.689    0.953    0.739    0.801    0.083   -1.111   -2.934   -7.259

 -0.573    0.389    0.979    1.206    1.339    0.996    0.271   -0.916   -2.671   -5.863

 -1.181   -1.284   -1.259   -0.706   -0.260   -0.440   -0.849   -1.422   -2.321   -3.628
(1,1)                                                                    (33,1)
(0,0)                                                                     (9,0)

again, I know I am really abusing the UBL, but would be so nice if it works... : )

Roxy-3D, 2018-03-25:
A G29 W would be helpful...   That will give us details about the mesh point locations.
I'm guessing...  But it looks like that area of the mesh (X,Y)=(22,47) is fairly tame.   Nothing much is going on in that area of the bed.

rooiejoris, 2018-03-25:
Here we go:
>>>G29 W
SENDING:G29 W
Unified Bed Leveling System v1.01 active.
Mesh 1 Loaded.
UBL object count: 1
planner.z_fade_height : 50.0000
# of samples: 150
Mean Mesh Height: -0.880550
Standard Deviation: 1.857006
zprobe_zoffset: -5.0000000
MESH_MIN_X  ((((((35) / 2) - (35) / 2) + 1)>(0 + -5)?((((35) / 2) - (35) / 2) + 1):(0 + -5)))=1
MESH_MIN_Y  ((((((200) / 2) - (200) / 2) + 1)>(0 + 0)?((((200) / 2) - (200) / 2) + 1):(0 + 0)))=1
MESH_MAX_X  ((((((35) / 2) + (35) / 2) - (1))<(40 + -5)?((((35) / 2) + (35) / 2) - (1)):(40 + -5)))=33
MESH_MAX_Y  ((((((200) / 2) + (200) / 2) - (1))<(200 + 0)?((((200) / 2) + (200) / 2) - (1)):(200 + 0)))=199
GRID_MAX_POINTS_X  10
GRID_MAX_POINTS_Y  15
MESH_X_DIST  3.56
MESH_Y_DIST  14.14
X-Axis Mesh Points at: 1.000  4.556  8.111  11.667  15.222  18.778  22.333  25.889  29.444  33.000
Y-Axis Mesh Points at: 1.000  15.143  29.286  43.429  57.571  71.714  85.857  100.000  114.143  128.286  142.429  156.571  170.714  184.857  199.000
Unified Bed Leveling sanity checks passed.

[I totally agree on the X,Y positions, nothing weird, thats also why I think its a bit strange...]
Is there a way how the 'generated' mesh look like...!? like in with the interpolated point in there how it calculates its value during the movements...?
also: I changed the LEVELED_SEGMENT_LENGTH 0.5 [default 5, but with this surface...]

Roxy-3D, 2018-03-25:
> MESH_X_DIST 3.56

Everything should work...  But 3.5mm between mesh points is a bit excessive.   LEVELED_SEGMENT_LENGTH is not currently used in UBL.   UBL has its own motion routine.  

I guess its time for a video...  Can you show us the expected Z movement at  (X,Y)=(22,47) ?

rooiejoris, 2018-03-25:
hmmm... I wanted to wait a few more days to make the setup going public... : )
but here we go, you will get it in a few minutes...

[maybe I will remove the video and republish it later...]

Roxy-3D, 2018-03-25:
We can use email to keep things private....
If you go to www.3dprintboard.com you can join and send me a private message there.  (Send to Roxy)

Roxy-3D, 2018-03-25:
I've sent you an email!

Roxy-3D, 2018-03-25:
Let's try this configuration.h file....   (You need to rename the type to .h)

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1845850/Configuration.txt)

Roxy-3D, 2018-03-25:
Try 2:

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1845861/Configuration.txt)

Roxy-3D, 2018-03-25:
compiling now with:
```
#define X_BED_SIZE (X_MAX_POS-X_MIN_POS)   // Altered from original configuration.h file
#define Y_BED_SIZE (Y_MAX_POS-Y_MIN_POS)

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
//#define X_MAX_POS X_BED_SIZE
//ovc first test
//#define X_MAX_POS 118
//egg
#define X_MAX_POS 40
#define Y_MAX_POS 205 // Y_BED_SIZE
//joris set this to unlimited for continues printing
#define Z_MAX_POS 100
```

Roxy-3D, 2018-03-25:
Try 3:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1845875/Configuration.txt)

Roxy-3D, 2018-03-25:
Fresh copy of files...
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1845880/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1845881/Configuration.txt)

rooiejoris, 2018-03-26:
Fresh download of Bugfix-1.1.x solved the compiling issue and also the strange z moves.

THANKS A LOT...!! @Roxy-3D for the really fast 1 to 1 help on this issue...!

I think this may be closed, needs some more tuning on my side

cheers / joris

Roxy-3D, 2018-03-28:
Hey people!!!!   Check this out:   Marlin is 3D-Printing on Easter Eggs!   

www.rooiejoris.nl/3d-printing-on-eggs/

rooiejoris, 2018-03-28:
>> Maybe next year we can use colored Sharpies to draw very intricate patterns and designs on the Easter Eggs.

I might try it this year if I find a few spare minutes... : )

Roxy-3D, 2018-03-28:
>> Maybe next year we can use colored Sharpies to draw very intricate patterns and designs on the Easter Eggs.

> I might try it this year if I find a few spare minutes... : )

Here is a Marlin logo...   If you do manage to get setup...   Try drawing this on an egg:
(you need to rename the .txt extension to .stl)
[marlin logo-ascii.txt](https://github.com/MarlinFirmware/Marlin/files/1858153/marlin.logo-ascii.txt)

rooiejoris, 2018-04-08:
Related to https://github.com/MarlinFirmware/Marlin/issues/10267 and https://github.com/MarlinFirmware/Marlin/issues/10265 i think...
[don't know if this should be a new issue...?!]

Is it possible to  have a smaller print bed, but a larger print area...?!
Let me explain, the bed size is for example 100x100, but the movement area is 200x200, then is theory you can print a pyramid or cone 'upside down'...?!

That's the theory, in real life I would like to implement these settings for the egg printing.

Roxy-3D, 2018-04-08:
One answer would be to define the bed size at the full amount (200x200).   But then put a 50mm inset on all sides.   That would restrict the mesh to the middle 100x100 of the 200x200 movement.

And of course...  nothing is ever easy.   So that 100x100 mesh isn't going to align perfectly.   In configuration_adv.h you can adjust the mesh inset for all 4 sides.

rooiejoris, 2018-04-08:
I think these settings can help me out indeed

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X MESH_INSET
  //#define MESH_MIN_Y MESH_INSET
  //#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  //#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
#endif

I am a bit confused with all the different places where mesh insets or alike are set...
confused in a way that I don't know which are used for what type of leveling [or combined]
also with the sanity check i had the feeling there are mixed up sometimes, but cannot reproduce at the moment

there is:
`  // Set the boundaries for probing (where the probe can reach).
`
but also:
  `//===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  //#define MESH_EDIT_GFX_OVERLAY   // Display a graphics overlay while editing the mesh

  #define MESH_INSET 1              // Mesh inset margin on print area
`
and

`  //===========================================================================
  //=================================== Mesh ==================================
  //===========================================================================

  #define MESH_INSET 1          // Mesh inset margin on print area
`

and

`  #define PROBE_PT_1_X 3       // Probing points for 3-Point leveling of the mesh
  #define PROBE_PT_1_Y 190
  #define PROBE_PT_2_X 3
  #define PROBE_PT_2_Y 10
  #define PROBE_PT_3_X 33
  #define PROBE_PT_3_Y 10
`

will try later on, but since easter is over I am not in a hurry anymore... : )

Roxy-3D, 2018-04-08:
```
#define MESH_INSET 1 // Mesh inset margin on print area
```
If you change this one...   (In configuration.h) it is used for all four sides of the mesh.
If you need more granular control and want to set the inset of each side of the mesh, you do that in configuration_adv.h
```
#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X MESH_INSET
  //#define MESH_MIN_Y MESH_INSET
  //#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  //#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
#endif
```

boelle, 2019-02-20:
@rooiejoris is this still relavant? is it a bug or hardware problem?

closed 2019-02-20 by thinkyhead

rooiejoris, 2019-02-20:
I managed to print, so I think it is solved now...
It's too long ago to remember exactly the problem... : )
I would say it can be closed.

cheers / joris