MarlinFirmware/Marlin #10219: No direct serial control (Serial 1)

pyr0ball, 2018-03-25:
### Description

I'm attempting to control a marlin based printer over direct serial using the UART pins on a raspberry pi 3, however the connection simply times out without any response. The same behavior occurs using a Windows PC and USB FTDI dongle with Simplify3D

This was tried with Marlin 1.1.6 as well as a fresh clone of 1.1.8

### Steps to Reproduce

1. Flash Marlin firmware with below settings under Configuration.h:

```
/**
 * Select which serial port on the board will be used for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port 0 is always used by the Arduino bootloader regardless of this setting.
 *
 * :[0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 1

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200
```

2. Connect to AUX-1 port on board.
pinout: `5v - GND - Tx - Rx`

3. Attempt to communicate with printer either using octoprint, or Simplify3D

**Expected behavior:** [What you expect to happen]
Communication established, printer responds to temperature requests, etc.

**Actual behavior:** [What actually happens]
Serial port times out:
```
Attempting connection at \\.\COM7...
  Testing plaintext communication protocol...
  Testing binary communication protocol...
  Testing alternate communication protocols...
  Attempting RTS reset and trying again...
  Connection failed.
```

#### Additional Information

I tried using not only the UART pins on a Rpi 3, but also a windows system running Simplify3D using a  USB FTDI dongle

I tried multiple control boards:

- MKS Base 1.4
- MKS Gen L 1.0
- RAMPS 1.4
- RAMPS 1.6

On the Raspberry Pi 3 I did get some feedback about disabling the UART for the bluetooth to allow pins 14/15 to be assigned to ttyAMA0 instead of ttyS0 which apparently can be finicky.

I also tried running at multiple baud rates:

- 115200
- 203400
- 250000

I did also try using `#define SERIAL_PORT 2` as well just for kicks, as well as reversing the tx/rx pins.

Just for clarification, the wires hooked up to the Rpi3 are using a voltage divider to bring the 5v down to 3.3v

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1845665/Configuration.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

thinkyhead, 2018-03-25:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

pyr0ball, 2018-03-25:
Good news!

V2.0.x works when I use:

```
#define SERIAL_PORT 0

#define SERIAL_PORT_2 1
```

bugfix_1.1.x still does not work

I'm about to try running:

```
#define SERIAL_PORT 1

#define SERIAL_PORT_2 -1
```

pyr0ball, 2018-03-25:
OK tested with 2.0.x and below config, and serial communcation was not established. So it looks like setting the primary serial port to `1` produces bad behavior on both firmwares

```
#define SERIAL_PORT 1

#define SERIAL_PORT_2 -1
```

gecko749, 2018-03-25:
Maybe it is because AUX1 is connected to SERIAL_PORT 0 ?

pyr0ball, 2018-03-25:
I'll try re-flashing SERIAL_PORT 0 and check again, but I think I did already try that.

I'll let you know shortly

thinkyhead, 2018-03-29:
@pyr0ball — At the moment I'm not sure where we would add more documentation on the website. We haven't really got a section devoted to Marlin 32-bit and the expanded options just yet.

But, if we were to add new comments to the configuration above the `SERIAL_PORT` and `SERIAL_PORT_2` settings, what would you recommend we write there?

pyr0ball, 2018-04-11:
Apologies, I thought I'd already replied to this but apparently I forgot to hit the send button

How about this as an example:

```
/**
 * Select which serial port on the board will be used for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port 0 is always used by the Arduino bootloader regardless of this setting.
 * Most 8-bit microcontrollers that also have an AUX-1 header share this serial port with the USB. If using AUX-1's serial to control your device, leave this as port 0
 *
 * :[0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 0
```

merlin2105, 2018-05-06:
Hello 

I have the same problems, Marlin 1.18... ramps 1.4

I want to connect a MKS TFT to aux 1, this also works, but then the USB port is dead and I can't use a PC or Octoprint anymore..

I tried this:

#define SERIAL_PORT 0
#define SERIAL_PORT_2 1

Works for me but not, then the MKS TFT still works but Octoprint does not connect via USB anymore, or I have to adjust something else in marlin...

Can i install Marlin 2. x on a mega 2560/ramps 1.4..?

thanks for your help...

thinkyhead, 2018-05-08:
> I tried this:
#define SERIAL_PORT 0
#define SERIAL_PORT_2 1

Marlin 1.1.x doesn't support a second serial port.

> Can i install Marlin 2.x on a mega 2560/ramps 1.4..?

Yes.

rockman256, 2018-05-10:
For me same problem with second serial port. 
It don't work on ramps fdv1. 
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
Work usb, don't work serial 1 (tested 0,1,3 ports).
When 
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1
Serial working, but by usb can't connect.

thinkyhead, 2018-05-14:
Examining the code, I see that no support for `SERIAL_PORT_2` has been implemented for DUE.

wbarber69, 2018-06-30:
@merlin2105 ive found the same to be true when using the mks tft as well as octoprint over usb.  however, i have a sort of workaround that 60% of the time works 100% of the time..... lol I have found that when i cannot connect to my board i can either power down the pi and it will find the port and connect during the service startup.  or if that doesnt work, and im not sure why this works  but, you can power down the printer, reconnect the usb to power up the board from usb, then refresh your octoprint session and the com port should show up in the drop down menu.

pyr0ball, 2018-06-30:
I figured that one out actually. There's a race condition conflict with the Bluetooth serial connection, as the rpi3 uses ttyAMA0 for Bluetooth when it's enabled and relegates the gpio serial to ttyS0. If you disable the Bluetooth in the boot config then ttyAMA0 gets reassigned to the gpio pins and is a ton more reliable

wbarber69, 2018-07-01:
Most definitely.  My level shifters finally arrived and I just hit connect for the first time.  It seems to be a faster connection even at the same baud rate.

philoupat, 2018-12-15:
there is a usb problem with map mks1.4 and Ecran tft mks 3.5.
a conflict exists usb=Tx0 rx0 and aux1=rx0 Tx0.
the port rx2 and tx2 is not defini for the WIFI output.
there is a solution to connect the OCTOPRINT ?

davidcgu, 2019-01-14:
Do you know if finally second serial is working with 2.0 and the setup shown for an mks gen L?

#define SERIAL_PORT 0
#define SERIAL_PORT_2 1

I would like to use a wifi module to get it working on the main board while keeping the mks tft screen connected at once

sandog78, 2019-02-18:
Hello all
                 So just so I'm following. When using a TFT touch screen on a MKS Gen l board with a Raspberry Pi running Octo Print I wont beable to connect Simpfly 3d to the MKS Board correct.

wbarber69, 2019-02-18:
Why wouldn’t you?

boelle, 2019-02-20:
@pyr0ball still having problems?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-11 by shitcreek

LF-fr, 2019-10-18:
You can try 👍 /**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 2