MarlinFirmware/Marlin #10222: how to perform a delay when change stepper direction

mnawaf, 2018-03-26:
I'm using a stepper driver that has a very long time response to direction signal .
I want to make a 200 micro second after any change in direction so the driver can understand this change.
how can I do this ? 
thanks in advanced .

thinkyhead, 2018-03-28:
That is such a long delay that it would essentially break the firmware to include it in the stepper ISR. I think we have to conclude that your stepper driver is not supportable by Marlin.

boelle, 2019-02-20:
@mnawaf 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

thinkyhead, 2019-02-20:
The current Marlin allows you to specify your stepper drivers and set custom delays for stepping, if needed.

closed 2019-02-20 by thinkyhead