MarlinFirmware/Marlin #10223: [bugfix-2.0.x]: STEVAL-3DP001V1 status

TerryBarnaby, 2018-03-26:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

TerryBarnaby, 2018-03-26:
I note that the STEVAL-3DP001V1 is stated as a HAL in development. Is there any status on this ?
I have checked out the bugfix-2.0.x branch and I don't see this board listed in boards.h.

thinkyhead, 2018-03-28:
It is not under very active development and is only in the early alpha stages.

closed 2018-03-28 by thinkyhead

TerryBarnaby, 2018-03-29:
So is there any code for this board somewhere to start with ?
I have some issues with the Marlin 1.0.2 based code that ST supplies for the board and was thinking of trying to port/use Marlin 2.0.x with this board. Or is it best to start with the HAL_STM32F7 port ?

thinkyhead, 2018-03-30:
From our perspective it would be most valuable to get more complete support — pins file, HAL changes, `platformio.ini` target, etc. — so at least we can get a successful compilation and the minimum of functionality. Some libraries are unusable with STM32F1, as you can see in `platformio.ini` with the `lib_ignore` section…
```ini
[env:STM32F1]
platform     = ststm32
framework    = arduino
board        = genericSTM32F103RE
build_flags  = !python Marlin/src/HAL/HAL_STM32F1/stm32f1_flag_script.py
  ${common.build_flags}
lib_deps     = ${common.lib_deps}
lib_ignore   = U8glib-HAL
  c1921b4
  libf3c
  lib066
  Adafruit NeoPixel_ID28
  Adafruit NeoPixel
  libf3e
  TMC26XStepper
lib_ldf_mode = 1
src_filter   = ${common.default_src_filter}
monitor_baud = 250000
```