MarlinFirmware/Marlin #10224: MATLAB Interface instead of the LCD

AhmadGhzawi, 2018-03-26:
Hi there,

I wish to make a friendly user interface using MATLAB/GUI instead of the LCD. I know that i have to make a serial connection between the computer and arduino but i can't find where to write my commands of code in "ultralcd.cpp".
Any help will be much appreciated.

Thanks in advance.

AhmadGhzawi, 2018-03-26:
@Roxy-3D 
Thank you, it will. But, I'm not in that phase yet. I just want to get rid off the LCD at first, so everything the LCD displays and the commands it sends need to be done by MATLAB/GUI. How to make all of that happen? 
Thanks very much

Roxy-3D, 2018-03-26:
Just turn off the LCD (by not enabling one) at the bottom of Configuration.h      But really....   Especially when trying to debug, you are going to benefit from having a display that tells you where the machine thinks its nozzle is at, and the nozzle and bed temperature.   

I would leave the LCD Panel active...    It will burn some CPU cycles...   But pretty much, it is inert and won't affect the other work you are doing.

AhmadGhzawi, 2018-03-26:
you are absolutely right! but what i am trying to do is the following
instead of sending the commands using the LCD menus and controllers, i wish to control and monitor the whole process using matlab, in order to get rid of the LCD and SDcard after a while when everything is stable. So, my matlab interface should display the location of the nozzle (on X,Y, and Z axes), nozzle and bed temperature, feed rate,...etc

I want to build an interface just like the LCD to ease the process. But in order to do that, i need to know exactly where i should add new lines of code to govern the process and give exactly what i need.

Thank you very much for your help, it's really appreciated.

Roxy-3D, 2018-03-26:
You might want to look and PronterFace and Repetier Host.   Both are Open Source and used heavily in different Marlin applications.    If the printer is running, it will be hard to communicate X & Y information just because there is a lot of time spent doing serial communications and the printer will be moving during this time.    

But both PronterFace and Repetier Host dynamically track temperatures and the Z Height.

closed 2018-03-28 by thinkyhead