MarlinFirmware/Marlin #10226: BLTouch and Z offset not working

ikarisan, 2018-03-26:
Hi!

I am using Marlin bufix-1.1.x and recently upgraded from a capacitive sensor to an original BLTouch from Antclabs. But there seems to be some issues with the Z offset when using a BLTouch.

I mounted the sensor behind the nozzle using this mount: https://www.thingiverse.com/thing:2618601 
And after finally mounting it I had to set " Z_PROBE_OFFSET_FROM_EXTRUDER"  to around -2.0. That is very strange because "-x.xx" seems to be "sensor BELOW the nozzle". But it isn't! The sensor is 3-4mm above the tip of the nozzle! And when I set this value to +2 the nozzle stops at +5mm above the bed! And when using -1.xx the nozzle is getting much more closer to the bed!

The next step was to disable the software endstops und lower the nozzle to its final position (using the paper test). Marlin told me a Z offset of -1.09. So I executed "M851 Z-1.09" and saved this value with "M500". After a reset I executed "G28 Z" followed by a "G0 Z". The nozzle moved down and the gap between the nozze and the sheet of paper was larger than before. So I disabled the endstops again and lowered the nozzle. Suddenly I got an offset of -1.29! :( :( Using this new values results in a much larger gap.... :/

GhostlyCrowd, 2018-03-26:
Chances are your bl touch is too low or too high. the shoulder of the bltouch (the plastic where the pin goes to) should be no greater then 8mm and no less than 7.5mm above the bed when the nozzle is touching the bed. the pin while retracted should be 2mm +/- .2mm from the bed with the nozzel touching the bed.  

 Once you have it set properly the bltouch is very accurate and everything else works. I am also using 1.1.x bugfix with my bltouch no issues.

ikarisan, 2018-03-27:
Thanks! That gave me the needed hint! I used the wrong data sheet. The currently sold "BLTouch smart" needs a distance of 2.5mm +/- 0.5mm! So, with my "The sensor is 3-4mm above the tip of the nozzle!" I was quite out of the tolerance. :(

closed 2018-03-28 by thinkyhead