MarlinFirmware/Marlin #10231: Filament runout sensor on RAMPS 1.4

MariusBeul, 2018-03-27:
When installing a filament runout sensor on a RAMPS 1.4 board, you have to define FIL_RUNOUT_PIN.
This however does only work when in line 66, #define SERVO3_PIN is uncommented sinze both share D4.

Zer0t3ch, 2018-03-28:
Does it fail to compile or not work in practice? Also: what branch are you on?

I was able to compile with FIL_RUNOUT_PIN enabled on the bugfix-2.0.x branch. (with the prerequisites NOZZLE_PARK_FEATURE and ADVANCED_PAUSE_FEATURE also enabled)

MariusBeul, 2018-03-28:
It compiles, but for me it does not work in practice. I am on  1.1.x. (1.1.8). Of course you have to enable NOZZLE_PARK_FEATURE and ADVANCED_PAUSE_FEATURE. Otherwise it will not compile.

Zer0t3ch, 2018-03-28:
> Of course you have to enable NOZZLE_PARK_FEATURE and ADVANCED_PAUSE_FEATURE. Otherwise it will not compile.

Yeah, that's what I was saying. Just wanted to make sure you had already done so.

Could you give `bugfix-2.0.x` a shot? In my experience so far, it's been just as stable as the main branch, enough for day-to-day use. Definitely stable enough to try one print and see if it works, in which case you can stay on that branch or look into the diff to see what may have changed/fixed it.

thinkyhead, 2018-03-28:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

Note that you can check the filament runout pin status with `M119`. You may or may not need `FIL_RUNOUT_INVERTING` and `FIL_RUNOUT_PULLUP`.

TwatBurglar, 2018-11-16:
@MariusBeul  This may not be related, but I had similar symptoms in that FilamentSensor connected to SERVO3_PIN was always stuck M119 TRIGGERED.   After much dikkery,  I discovered the ReArm board (a 3.3v cpu) has a unidirectional level-shifter on that pin, so evidently SERVO3 *cannot* be used as an input.   Connecting filament to SERVO2 instead (which does not have a level shifter) seems to toggle fine.