MarlinFirmware/Marlin #10232: Problem in branch 2.0.0.x M218 with DELTA

zogtrog, 2018-03-27:
Thinkyhead recently posted a fix to branch 2.0.0.x for delta printers with multiple nozzles to allow
z-positioning. I have built such a printer, but I can't get branch 2.0.0.x to compile. I am getting this error

Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\src\gcode\config\M218.cpp: In static member function 'static void GcodeSuite::M218()':

sketch\src\gcode\config\M218.cpp:77:82: error: 'planner' was not declared in this scope

       do_blocking_move_to_xy(current_position[X_AXIS], current_position[Y_AXIS], planner.max_feedrate_mm_s[X_AXIS]);

It is this code in M218.cpp that is causing the problem.

  #if ENABLED(DELTA)
    if (target_extruder == active_extruder)
      do_blocking_move_to_xy(current_position[X_AXIS], current_position[Y_AXIS], planner.max_feedrate_mm_s[X_AXIS]);
  #endif

Commenting out this line allow compilation. I am assuming this line is supposed to move the current hotend / tool to it's new position if it's already selected on a delta printer.

                                                                                  ^

zogtrog, 2018-03-27:
I also don't need stepper motors for my switching extruders / hot end positioning system. I am using a DC geared motor with limit switches and some electronics to control the hotend positioning. Switching on steppers to > 1 in this version consumes a whole load of pins which then become unusable to the rest of the system in pins_RAMPS.h.
With my system I need to write either a 1 or a zero to a pin using m42 and my control electronics takes care of the positioning on the nozzles.

my fudge around was to set  #define E1_CS_PIN -1 instead of 44 which seems to free the pin up for usage by M42.

Bob-the-Kuhn, 2018-03-27:
FYI on the -1 

The pins for the active sockets are put into the protected list.  M42 won't touch anything in the protected list.  Setting the pin to -1 says the pin doesn't exist so it doesn't get put into the protected list.

zogtrog, 2018-03-27:
Yeah I figured that - that's why I set it to -1. I am not using any stepper motors, but to use the SWITCHING_NOZZLE configuration I need to define 1 stepper motor, which was fine in version 1.1.8. and no conflict with pins 42 or 44. However there was no Z-POSITIONING for delta printers for two or more tilting hotends so I wanted to try out branch bugfix 2.0.0.x as Thinkyhead added this functionality 11 days ago. I could use the SERVO_PIN_NR 0 but I need to set angle which would correspond to it being totally off or on.

zogtrog, 2018-03-27:
Yeah I figured that - that's why I set it to -1. I am not using any stepper motors, but to use the SWITCHING_NOZZLE configuration I need to define 1 stepper motor, which was fine in version 1.1.8. and no conflict with pins 42 or 44. However there was no Z-POSITIONING for delta printers for two or more tilting hotends so I wanted to try out branch bugfix 2.0.0.x as Thinkyhead added this functionality 11 days ago. I could use the SERVO_PIN_NR 0 but I need to set angle which would correspond to it being totally off or on. This doesn't address the original issue which was that M218 is broken with Delta enabled.

zogtrog, 2018-03-27:
I don't know the code base well enough to start fixing issues in Marlin. I thought it would be better if someone who made changes recently fixed the M218 issue.

thinkyhead, 2018-03-28:
```cpp
#include "../../module/planner.h"
```

commit 114ab227 (2018-03-28): M218 with DELTA needs planner.h

Fix #10232
commit 38378cd0 (2018-03-28): M218 with DELTA needs planner.h

Fix #10232
closed 2018-03-28 by thinkyhead