MarlinFirmware/Marlin #10236: Firmware Filament retract while heating (~150C)

G3DBuilder, 2018-03-27:
I would like to see the Firmware do an Extruder retraction during the initial heating phase. Repetier Firmware has this feature.  It keeps the plastic from oozing all over the printer while the extruder heats up. On my printer that has repetier fw the extruder retracts 5mm when the temp reaches 140C.  I cannot find this setting in configuration.h or configuration_adv.h

Zer0t3ch, 2018-03-28:
Not sure how we would go about doing this. Maybe add it as part of M109? (I'd think not as a part of M104, to avoid tampering with motors in the background)

How does Repetier handle it? I'm looking around, and I can't find any clear documentation about Repetier or their GCode, specifically in relation to this.

AnHardt, 2018-03-28:
Dopends a bit on if in relative mode or not, but 
```
M140 S140
G0 Z-3
M140 S230
```
does not need an extra g-code.

Zer0t3ch, 2018-03-28:
@G3DBuilder If you want to use @AnHardt's suggestion and you're using CURA, put the first two lines of that gcode at the end of your "start" gcode in your printer config. It should work sufficiently to get it up to a minimal temp and retract, then the CURA-generated stuff will take over and set the actual long-term temp that you have set in your print profile.

It's probably doable to add what you want as an integrated part of the firmware gcode handling or something, but I doubt it's necessary given how well this "manual" solution should get the job done. The only point of that would be to offload the heat/retraction decisions to the firmware layer, but it seems redundant considering how easily it can be handled by existing gcode.

boelle, 2019-02-20:
@thinkyhead maybe change to feature request issue?

@G3DBuilder have you considered heating up when nozzle is 0.05 to 0.1mm from bed? i do that and have no ooze

closed 2019-05-16 by thinkyhead