MarlinFirmware/Marlin #10244: 2.0bugfix fails to compile because of too few arguments for gcode_line_error

reven, 2018-03-28:
### Description

2.0bugfix branch fails to compile on Arduino IDE 1.8.5. The complete error that the IDE gives is:
```sketch/src/gcode/queue.cpp: In function 'void get_serial_commands()':
sketch/src/gcode/queue.cpp:345:53: error: too few arguments to function 'void gcode_line_error(const char*, uint8_t)'
           gcode_line_error(PSTR(MSG_ERR_NO_CHECKSUM));
                                                     ^
sketch/src/gcode/queue.cpp:238:6: note: declared here
 void gcode_line_error(const char* err, uint8_t port) {
      ^
exit status 1
Error compilando para la tarjeta Anet V1.0.
```
I'm using the definition for Anet V1.0 boards

### Steps to Reproduce

1. Load file in IDE
2. Edit config
3. Compile

**Expected behavior:** For it to compile

**Actual behavior:** It throws the error above

#### Additional Information

* I've included a zip of my config files: [config files.zip](https://github.com/MarlinFirmware/Marlin/files/1857755/config.files.zip)

alexxy, 2018-03-28:
should be fixed by #10245

reven, 2018-03-28:
Thanks @alexxy. The change to queue.cpp does indeed fix the issue. I'll close this issue.

PR #10245 (2018-03-28): Make compile with platformio for stm32
This pull makes Marlin compile with pio and config from Marlin/src/config/examples/stm32f103ret6/Configuration.h

#8796 #8585 
also fixes #10244
closed 2018-03-28 by reven