MarlinFirmware/Marlin #10251: Bugfix-1.1.x and 2.0.x compile error with BLTouch enabled

JimBrown, 2018-03-29:
### Description

Both Bugfix-1.1.x and 2.0.x are unable to compile if BLTouch feature is enabled.

Source was downloaded 30 minutes ago from both branches.

The following error is generated:

`Bugfix-1.1.x`
```
C:\Users\jbwin\AppData\Local\Temp\arduino_build_9103\sketch\Marlin_main.cpp: In function 'void do_homing_move(AxisEnum, float, float)':

Marlin_main.cpp:2940:11: error: 'deploy_bltouch' was not declared in this scope

       if (deploy_bltouch) set_bltouch_deployed(false);

           ^

Using library LiquidCrystal at version 1.0.7 in folder: D:\Documents\arduino-1.9.0.Beta\libraries\LiquidCrystal 
exit status 1
'deploy_bltouch' was not declared in this scope
```
`Bugfix-2.0.x`
```
C:\Users\jbwin\AppData\Local\Temp\arduino_build_957829\sketch\src\module\motion.cpp: In function 'void do_homing_move(AxisEnum, float, float)':

C:\Users\jbwin\AppData\Local\Temp\arduino_build_957829\sketch\src\module\motion.cpp:1047:11: error: 'deploy_bltouch' was not declared in this scope

       if (deploy_bltouch) set_bltouch_deployed(false);

           ^

Using library LiquidCrystal at version 1.0.7 in folder: D:\Documents\arduino-1.9.0.Beta\libraries\LiquidCrystal 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

GhostlyCrowd, 2018-03-29:
I have this exact same error with Bugfix 2.0

Marlin\src\module\motion.cpp: In function 'void do_homing_move(AxisEnum, float, float)':
Marlin\src\module\motion.cpp:1047:11: error: 'deploy_bltouch' was not declared in this scope
if (deploy_bltouch) set_bltouch_deployed(false);

thinkyhead, 2018-03-29:
Patched.

closed 2018-03-29 by thinkyhead