MarlinFirmware/Marlin #10255: [2.0.x] Makefile does not seem to work

marcio-ao, 2018-03-29:
### Steps to Reproduce

When I attempt to build the bugfix-2.0.x branch, I get:

make: *** No rule to make target 'applet/wiring.o', needed by 'applet/src.elf'.  Stop.

I printed out the $(VPATH) and saw that the path to the "wiring.c" file was in there and it was correct. As best as I can figure it out, when Make tries to compile "applet/wiring.o", it is actually searching for "applet/wiring.c", which does not exist. If I go to the directory where "wiring.c" exists, and I create a symlink to "." called "applet", then it proceeds to compile (but encounters other errors).

thinkyhead, 2018-03-29:
This doesn't surprise me. There hasn't really been any attention paid to the `Makefile` in a long while.

marcio-ao, 2018-03-30:
Inexplicably, moving the Makefile up one directory to the "Marlin" directory, instead of the "src" directory, is getting me further in the process. I guess if I can get this to compile I'll make a PR with my changes. We rely on the Makefile for automatic builds so this is sort of important to us.

marcio-ao, 2018-03-30:
@thinkyhead: It looks like I am close to getting the Makefile to work, however, because it relies on VPATH, it does not work when files with duplicate names are present in multiple directories. It appear as if the only files that have this problem are the following:

./gcode/bedlevel/ubl/G29.cpp
./gcode/bedlevel/abl/G29.cpp
./gcode/bedlevel/mbl/G29.cpp
./gcode/bedlevel/ubl/M421.cpp
./gcode/bedlevel/abl/M421.cpp
./gcode/bedlevel/mbl/M421.cpp
./gcode/bedlevel/ubl/M421.cpp
./gcode/bedlevel/abl/M421.cpp
./gcode/bedlevel/mbl/M421.cpp
./HAL/servo.cpp
./module/servo.cpp
./HAL/HAL_STM32F7/persistent_store_impl.cpp
./HAL/HAL_DUE/persistent_store_impl.cpp
./HAL/HAL_TEENSY35_36/persistent_store_impl.cpp
./HAL/HAL_STM32F1/persistent_store_impl.cpp
./HAL/HAL_AVR/persistent_store_impl.cpp
./HAL/HAL_LPC1768/persistent_store_impl.cpp

Would you accept a PR that renamed these files so they had unique names? If that is not possible, then the Makefile would need to be rewritten from scratch and I think that may be beyond my abilities.

p3p, 2018-03-30:
I would prefer not having to have project wide unique filenames, my intention was to clean up the platform folders by removing the unnecessary prefixes when I get round to changing how we include the correct platforms headers, the current routing headers are a bit of a mess.

> We rely on the Makefile for automatic builds so this is sort of important to us.

Is there a reason you can't use the platformio build system in your automatic builds?, it is pretty clean system when you ignore Atom,

```platformio run -e megaatmega2560```

marcio-ao, 2018-03-30:
@p3p: I'm not the one who manages our build system, so I'm not sure what it would take to implement platformio. Historically at LulzBot we have either used Arduino IDE or the Makefile for builds and I would hate to see those options go away. I think choices are good!

The way I have gotten the Makefile to work is by adding the following lines:

```
# Add all non-HAL related files to the build
CXXSRC += $(notdir $(shell find src -name '*.cpp' -type f  -not -path "src/HAL/*"))
VPATH  += $(shell find src -path src/HAL -prune -o -type d -not -path "src/HAL/*")

# Add HAL related files to the build
CXXSRC += $(notdir $(wildcard src/HAL/*.cpp))
CXXSRC += $(notdir $(wildcard src/HAL/$(WHICH_HALL)/*.cpp))
VPATH  += src/HAL src/HAL/$(WHICH_HALL)
```

This methods works by finding all .cpp files and adding them to CXXSRC and then adding all the directories to VPATH. This works so long as there are no duplicates, as Make will only use the first file it finds in the VPATH.

I've separated out the HALL directory, so in fact, the only ones that need to have unique names are the bed leveling files, which could be renamed like this:

- ./gcode/bedlevel/ubl/G29.cpp  -> ./gcode/bedlevel/G29_ubl.cpp
- ./gcode/bedlevel/abl/G29.cpp  -> ./gcode/bedlevel/G29_abl.cpp
- ./gcode/bedlevel/mbl/G29.cpp -> ./gcode/bedlevel/G29_mbl.cpp
- ./gcode/bedlevel/ubl/M421.cpp -> ./gcode/bedlevel/M421_ubl.cpp
- ./gcode/bedlevel/abl/M421.cpp -> ./gcode/bedlevel/M421_abl.cpp
- ./gcode/bedlevel/mbl/M421.cpp -> ./gcode/bedlevel/M421_mbl.cpp

I believe this is really all that it would take to get the Makefile to work again.

thinkyhead, 2018-03-30:
> it is a pretty clean system when you ignore Atom

**Visual Studio Code** is a superior option.

marcio-ao, 2018-03-30:
> Visual Studio Code is a superior option.

Ahhh. Let the religious wars begin!

thinkyhead, 2018-03-30:
> Ahhh. Let the religious wars begin!

Project search is broken in Atom. Atom is also sluggish.

> the only ones that need to have unique names…

I would have to concur that imposing unique filenames on everything is not desirable, and should not be necessary in the 21st century when we've been compiling C for 50 years. If VPATH isn't working, then maybe a replacement for that can be concocted.

marcio-ao, 2018-03-30:
> If VPATH isn't working, then maybe a replacement for that can be concocted.

True. Fixing the Makefile is probably the best no-compromise solution.

> Project search is broken in Atom. Atom is also sluggish.

All coding should be done in "vi"

... just kidding :)

thinkyhead, 2018-03-30:
Are you using old-school `make` or the somewhat better `cmake`?

marcio-ao, 2018-03-30:
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

thinkyhead, 2018-03-30:
> All coding should be done in "vi"

Vi and Emacs users can't see their tabs and trailing whitespace and tend to leave these lying around.

thinkyhead, 2018-03-30:
This is a bit old, but should be adaptable… #5251

marcio-ao, 2018-03-30:
Okay. I'll see if I can find some good Makefile tutorials online and come up with something better.

I suppose if that fails, I could simply create a bunch of symlinks and give the files unique names that way, but that would be a pretty ugly hack.

thinkyhead, 2018-04-02:
https://github.com/tohara/marlin-cmake

marcio-ao, 2018-04-02:
Sorry for the repeated commits. I wanted to get a "clean" PR and ended up resetting the branch on my fork a couple times before I got it all squared away. 8546cc1 is good.

marcio-ao, 2018-04-02:
Here is the PR. No need to rename any files :)

https://github.com/MarlinFirmware/Marlin/pull/10281

marcio-ao, 2018-04-03:
Closing this out since it has been merged.

closed 2018-04-03 by marcio-ao