MarlinFirmware/Marlin #10258: ENABLE_LEVELING_FADE_HEIGHT causes back corner to be incorrect with Bilinear Bed Levellng

asantoni, 2018-03-29:
### Description

The ENABLE_LEVELING_FADE_HEIGHT flag causes bilinear bed leveling to consistently probe too low on the final point in a 9-point mesh. I tried to figure out why bilinear bed leveling kept crashing the nozzle into the bed just in the last corner of the G29 bed leveling process, and this is why. 

I tried it like 25 times, it reproduced the issue every time, until I made a single change to the firmware - I commented out ENABLE_LEVELING_FADE_HEIGHT, on a hunch after reading bug #9529.

With ENABLE_LEVELING_FADE_HEIGHT:

```Bilinear Leveling Grid:
      0      1      2
 0 +0.997 +0.091 -0.359
 1 +0.721 -0.070 -0.557
 2 +0.391 -0.404 -3.070```
(Note the last value is crazy low - For some reason Marlin isn't picking up the Z probe value correctly just in that one corner.)

After disabling ENABLE_LEVELING_FADE_HEIGHT:

```Bilinear Leveling Grid:
      0      1      2
 0 +0.992 +0.088 -0.354
 1 +0.724 -0.076 -0.547
 2 +0.437 -0.357 -0.875```

These runs were about 4 minutes apart, and no mechanical or thermal changes were made, just this 1 line of code.

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. Enable AUTO_BED_LEVELING_BILINEAR and ENABLE_LEVELING_FADE_HEIGHT.
2. Set your Z_PROBE_OFFSET_FROM_EXTRUDER. I tried all sorts of values, and they all reproduced the bug.
3.  Compile Marlin and flash the fresh firmware.
4. Run G28, then a G29. 
5. Watch the bed leveling get an accurate value for the first 8 points, then crash the nozzle deep into the bed on the final (9th) point. From the paste above, you can see it pushes about 3 mm below the bed.

I'm using the bugfix-1.1.x branch from March 27th, 2018.

**Expected behavior:** 

ENABLE_LEVELING_FADE_HEIGHT should not cause the last bed height probe point to mess up.

**Actual behavior:** 

It smashes 3 mm below my bed before it thinks it's hit the bed.

Thanks!

thinkyhead, 2018-04-01:
> ENABLE_LEVELING_FADE_HEIGHT should not cause the last bed height probe point to mess up.

Indeed! I can find no cause for this. Do you get the same problem if you include `F0` in your `G29` command?

thinkyhead, 2018-04-01:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

houseofbugs, 2018-04-10:
It looks like the probe isn't triggering. Likely a hardware issue. I cannot replicate with 1.1.8.

thinkyhead, 2018-04-12:
We've also recently added `Z_PROBE_LOW_POINT` to prevent probes from getting harmed. Get the latest to use this option.

boelle, 2019-02-20:
@asantoni tried what thinky suggested?

closed 2019-02-20 by thinkyhead

asantoni, 2019-02-20:
@boelle: No, I ran out of time and just ended up making too many other changes to my printer. (I updated Marlin, rebuilt it twice, switched Z probes 3 times, etc. - I haven't had the problem since, but I have definitely kept ENABLE_LEVELING_FADE_HEIGHT disabled since this bug report.) Feel free to leave closed unless someone else reproduces in latest Marlin.