MarlinFirmware/Marlin #10259: Marlin and platformio.io

vMeph, 2018-03-30:
this is not a issue, just would like to know what option we have in making updates on firmware
everytime new things i sadd to the bugfixes you got redo project in atom plaformio, was wondering if there is any way to only do the updates of new commits with out losing the settings on platformio

thanks

thinkyhead, 2018-03-30:
Which settings are you losing? How are you maintaining your local copy? You should be able to obtain the latest changes from `upstream` while preserving your local changes to `platformio.ini`.

vMeph, 2018-03-30:
@thinkyhead im very new to atom platformio never used it before so im not sure how to mantain and how to do upstream i basikly have downloaded marlin bug fix and created a  project folder, so im not sure how to get the new updates from git while preserving my local changes....
it would be great if you could point me out the steps to get that acomplished

thanks

thinkyhead, 2018-03-30:
I don't use Atom. (**Visual Studio Code** is superior to **Atom**. Project search is broken in Atom. Atom is also sluggish.) I also don't have any issues with losing my local changes, since I don't try to modify the branches named `bugfix-1.1.x` etc. If you want to ensure your changes are preserved, make new branches and apply your changes there.

mamama1, 2018-04-10:
How does that work in terms of main.cpp/marlin.ino? I mean, platformio does not use ino files.
What about moving Marlin completely to PlatformIO? I love the library manager and the possibility to add library version dependencies to a project. Especially in the recent past there were some issues with incompatible TMC libs when using the most recent ones with Marlin 1.1.8, for example, which could have been completely avoided by setting mandatory version dependencies within the project...

thinkyhead, 2018-04-12:
> What about moving Marlin completely to PlatformIO?

Marlin completely supports PlatformIO and it completely supports Arduino IDE for those who just want to build for AVR and the other platforms that Arduino IDE supports. We aren't dropping support for Arduino IDE. Note that the new `Marlin.ino` file is empty except for commentary.

Adremelej, 2018-07-12:
Hello @thinkyhead. How can I use VScode to write code for Arduino? I want to add a Nextion LCD

closed 2018-08-12 by vMeph

thinkyhead, 2018-08-13:
@Adremelej — Sorry, very busy lately… Hope you figured it out. See http://marlinfw.org/docs/basics/install_platformio.html for full instructions.