MarlinFirmware/Marlin #10262: Autobed leveling unstable

mogey336, 2018-03-30:
I made Hbot printer , i using marlin 1.1.x ,I'm using proximity sensor 8 mm range
The problem is that the bed leveling some time works good and some time does work at all 
The bed will move up then suddenly it moves down the proximity even does activate(Red light on the sensor)
I try M119 command the limit swith is open not triggered 
It's there a bug or something!!!!

thinkyhead, 2018-03-30:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

---
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll check them over and see if anything is amiss.

---
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

mogey336, 2018-04-01:
Thanks for your reply
I realized that the GND connection of the proximity sensor was not nearly loose after i fixed it the autolevel was correct , I will feedback you if it does it again

closed 2018-04-02 by thinkyhead

mogey336, 2018-04-09:
The problem appeared again 
[log.txt](https://github.com/MarlinFirmware/Marlin/files/1891276/log.txt)

mogey336, 2018-04-09:
I don't have winzip program so here are my configuration and Configuration_adv
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1891316/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1891317/Configuration_adv.txt)

thinkyhead, 2018-04-10:
I'm sure it's not a bug in Marlin, and is unlikely to be fixed by changes in configuration. Try asking for help from the community in these forums:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)