MarlinFirmware/Marlin #10263: [2.0] MSK SBASE - Error:EEPROM datasize error.

manutenfruits, 2018-03-30:
### Description

Gcode M500 to save EEPROM doesn't work, throws error.

### Steps to Reproduce

1. Reset board
2. Send M500

**Expected behavior:** The settings are saved and you get back control of the printer.

**Actual behavior:** Throws an error and OctoPrint cuts communication.

#### Additional Information

* I tried to isolate the issue by changing the minimum things and discard that it was some misconfiguration. My `Configuration.h` is just uncommenting `EEPROM_SETTINGS`, adding `DEBUG_EEPROM_READWRITE` like I read on another issue and setting the board to `BOARD_MKS_SBASE`. You can check out all the files on my [GitHub fork](https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...manutenfruits:test-m500).
* The error seems to be related to some format mismatch with `esteppers`. Check out the console output on M500:

```
Send: M500
[...]
Recv:  write_data(100,268467976,4...)
Recv:  f_lseek()=9
Recv: Error:Field esteppers mismatch.
Changing monitoring state from 'Operational' to 'Error: Field esteppers mismatch.
'
Recv:  write_data(102,268467958,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443364,16...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443384,16...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443432,16...)
Recv:  f_lseek()=9
Recv:  write_data(102,268441992,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443412,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443408,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443400,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443428,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443404,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443464,16...)
Recv:  f_lseek()=9
Recv:  write_data(102,268441920,12...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467984,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467959,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467960,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467988,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467961,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467962,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468024,8...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468016,8...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467963,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467964,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467992,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467996,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468000,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443668,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443664,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268443660,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468004,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467968,2...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467965,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468068,32...)
Recv:  f_lseek()=9
Recv:  write_data(102,268441193,1...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468044,22...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468100,44...)
Recv:  f_lseek()=9
Recv:  write_data(102,268468008,6...)
Recv:  f_lseek
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv:  write_data(102,268467972,4...)
Recv:  f_lseek()=9
Recv: ok
```

p3p, 2018-03-30:
This has come up a bit recently, the LPC1768 platform running on the MKS SBASE seems to have an issue reading the onboard sdcard (where we save the eeprom.dat file on the LPC1768 platform), I assume that you can't access the usb drive to flash firmware either?

manutenfruits, 2018-03-30:
> I assume that you can't access the usb drive to flash firmware either?

I'm not sure I understand. My process to flash the firmware is to copy it on the sd card and pop it in the board. I've connected to the board using USB and that used to work fine, the SD would be mounted fine as an external drive in my computer, but haven't tried with Marlin.

p3p, 2018-03-30:
It should also be mounted as an external drive with Marlin, well unless the sd-card cannot be accessed. As flashing does work, meaning the boot-loader can access the sd-card, the issue must be with the pin configuration but unfortunately I don't have an mks sbase to test with and everything works as expected on my re-arm.

thinkyhead, 2018-04-01:
The error is interesting. "`Field esteppers mismatch`" indicates that the eeprom index testing is failing right on the first field to be written. Try adding some more logging so we can see what's going on with the test values:

```diff
+   SERIAL_ECHO_START_P(port);
+   SERIAL_ECHOPAIR_P(port, "eeprom_index=", eeprom_index);
+   SERIAL_ECHOLNPAIR_P(port, "Expected index is ", offsetof(SettingsData, esteppers) + EEPROM_OFFSET);
    _FIELD_TEST(esteppers);
```

p3p, 2018-04-01:
@thinkyhead I think that may be a false positive due to the sd-card access error though I could be jumping to conclusions, `f_lseek()=9` indicates the file was never opened (empty data structure) so the seek failed.

manutenfruits, 2018-04-01:
@thinkyhead I added the logging, this is what I got.
```
Send: M500
Recv:  write_data(100,268467976,4...)
Recv:  f_lseek()=9
Recv: echo:eeprom_index=102Expected index is 106
Recv: Error:Field esteppers mismatch.
Changing monitoring state from 'Operational' to 'Error: Field esteppers mismatch.
'
Recv:  write_data(102,268467958,1...)
Recv:  f_lseek()=9
```

manutenfruits, 2018-04-01:
I used a different SD card and it worked perfectly! I plugged it in my computer and I can see an `eeprom.dat` file now. Not sure what the problem is, but top card (4GB class 6, Transcend MicroSDHC) *doesn't* work, but bottom card (16GB Class 10, Centon MicroSDHC) *does* work just fine. Funny thing is, the one that works fine here had trouble running in a Raspberry Pi before. I need a better SD Card  supplier 😒 
[image]

Thanks everybody for helping me troubleshoot!

closed 2018-04-01 by manutenfruits

p3p, 2018-04-01:
As nice as it is to know the onboard sd-card does work on the MKS-SBase, that was unexpected. On other issues changing the sd-card hadn't seemed to help (and I've had no problams with any of mine diverse collection on the Re-ARM), can you confirm flashing was working correctly from the other card, if it was there should be no reason Marlin cannot access it as well.

manutenfruits, 2018-04-01:
so reading works fine, but in the faulty card it seems to be unable to write to it.

thinkyhead, 2018-04-02:
Hmm. Hopefully in future we can provide a better error message in the case of faulty SD card writes.