MarlinFirmware/Marlin #10264: [bugfix-1.1.x] G33 / G29 does not work. (Kossel Delta Servo Probe)

yoch2015, 2018-03-31:
Hello.
The servo probe does not work with G33 / G29.
It does not work in M401 / M402, but M280 P0 S0 and M280 P0 S90
 work properly.
Also, Z Probe Fail error occurs in M401 / M402.
Thank you.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1865426/Configuration.h.zip)

yoch2015, 2018-03-31:
Debug Log Send.

[TestCase1]
1. G28 [ok]
2. M280 P0 S0 [ok]
3. M280 P0 S90 [ok]
4. M401 [no deploy]
5. M402 [z probe fail]

[TestCase2]
1. G28
2. G0 Z15
3. M401 [move up,no deploy]
4. M402 [z probe fail]

[M401M402_Log.txt.zip](https://github.com/MarlinFirmware/Marlin/files/1865999/M401M402_Log.txt.zip)

thinkyhead, 2018-04-01:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

thinkyhead, 2018-04-01:
Pretty strange. Everything looks ok in your configuration. And the code to deploy the servo is present in the `set_probe_deployed` function.

thinkyhead, 2018-04-01:
Try this: Enable `PINS_DEBUGGING` and then use `M43 S` to do the servo probe test. You should also do a plain `M43` to make sure there are no pins conflicts.

yoch2015, 2018-04-01:
Thank you reply.
I tested it with M43, but it seems that it worked normally.
But, M401 / M402 fails.
[M43S_LOG.txt.zip](https://github.com/MarlinFirmware/Marlin/files/1866018/M43S_LOG.txt.zip)

thinkyhead, 2018-04-01:
I'm not able to compile Marlin `bugfix-1.1.x` using your `Configuration.h` file due to several errors. Are you sure you've attached the correct configuration? The errors are:
```
MKS 1.3/1.4 supports up to 2 hotends / E-steppers. Comment out this line to continue.
Please enable only one probe option: PROBE_MANUALLY, FIX_MOUNTED_PROBE, BLTOUCH, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo.
E2_STEP_PIN, E2_DIR_PIN, or E2_ENABLE_PIN not defined for this board.
DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.
DEFAULT_MAX_FEEDRATE has too many elements.
DEFAULT_MAX_ACCELERATION has too many elements.
```

If you're not actually using `bugfix-1.1.x` then you might want to start with a fresh copy. Meanwhile, ZIP and attach your whole `Marlin` folder so we can better see what's going on.

thinkyhead, 2018-04-02:
I've just tested `bugfix-1.1.x` with basic servo probe configuration, and both `M401` and `M402` operated normally. I will check again using some modifications from your attached configuration and see if some other setting is involved.

thinkyhead, 2018-04-02:
The only thing I can see that could be potentially problematic with those additional files is that `E2_DIR_PIN` and `E2_CS_PIN` are being assigned to pins 1 and 0. As I understand it, pins 0 and 1 are reserved for the built-in serial port, and assigning them to other functions could lead to strange behavior. I don't see any how that could affect `M401 /`M402`, but you should try some other pins just to rule out the conflict. Be sure to also use the latest `bugfix-1.1.x` so that we are comparing apples to apples.

yoch2015, 2018-04-02:
Thank you.
Based on the test results, I tried to fix it to 2 stepper, but the result did not change.
I changed the example_configurations from kossel_mini to generic and tested it at branch # be098cd and it worked normally.
Thank you very much.

yoch2015, 2018-04-02:
close issue

closed 2018-04-02 by yoch2015