MarlinFirmware/Marlin #10265: UBL +  MESH_INSET = cannot print outside of mesh?

alexxy, 2018-03-31:
Hi!

I'm using UBL and since my Z-Probe has large offset i use MESH_INSET to fix issue with it.
However it seems quite strange that i cannot print outside UBL mesh....

I'm using bugfix-2.0.x branch 
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1865438/configs.zip)
[image]

Roxy-3D, 2018-03-31:
Can you move the nozzle around off the mesh?    Can you do a G29 W T and post the results here?    And then do a G26 Mesh Validation pattern and post a picture of that?

alexxy, 2018-03-31:
I can move nozzle, but it cannot print

```
Send: G29 W T
Recv: Unified Bed Leveling System v1.01 active.
Recv: Mesh 1 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 10.0000
Recv: # of samples: 9
Recv: Mean Mesh Height: 0.634069
Recv: Standard Deviation: 0.194173
Recv: zprobe_zoffset: -0.9000000
Recv: MESH_MIN_X  ((((((200) / 2) - (200) / 2) + 50)>(0)?((((200) / 2) - (200) / 2) + 50):(0)))=50
Recv: MESH_MIN_Y  ((((((200) / 2) - (200) / 2) + 50)>(0)?((((200) / 2) - (200) / 2) + 50):(0)))=50
Recv: MESH_MAX_X  ((((((200) / 2) + (200) / 2) - (50))<(200)?((((200) / 2) + (200) / 2) - (50)):(200)))=150
Recv: MESH_MAX_Y  ((((((200) / 2) + (200) / 2) - (50))<(200)?((((200) / 2) + (200) / 2) - (50)):(200)))=150
Recv: GRID_MAX_POINTS_X  3
Recv: GRID_MAX_POINTS_Y  3
Recv: MESH_X_DIST  50.00
Recv: MESH_Y_DIST  50.00
Recv: X-Axis Mesh Points at: 50.000  100.000  150.000
Recv: Y-Axis Mesh Points at: 50.000  100.000  150.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,2)           (2,2)
Recv: (50,150)        (150,150)
Recv:   0.551    0.755  [ 0.933]
Recv: 
Recv:   0.404    0.621    0.863
Recv: 
Recv:   0.263    0.555    0.762
Recv: (50,50)            (150,50)
Recv: (0,0)             (2,0)
Recv: ok
```

Roxy-3D, 2018-03-31:
That is strange...   The same motion routines are used for the X & Y axis as the extruder.  If the X & Y can move off the mesh, the extruder should be able to move while off the mesh.

Can you do a G26 so we can see the location of the mesh and its boundaries?

gloomyandy, 2018-03-31:
When the head moves off the mesh what actually happens? Does extrusion stop or does the nozzle lift (or crash) stopping the the filament from staying on the bed? 

Also could you explain why you need to use MESH_INSET? I've found having the mesh cover all of the bed keeps things simple. It is possible to use MIN_PROBE_EDGE to stop the probe being used on the edges of the bed. So in your case you could have a MIN_PROBE_EDGE of 50 and a MESH_INSET of 0. This would have a mesh that covers all of the bed (using G20 P3 or manual probing to fill in the missing points), but the probe would only be used in the same area as you have now.

thinkyhead, 2018-04-01:
> Does extrusion stop or does the nozzle lift (or crash) stopping the the filament from staying on the bed?

If the nozzle was to move down to 0 when outside the mesh —thus blocking the nozzle opening— that would prevent extrusion. Try enabling and setting this value:
```cpp
  //#define UBL_Z_RAISE_WHEN_OFF_MESH 2.5 // When the nozzle is off the mesh, this value is used
                                          // as the Z-Height correction value.
```

alexxy, 2018-04-01:
I need MESH_INSET because i have large Z_PROBE_OFFSETS (10, -40, -0.9). So If I dont use it, probe cannot probe the bed. 

It seems when nozzle is out of mesh, then it crashed on the table (difference ~0.6mm)

So does UBL interpolate tilt when its out of mesh? And it seems that UBL doesnt use Z_PROBE_OFFSETS values when generating mesh

gloomyandy, 2018-04-01:
UBL should work fine with your probe settings without needing the large MESH_INSET. What will probably happen is that it will not probe some parts of your bed (because the probe can not get there). So after a G29 P1 you will have some points that have not been filled in (you can see this if you do a G29 T). You can either fill these points in manually, or you can fill them automatically using G29 P3. That way you will have a mesh that covers all of the area you want to print on. It really doesn't make sense to try printing to areas that are not covered by the mesh. 

You may find that to get good automatic probe coverage of the bed you need to specify a larger number of mesh points than you expect. So for instance rather than using a 3x3 mesh you may find a 4x4 or 5x5 works better because with the 3x3 mesh two of the rows/cols may not get probed because the probe can not reach the mesh points 

What happens if you do not set the large MESH_INSET? If only a small number of points are probed try increasing the mesh size and try it again.  

The following provides a pretty good description of how UBL works and in particular how the to handle parts of the mesh that can not be reached by the probe...
http://marlinfw.org/docs/features/unified_bed_leveling.html

alexxy, 2018-04-01:
Problem is that when i dont set mesh inset hotend crashes on heatbed when the probe is actualy outside of bed area. So good idea is to take Z_PROBE_OFFSET into accaunt while generating mesh.

gloomyandy, 2018-04-01:
If it is trying to probe an area that is not on the bed then something is wrong with that. UBL will normally use the probe offset settings to ensure it only probes within the bed. I have a similar setup to you (my probe is at -23, -40 offset) and UBL works fine only probing within the bed, but I'm using the 1.1.x bugfix, not 2.0.

alexxy, 2018-04-01:
OK. I'll update to latest code today and check if there is a problem.
And will test with bugfix-1.xx

alexxy, 2018-04-02:
G29 W T with 4 points, some of them are unreachable for Z_PROBE
```
Send: G29 W T
Recv: Unified Bed Leveling System v1.01 inactive.
Recv: Mesh 0 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 10.0000
Recv: # of samples: 12
Recv: Mean Mesh Height: 0.887864
Recv: Standard Deviation: 0.176962
Recv: zprobe_zoffset: -0.8000000
Recv: MESH_MIN_X  ((((((200) / 2) - (200) / 2) + 10)>(0)?((((200) / 2) - (200) / 2) + 10):(0)))=10
Recv: MESH_MIN_Y  ((((((200) / 2) - (200) / 2) + 10)>(0)?((((200) / 2) - (200) / 2) + 10):(0)))=10
Recv: MESH_MAX_X  ((((((200) / 2) + (200) / 2) - (10))<(200)?((((200) / 2) + (200) / 2) - (10)):(200)))=190
Recv: MESH_MAX_Y  ((((((200) / 2) + (200) / 2) - (10))<(200)?((((200) / 2) + (200) / 2) - (10)):(200)))=190
Recv: GRID_MAX_POINTS_X  4
Recv: GRID_MAX_POINTS_Y  4
Recv: MESH_X_DIST  60.00
Recv: MESH_Y_DIST  60.00
Recv: X-Axis Mesh Points at: 10.000  70.000  130.000  190.000
Recv: Y-Axis Mesh Points at: 10.000  70.000  130.000  190.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,3)                   (3,3)
Recv: (10,190)                (190,190)
Recv:      .          .          .          .
Recv: 
Recv:   0.757    0.872  [ 1.078]   1.132
Recv: 
Recv:   0.597    0.847    0.998    1.147
Recv: 
Recv:   0.587    0.733    0.900    1.006
Recv: (10,10)                    (190,10)
Recv: (0,0)                     (3,0)
Recv: ok
```

Roxy-3D, 2018-04-02:
Yes...   this is expected if you have an X and Y _OFFSET_FROM_EXTRUDER.    You can fill those points with a G29 P3.      And once you have done a G26, you can use the Mesh Validation Pattern to edit those points to exactly reflect reality.

alexxy, 2018-04-02:
@Roxy-3D why not interpolate mesh data outside of probbed mesh? like it does bilinera bed leveling

alexxy, 2018-04-02:
Ok. Seems like i miss the concept of UBL. So G29 P3 actualy do extrapolation.

Roxy-3D, 2018-04-02:
Because the mesh should cover the usable print envelope.    But I'm currently looking at adding an option to interpolate the Z-Height correction when the nozzle is off the mesh.    Right now, there is a Z-Lift option for when the nozzle is off the mesh.   That option will remain.   But some flavor of extrapolation will be there.    And maybe configurable.

G29 P3 does a 'Smart Extrapolation' to fill the unreachable points.   But it is much to expensive to do during an actual print.   Something more 'cost effective' will have to be done.

thinkyhead, 2018-04-03:
The automatic extrapolation of mesh height where your delta cannot reach was based on the original "delta bed leveling" by @jcrocholl. ABL and UBL improved on that by allowing you to set new Z values for those points, making it best in terms of control to have the mesh extend all the way to the edges of the print area.

The main reason I ended up adding extrapolation options for the ABL mesh was to accommodate a SCARA printer where the mesh area can be specified to be a small square area, although the machine itself can move nearly 360° in a large (800mm wide) area.

The option to "continue the slope" was overkill for that machine, so it uses the "stay level past the edge" behavior. The "continue the slope" option works ok for small beds where the area outside the mesh is usually only a centimeter or two at most.

In both cases, there's no extra computational cost because they simplify the bilinear interpolation. One interpolates the outer edge with itself ("stay level") and the other uses a "proportion of the grid box" greater than 1.0 or less than 0.0 ("continue the slope").

Roxy-3D, 2018-04-03:
In the case of extending the edge Z height correction I agree, there is no extra computation.
And I think that should be one of the options.   

The other option I want to add is similar to the Smart fill for UBL's G29 P3 to populate unprobed mesh points.    It looks at the angle the bed is going up or down.   If the bed appears to be going down, it just extends the near by mesh points in that direction because that is safest.    If the bed appears to be going up, it calculates the angle of the rise and extends that to the mesh point.

I suspect there will be other ideas that appear as I start working through this.