MarlinFirmware/Marlin #10266: probe offset from extruder

vMeph, 2018-03-31:
when i put the correct values from the distance betwen bl touch probe and hotend
```
#define X_PROBE_OFFSET_FROM_EXTRUDER -21.5  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -11.6  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.37  // Z offset: -below +above  [the nozzle]
```
i get the folowing errors:

```#define MAX_PROBE_X _MAX_PROBE_X
^
Marlin\SanityCheck.h:894:53: note: in expansion of macro 'MAX_PROBE_X'
#if !WITHIN(Z_SAFE_HOMING_X_POINT, MIN_PROBE_X, MAX_PROBE_X)

^
Marlin\Configuration.h:702:39: error: floating constant in preprocessor expression
#define Y_PROBE_OFFSET_FROM_EXTRUDER -11.6  // Y offset: -front +behind [the nozzle]
^
Marlin\macros.h:150:32: note: in definition of macro 'WITHIN'
#define WITHIN(V,L,H) ((V) >= (L) && (V) <= (H))
^
Marlin\Conditionals_post.h:1007:27: note: in expansion of macro 'max'
#define _MIN_PROBE_Y (max(Y_MIN_BED + MIN_PROBE_EDGE, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
^
Marlin\Conditionals_post.h:1007:71: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'
#define _MIN_PROBE_Y (max(Y_MIN_BED + MIN_PROBE_EDGE, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
^
Marlin\Conditionals_post.h:1022:25: note: in expansion of macro '_MIN_PROBE_Y'
#define MIN_PROBE_Y _MIN_PROBE_Y
^
Marlin\SanityCheck.h:896:42: note: in expansion of macro 'MIN_PROBE_Y'
#elif !WITHIN(Z_SAFE_HOMING_Y_POINT, MIN_PROBE_Y, MAX_PROBE_Y)
^
compilation terminated due to -fmax-errors=5.
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\megaatmega2560\src\Sd2Card.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\Max7219_Debug_LEDs.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\M100_Free_Mem_Chk.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\Marlin_main.cpp.o] Error 1
```

im guessing this is cause its does accept floating value?

thats the exact values off set from hotend, how to go about it?

Roxy-3D, 2018-03-31:
The compiler is complaining because Z_SAFE_HOMING_X_POINT is not with the range specified by   MIN_PROBE_X and  MAX_PROBE_X.

What are the values for those 3 items?

vMeph, 2018-03-31:
@Roxy-3D  from what i see the X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER dont accept floating values  (.x)
the Z accepts .x values but for some reason x and y dont

vMeph, 2018-03-31:
Z Probe to nozzle (X,Y) offset, relative to (0, 0).
 *   X and Y offsets must be integers.
if X and Y must be integer how i compensate de X .5 and Y .6 values to be exact?

th33xitus, 2018-03-31:
Do you really think that half a millimeter in X and Y offset makes such a difference in your probing results?

Roxy-3D, 2018-03-31:
The reason those don't take floating point values is because the pre-processor can not do floating point math.  (Which is, admit-ably, lame...)    But a 1/2 mm either way isn't going to affect things.   Just get the X and Y value as close as possible to being accurate.

closed 2018-03-31 by thinkyhead

vMeph, 2018-03-31:
thanks for the explanation!!!

thinkyhead, 2018-04-01:
Yes (_**admittedly**_) it is lame. I've pushed changes so this sanity-check uses `static_assert` instead — now in `bugfix-1.1.x` and `bugfix-2.0.x`. See if that resolves all the compile errors for you.