MarlinFirmware/Marlin #10267: Probing off of bed and other issues

nkuck, 2018-03-31:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

gloomyandy, 2018-03-31:
This may be a difference between 1.1.8 and the bugfix (which I'm using), but for me G28 will place the probe in the centre of the bed not the nozzle.

Where in relation to your nozzle is your probe (in front/behind, left, right - all when looking from the front of the machine). Having a positive value for your Z probe offset seems a little odd to me. My understanding is that for an inductive type probe the offset normally describes the "virtual" position at which the probe triggers. Typically this will be below the nozzle tip (you want the probe to trigger before the nozzle hits the bed!) and so it will have a negative value, even though the position of the "end" of the probe is above the nozzle.  So for instance if your probe triggers when the nozzle is 2mm above the bed then you need to set the Z probe offset to -2.0

nkuck, 2018-03-31:
I tuned it in to place the nozzle at bed center. Didn’t know it should be the probe. I can certainly retune it. My probe is in front of, and to the left of, the nozzle. The probe tip is about 0.5mm above the nozzle. That seems to work pretty well, so far.

Noel Kuck

Live simply so others can simply live!

ManuelMcLure, 2018-03-31:
You might want to update to bugfix-1.1.x - a lot of work has gone into ensuring that probes don't go off the bed. Also, assuming you're no longer using the Z-endstop and instead are using the Z-probe for himing you should definitely also set Z_SAFE_HOMING to ensure that it homes with the probe in the center of the bed.

If you actually measure the probe offset Marlin should automagically place the probe at the center of the bed when you set Z_SAFE_HOMING.

nkuck, 2018-03-31:
I have safe homing enabled:
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
#define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

Roxy-3D, 2018-03-31:
If you turn on #define DEBUG_LEVELING_FEATURE and do a M111 S255 prior to doing a G28....

The output will tell us a lot about what is going wrong.    It will be especially helpful if you indicate exactly in the log text where the probe goes off the bed.

nkuck, 2018-03-31:
My probe/nozzle settings are:
define X_PROBE_OFFSET_FROM_EXTRUDER -40
#define Y_PROBE_OFFSET_FROM_EXTRUDER -22
#define Z_PROBE_OFFSET_FROM_EXTRUDER +0.5
I have a 220 x 200 Cartesian bed, and G28 set the nozzle squarely in the center of the bed. So at least that part looks good.
G29 places the probe just on the very left edge of the bed (about 110mm to the left of bed center), and two positions off of the front of the bed, about 140mm from bed center.

Noel Kuck

Live simply so others can simply live!

gloomyandy, 2018-03-31:
You shouldn't really be "tuning" the probe offset x and y values. They need to reflect the actual position of the probe relative to the nozzle. These figures are then used to position the probe when probing the bed.

nkuck, 2018-03-31:
ROXY-3D
I turned on the debugger, but not sure on what I should be focusing. Not too helpful, for me.

nkuck, 2018-03-31:
I shouldn’t have said “tuning”, but rather “adjusting” the offsets to keep the probe on the bed. Rather unsuccessfully.

Noel Kuck

Live simply so others can simply live!

nkuck, 2018-03-31:
I actually measured the probe offsets from the nozzle and entered those dimensions.

Noel Kuck

Live simply so others can simply live!

thinkyhead, 2018-03-31:
Some of our canned responses:

---
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

---
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. (This will also enable bed leveling.)
- Copy the log output into a **.TXT file** and attach it to your next reply.

nkuck, 2018-04-01:
Here is my console dump. Does anything look wrong? Probing two spots off of the left side of bed, and one in front of bed front. Also, the nozzle and probe locations seemed to be swapped.
SENT: M111 S247
READ: echo:M111 S247
M111 S247
READ: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
READ: ok
SENT: M105
READ: echo:M105
M105
READ: ok T:21.41 /0.00 B:22.19 /0.00 @:0 B@:0
SENT: M105
READ: echo:M105
M105
READ: ok T:21.41 /0.00 B:22.11 /0.00 @:0 B@:0
SENT: M105
READ: echo:M105
M105
READ: ok T:21.52 /0.00 B:22.34 /0.00 @:0 B@:0
SENT: G28
READ: echo:G28
G28
READ: >>> gcode_G28
READ: Machine Type: Cartesian
READ: Probe: FIX_MOUNTED_PROBE
READ: Probe Offset X:-30 Y:-22 Z:0.50 (Left-Front & Above Nozzle)
READ: Auto Bed Leveling: 3POINT (enabled)
READ: ABL Adjustment X+0.00 Y+0.00 Z-0.08
READ:   current_position=(200.00, 42.00, 4.42) : sync_plan_position
READ:   current_position=(200.00, 42.00, 4.42) : setup_for_endstop_or_probe_move
READ: > endstops.enable(true)
READ: >>> homeaxis(X)
READ: Home 1 Fast:
READ: >>> do_homing_move(X, -322.50, 0.00)
READ:   current_position=(0.00, 42.00, 4.42) : sync_plan_position
SENT: M105
READ: echo:busy: processing
busy: processing
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(X)
READ: Move Away:
READ: >>> do_homing_move(X, 5.00, 0.00)
READ:   current_position=(0.00, 42.00, 4.42) : sync_plan_position
READ: <<< do_homing_move(X)
READ: Home 2 Slow:
READ: >>> do_homing_move(X, -10.00, 25.00)
READ:   current_position=(0.00, 42.00, 4.42) : sync_plan_position
READ: <<< do_homing_move(X)
READ: >>> set_axis_is_at_home(X)
READ: For X axis:
READ:  home_offset = 0.00
READ:  position_shift = 0.00
READ:  soft_endstop_min = 0.00
READ:  soft_endstop_max = 215.00
READ: > home_offset[X] = 0.00
READ:   current_position=(0.00, 42.00, 4.42) : 
READ: <<< set_axis_is_at_home(X)
READ:   current_position=(0.00, 42.00, 4.42) : sync_plan_position
READ:   current_position=(0.00, 42.00, 4.42) : > AFTER set_axis_is_at_home
READ: <<< homeaxis(X)
READ: >>> homeaxis(Y)
READ: Home 1 Fast:
READ: >>> do_homing_move(Y, -322.50, 0.00)
READ:   current_position=(0.00, 0.00, 4.42) : sync_plan_position
SENT: M105
READ: <<< do_homing_move(Y)
READ: Move Away:
READ: >>> do_homing_move(Y, 5.00, 0.00)
[... 181 lines cut ...]
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ:   current_position=(200.00, 42.00, -0.91) : sync_plan_position
READ:   current_position=(200.00, 42.00, -0.91) : <<< do_probe_move
READ: >>> do_blocking_move_to(200.00, 42.00, 4.09)
SENT: M105
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ:   current_position=(200.00, 42.00, 4.09) : set_probe_deployed
READ: deploy: 0
READ: >>> do_blocking_move_to(200.00, 42.00, 4.09)
READ: <<< do_blocking_move_to
READ:   current_position=(200.00, 42.00, 4.09) : > probing complete
READ:   current_position=(200.00, 42.00, 4.09) : G29 uncorrected XYZ
READ: Z from Probe:4.50  Matrix:4.15  Discrepancy:0.35
READ:   current_position=(200.00, 42.00, 4.50) : G29 corrected XYZ
READ:   current_position=(200.00, 42.00, 4.50) : clean_up_after_endstop_or_probe_move
READ: <<< G29
READ:   current_position=(200.00, 42.00, 4.50) : sync_plan_position
READ: X:200.00 Y:42.00 Z:4.50 E:0.00 Count X:20000 Y:4200 Z:1776
READ: ok

gloomyandy, 2018-04-01:
How big is your actual bed? In the first post you said it is 220x200 but in the configuration.h file it is set to 215x215. Also what are the probe offsets? In the configuration.h file you have them set to X -40 and Y -22 but in the above log it seems they are set to X -30 Y -22.  

Does your bed actually start at 0, 0? If not then this may be part of the reason why you are probing off of the bed. 

The above log shows the system trying to probe point (15, 20) and then correctly offsetting that point to be a nozzle position of (45, 42) to allow for the -30, -22 probe offset. But if your offsets are not correct and/or if the bed does not start at (0,0) then this point may not be on the bed. 

If your printer is such that the bed is not at (0,0) you may need to either adjust the endstops or change your configuration to correctly describe the position of the bed.

nkuck, 2018-04-01:
Sorry about the variations. I had been making a number of adjustments with the last being with the bugfix version so, maybe, not all the same parameters were carried over.
I don’t see were a probing point of (15,20) is showing up.
A G28 lands the nozzle in the center of the bed, and I assume that to be (0,0).
The bed is actually 220x220 and the probe offsets from the nozzle is x-40 y-22. I will do another run and upload the controls dump.
Thanks for your review and comments.

gloomyandy, 2018-04-01:
> I don’t see were a probing point of (15,20) is showing up.

Do a search for "probe_pt(15.00, 20.00" and you will see the debug for probing that point and how the probe offset is applied before performing the blocking move.

point (0, 0) is not (usually) in the centre of the bed for a Cartesian machine  it is usually at the front left location. You can easily find it by either using the LCD to move the print head to 0,0 or by issuing a G1 X0 Y0 command this will move the nozzle to (0, 0) is this location actually on your bed?

Looking at the debug when performing a G28 the probe not the nozzle should be in the centre of the bed. With safe home configured it will use bed width/2 and bed length/2 as the point to perform the Z home (so in your case 215/2, 215/2), but because you have a probe you can see it applies the probe offsets and the target location for the nozzle becomes (137, 129) (215/2 + 30 and 215/2 + 22 using the offsets in the debug). This should place the probe in the centre of the bed.

nkuck, 2018-04-01:
G1 X0 Y0 puts the probe way off to the front-left side of the bed. About 70MM to the left and 50MM in front. Shall we start there, first, Two correct my error(s)? Thanks.

gloomyandy, 2018-04-01:
What sort of printer do you have? A picture of it would be good.

nkuck, 2018-04-01:
It’s a Prusa clone from Zonestar. 
[image]

nkuck, 2018-04-01:
The position you see is X0 Y0

gloomyandy, 2018-04-01:
Right so the nozzle can move a long way from the bed. I think you have 3 choices....
1. Move the X and Y endstops so that (0, 0) actually falls on the bed. This may not be easy to do.
2. Work out how to tell Marlin that your movement range is larger than the bed size and the relative position of the bed within that range. 
3. Lie to Marlin and tell it that the bed is bigger than it is (pretend that it covers all of the area that you can move the nozzle to), but change the points that the levelling code uses to perform a probe such that they fall on the bed.

A good first step would be to map out what the movement limits are for your printer and to establish where in this envelope the bed actually sits. So you know (0, 0) is outside the bed. What is the x,y values if you move the nozzle so that it is just at the bottom left corner of the bed? How far can you move the nozzle in the X and Y direction (be careful, you don't want to go too far!). I assume that you can move it off of the other sides of the bed? If you can't do this then you may need to do some work to make sure all of the bed is reachable (otherwise your printable area will be smaller than it should be).

I think that option 2 is the best solution. But to be honest I'm not 100% sure how to configure Marlin for this type of setup. I can think of a couple of ways but it is probably better if someone else provides guidance on the best way to set things up. You might want to close this issue and open a new one asking how to configure Marlin for a bed smaller than the movement range.

ManuelMcLure, 2018-04-01:
That looks similar to what my Anet A8 has when it's at the X and Y endstop - that means that you need to configure `X_MIN_POS` and `Y_MIN_POS` so that X=0, Y=0 is at the lower left corner of the bed.

The easiest way to do this is to home X and Y (`G28 X Y`) and then use the LCD "Prepare" menu to move the printhead until the nozzle is exactly at the lower left corner of the bed. Then write down the X and Y coordinates that show on the LCD, and set the `X_MIN_POS` and `Y_MIN_POS` values in `Configuration.h` to the negative values of what is shown on the LCD. For example, if the LCD shows X=33 and Y=8, set `X_MIN_POS=-33` and `Y_MIN_POS=-8`.

As an example, I have the following settings for my Anet A8:
```
// @section machine

// The size of the print bed
#define X_BED_SIZE 220
#define Y_BED_SIZE 220

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -36
#define Y_MIN_POS -7
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 240
```

nkuck, 2018-04-01:
Thanks, guys. Will dive into these thoughts later tonight.

nkuck, 2018-04-01:
This is always confusing. You say to adjust X & Y to position the “nozzle” at the corner. Others have been saying “probe”. Which should it be?

Noel Kuck

Live simply so others can simply live!

nkuck, 2018-04-01:
I made a quick adjustment as you state, using the probe, and the nozzle was way to the back-right. Will try again with the nozzle to the front-left. Thanks.

Noel Kuck

Live simply so others can simply live!

ManuelMcLure, 2018-04-01:
The home position (0, 0) should always be set so the _nozzle_ is at the front-left corner of the bed. This may require `X_MIN_POS` and `Y_MIN_POS` to be set to a negative number. Imagine `X_MIN_POS` and `Y_MIN_POS` as "this is the position of the nozzle relative to the front-left when I'm touching both the X and Y endstops". Once these are set correctly, when you do a `G28 X Y` the _nozzle_ should end up at the front-left corner of the bed.

Once you have that set up, you use `X_PROBE_OFFSET_FROM_EXTRUDER` and `Y_PROBE_OFFSET_FROM_EXTRUDER` to specify where the probe is in relation to the nozzle. If those are set correctly when you do a `G28` with `Z_SAFE_HOMING` the _probe_ should end up in the center of the bed.

nkuck, 2018-04-01:
Ok. Thanks.

nkuck, 2018-04-02:
Followed your suggestions.First did a G1 X0 Y0, then jogged the nozzle to the front-left bed corner, and had X+37 Y-54 which I entered into the X_MIN_POS and Y_MIN_POS. When I tried to flash it, I received the following error message:

In file included from sketch/MarlinConfig.h:39:0,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
SanityCheck.h:323: error: static assertion failed: Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.
 static_assert(X_MAX_LENGTH >= X_BED_SIZE && Y_MAX_LENGTH >= Y_BED_SIZE,
 ^
exit status 1
static assertion failed: Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.

Any thoughts? This is my first, prolonged attempt at getting ABL running on my printer which. otherwise, performs well.
Thanks for any further suggestions. My Configuration.h file is attached.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1869145/Configuration.h.zip)

Roxy-3D, 2018-04-02:
You can either just specify your bed size and let it calculate the Min and Max X & Y.   Or you can specify the Min and Max and let it calculate the bed size.   But you can't do both...    It might make sense to return that part of configuration.h back to its original state and do that part over again.

nkuck, 2018-04-02:
Ok, so will 0 the XY-MIN-POS and try again?

nkuck, 2018-04-02:
No luck with:
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 215
Just probes off the bed to left and ends at X175 Y69 above the bed.

gloomyandy, 2018-04-02:
Those X and Y figures seem a little strange. Given your previous description I would have expected that when the nozzle was over the left front corner  both would have +ve value. I thought you said that (0, 0) was to the left and in front of the front edge of the bed?

If to get the nozzle over the left front corner you need to move it to Y == -54 then this would mean that (0, 0) is off to the left of the bed but 54mm from the front edge of the bed towards the rear of the machine (so actually over the bed). That seems a rather odd setup for a machine that can move the nozzle to the  front/left point of the bed. Perhaps you could post a picture which clearly shows the position of the nozzle in relation to the bed when at (0, 0). Just to be clear, you need to perform a G28 before doing any of these operations.

@Roxy-3D How do you specify a machine that has a movement range that is greater than the size of the bed? It would seem that to do that you would need to specify both the bed size and the Min/Max X/Y?

ManuelMcLure, 2018-04-02:
> First did a G1 X0 Y0, then jogged the nozzle to the front-left bed corner, and had X+37 Y-54 which I entered into the X_MIN_POS and Y_MIN_POS.

Are you sure it was X+ and Y-? from the photo above it looks to me like both should have resulted in positive values (which you would enter as negative in `X_MIN_POS` and `Y_MIN_POS`. Your printer looks very much like a clone of my Anet A8 and on my A8 both `X_MIN_POS` and `Y_MIN_POS` are negative.

Unless... wait a sec. When you jogged Y, did the bed move or did the extruder move? If the extruder moved you have the X and Y axes swapped and that would explain why our suggestions have not been helping.

ManuelMcLure, 2018-04-02:
In any case, it makes sense to make sure that the motion is as expected. When jogging, if you increase X the extruder should move to the right, and if you increase Y the bed should move towards the front of the printer. If either of these are incorrect we need to fix that first.

nkuck, 2018-04-02:
Jogging Y moves the bed.
Here is after Home All (Image)
Here is after X0 Y0 (Image 1)

Roxy-3D, 2018-04-02:
One answer would be to specify the full movement area as the bed size.   And then specify each of the 4 insets to be appropriate to keep the nozzle on the bed.    I do that on my gMax.   Its bed is larger than the 400x400mm heat pad and glass I have.   And it homes to a corner that is off the glass.

nkuck, 2018-04-02:
Not sure that I understand how to implement what you are suggesting.

Roxy-3D, 2018-04-02:
You can't do that with 3-Point leveling.   You will need to use a mesh based system to have insets available.

gloomyandy, 2018-04-02:
@nkuck the images that you tried to post do not seem to work (at least not for me).

nkuck, 2018-04-02:
Here are the photos again:
Home All
G1 X0 Y0

Noel Kuck

Live simply so others can simply live!

gloomyandy, 2018-04-02:
I don't think you can post images to github by including them in email replies. I think you either need to post a link or probably better still go to github and post the images there.

nkuck, 2018-04-02:
OK, I followed this tutorial and am as close as I have ever been to getting G29 to do what I expected it to do in attempting the 3-point leveling. Maybe a little fine tuning.
https://github.com/JimBrown/MarlinTarantula/wiki/How-to-center-your-prints-on-the-bed

nkuck, 2018-04-02:
here they are

ManuelMcLure, 2018-04-02:
I think that tutorial assumes that the endstops will trigger at (0, 0) which isn't the case with your printer (or mine). Although it might get you to something that kind of works, I'd be wary of future issues because Marlin isn't understanding exactly what your printer is like.

nkuck, 2018-04-03:
Thanks for the insight. I will keep that in mind.

thinkyhead, 2018-04-03:
> here they are

Where? If you still want to share the images, paste them into a reply at https://github.com/MarlinFirmware/Marlin/issues/10267

nkuck, 2018-04-03:
[image]
[image]

nkuck, 2018-04-03:
> Hmmm... when I instructed you earlier on how to find the X_MIN_POS and Y_MIN_POS values by homing X and Y and then jogging until the nozzle was over the front-left corner, I forgot to mention that you needed to make sure X_MIN_POS and Y_MIN_POS were both 0 before starting that procedure. If they were set to something different from an earlier attempt the results would be incorrect. So the procedure is:
>
> Set X_MIN_POS and Y_MIN_POS to 0 and update the firmware.
> Home X and Y with G28 X Y
> The LCD should now show X=0 and Y=0.
> Jog the X and Y using the LCD until the nozzle is just over the front-left corner of the bed.
> Write down the X and Y numbers shown on the LCD.
> Use the negative value of whatever X and Y are to set X_MIN_POS and Y_MIN_POS. So if the LCD displays X=34 and Y=8, use X_MIN_POS=-34 and Y_MIN_POS=-8.
> Update your firmware again with the new settings.
> You could also get at least a close approximation by using a ruler and square to measure the X and Y offset of the nozzle from the front-left corner of the bed when the printer is in the state shown by your second picture. With that position for the nozzle you should use negative values for both X_MIN_POS and Y_MIN_POS.

Implemented your suggestions. G28 positioned nozzle to X135 Y162. G29 looked ok for probing X along the front of the bed, but the center/back spot was just off of the bed. I’m sure that there is another setting that I need to change. Thoughts? Thanks for the guidance.

nkuck, 2018-04-03:
And doing a HOME-ALL positions the probe close to the center of the bed with the nozzle per my offset entries (behind it and to the right - further from bed center). Is that the way it is supposed to be??

ManuelMcLure, 2018-04-03:
> And doing a HOME-ALL positions the probe close to the center of the bed with the nozzle per my offset entries (behind it and to the right - further from bed center). Is that the way it is supposed to be??

With `Z_SAFE_HOMING` set, the probe should end up spot-on in the center of the bed, which with your settings of:

```
// The size of the print bed
#define X_BED_SIZE 215
#define Y_BED_SIZE 215
...
#define X_PROBE_OFFSET_FROM_EXTRUDER -25  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -50  // Y offset: -front +behind [the nozzle]
```

should mean that the LCD shows X132 and Y157. So your X135 Y162 is very close to correct.

> G29 looked ok for probing X along the front of the bed, but the center/back spot was just off of the bed. I’m sure that there is another setting that I need to change.

For the probe locations, I haven't used 3-point leveling so I don't know how the automatic selection of leveling points works and whether it's been updated to use MIN_PROBE_EDGE and such. However, you can try setting manual points by setting the following:

```
#if ENABLED(AUTO_BED_LEVELING_3POINT) || ENABLED(AUTO_BED_LEVELING_UBL)
  //#define PROBE_PT_1_X 15
  //#define PROBE_PT_1_Y 180
  //#define PROBE_PT_2_X 15
  //#define PROBE_PT_2_Y 20
  //#define PROBE_PT_3_X 170
  //#define PROBE_PT_3_Y 20
#endif
```

Just make sure that the probe locations are reachable by the _probe_. Also, with an inductive or capacitive probe it's a good idea to not get too close to the edge of the bed - I'd make sure that the position is 20mm in from the edge of the bed.

With your setup I'd probably use something like

```
#if ENABLED(AUTO_BED_LEVELING_3POINT) || ENABLED(AUTO_BED_LEVELING_UBL)
  #define PROBE_PT_1_X 110
  #define PROBE_PT_1_Y 165
  #define PROBE_PT_2_X 20
  #define PROBE_PT_2_Y 20
  #define PROBE_PT_3_X 190
  #define PROBE_PT_3_Y 20
#endif
```

nkuck, 2018-04-03:
Excellent support. Thanks!

Noel Kuck

Live simply so others can simply live!

closed 2018-04-09 by thinkyhead