MarlinFirmware/Marlin #10268: Axis inverted after G29

gucio1200, 2018-03-31:
### Description

After G29 complited, X/Y axis are inverted doing G28 , when try to move by G1 moves ok to correct cordinates.

### Steps to Reproduce

1. G28
2. M915 S230
3. G29 D
4. G29 P1
5. G29 P3.1 R
6. G29 S0
7. G29 A
8. M500
9. M501
10. G28 - Goes wrong direction and hiting end of axis.

[marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/1865924/marlin-config.zip)

thinkyhead, 2018-03-31:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

gucio1200, 2018-04-01:
Yeap, seems to be fixed in last release.
Thank You!

thinkyhead, 2018-04-02:
Switch to `AUTO_BED_LEVELING_BILINEAR` and see if the issue still occurs. Then, please try disabling various optional features one at a time to see if any of them causes the issue to go away. Some options to try disabling…
- `EEPROM_SETTINGS`
- `NEOPIXEL_LED`
- `NOZZLE_CLEAN_FEATURE`
- `G26_MESH_VALIDATION`
- `RESTORE_LEVELING_AFTER_G28`
- `LIN_ADVANCE`
- `MINIMUM_STEPPER_PULSE` (disable or set to 2)
- `HAVE_TMC2130`
- `MONITOR_DRIVER_STATUS`

---
Also:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. (This will also enable bed leveling.)
- Do another `G28` to expose the axis reversal issue.
- Copy the log output into a **.TXT file** and attach it to your next reply.

gucio1200, 2018-07-15:
I have found reason of issue - that was octoprint - which during update put some extra lines of code to macros. I realised that when i saw octoprint config file. So that was not related to Marlin.

closed 2018-07-15 by gucio1200