MarlinFirmware/Marlin #10269: [2.0] MKS SBASE TMC2130 - MONITOR_DRIVER_STATUS kills serial connections

maccadelic, 2018-04-01:
I have 4 fystec TMC2130 drivers running on my MKS SBASE with SPI. Everything is fully fuctioning. I can run large prints without an issue all day.

As soon as i define MONITOR_DRIVER_STATUS in configuration_adv, the USB connection completly fails to work both in control software (tried repiter and octoprint), and also stops showing up as a USB device in windows. 

If i comment it back out, and flash from SD card, everything works fine again. My M122 output is as follows:

```
Send: M122
Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	500	500	600	500
Recv: RMS current	489	489	581	489
Recv: MAX current	689.490000	689.490000	819.210000	689.490000
Recv: Run current	15/31	15/31	18/31	15/31
Recv: Hold current	7/31	7/31	9/31	7/31
Recv: CS actual		7/31	7/31	8/31	7/31
Recv: PWM scale	24	21	20	25
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	8	3	3
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x       8       0:       0       7:       0       0:       0       0
Recv: 	Y = 0x       8       0:       0       7:       0       0:       0       0
Recv: 	Z = 0x       8       0:       0       9:       0       0:       0       0
Recv: 	E0 = 0x       8       0:       0       7:       0       0:       0       0
```

[Marlincofigs.zip](https://github.com/MarlinFirmware/Marlin/files/1866016/Marlincofigs.zip)

thinkyhead, 2018-04-02:
See if any of these issues gives any clues:

https://github.com/MarlinFirmware/Marlin/search?q=MONITOR_DRIVER_STATUS&type=Issues

Particularly, #9796

maccadelic, 2018-04-03:
Sorry, may bad, i thought it might be a bug, but was the servo pin set to the same as MSIO. For some reason that article didn't show up in my search. Thanks, ill try and be more thorough before i post again.

LemarinelNet, 2018-05-03:
Perhaps this one could be closed now?

closed 2018-05-04 by thinkyhead