MarlinFirmware/Marlin #10270: Can`t compil with EXTRUDER_AUTO_FAN_SPEED not equal 255

tayalex, 2018-04-01:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

tayalex, 2018-04-01:
1. Board RURAMPS4D
2. E0_AUTO_FAN_PIN 8
3. EXTRUDER_AUTO_FAN_SPEED 127

thinkyhead, 2018-04-02:
@Bob-the-Kuhn — Is the `_GET_TIMER` macro for Due simply incorrect or not yet implemented? It currently does nothing.
```cpp
/// check if pin is a timer
#define _GET_TIMER(IO)
```

numanair, 2018-06-19:
I can reproduce the issue with the malyan m200 board.

thinkyhead, 2018-06-20:
@numanair — This seems to compile fine.
I just commented out `E0_AUTO_FAN_PIN` so it uses pin `PB8`.
```cpp
//#define E0_AUTO_FAN_PIN -1
```

boelle, 2019-02-20:
@tayalex was problem solved ???

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead