MarlinFirmware/Marlin #10272: LIN_ADVANCE 1.5 -- extruder rattles *a lot*.

Artem-B, 2018-04-01:
### Description

I'm trying to enable LIN_ADVANCE on bugfix-2.0.x branch on my CR-10s + Cohesion 3d remix board (smoothieboard variant).
As soon as LIN_ADVANCE is enabled extruder starts rattling. X and Y seem to start jerking, too, but it's hard to tell with all the noise the extruder motor is making.

I've checked step/dir signals on the extruder and it confirmed that it frequently changes motor direction 
* ~20-70 ms worth of forward progress with about evenly spaced pulses.
* ~10-50 ms retraction
* repeat

Rattling happens regardless of what I print. Even when printing LIN_ADVANCE calibration pattern extruder starts rattling as soon as LIN_ADVANCE is enabled.

Despite the noise the printer *is* printing OK. At least, I don't see anything obviously wrong with the print, but the noise is rather bad. It's *way* worse when I had LIN_ADVANCE 1.0 working on the same printer. With 1.0 extruder was making 'tat-tat-tat-tat' noises. Now it's 'drrrrrrrrrrrr'.

### Steps to Reproduce

1. bugfix-2.0.x branch (as of March 30th)
2. start printing
3. M900 K0.6 (best value on calibration test print for my CR-10s with ~400mm long bowden)
4. enjoy the noise!

**Expected behavior:** [What you expect to happen]
LIN_ADVANCE 1.5 is supposed to `ensure a smooth extruder movement without rattling`

**Actual behavior:** [What actually happens]
Instead, the smooth movement with no retractions turns into constant back-and-forth.

#### Additional Information
Here is the album with the videos and oscilloscope shots demonstrating the problem:
https://photos.app.goo.gl/l8hzKHgBgaR0dw5I2
This is what is the extruder step(yellow)/dir(blue) signals look with LIN_ADVANCE disabled. 
[image] 

Here's what I get with LIN_ADVANCE enabled:
[image]

Attached config files:
[Marlin-configs.zip](https://github.com/MarlinFirmware/Marlin/files/1866550/Marlin-configs.zip)

thinkyhead, 2018-04-02:
Do you see any improvement with `MINIMUM_STEPPER_PULSE` set to 2?

Artem-B, 2018-04-02:
Tried that -- no difference. I've verified that with default settings pulses are ~2us wide (and pretty clean signal-wise) regardless of LIN_ADVANCE and that's way more than my steppers (TMC2224) need (100ns min step time). I've also verified that there's at least 1us delay between changing DIR and the next step. AFAICT, the timings are rather conservative for the driver.

As far as I can tell, the motor does obey the frequent direction changes and that's what makes the rattling noise. I mostly wonder whether such behavior is normal for LIN_ADVANCE or not.

Another things that looks suspicious to me is the high step rate with LIN_ADVANCE enabled. It looks like it tries to move filament at ~ 5mm/s with no visible acceleration/deceleration phase. Interestingly enough, jerk is set to 5 mm/s, so that may be the setting to play with.

Sebastianv650, 2018-04-02:
> It looks like it tries to move filament at ~ 5mm/s with no visible acceleration/deceleration phase. Interestingly enough, jerk is set to 5 mm/s, so that may be the setting to play with.

That's exactly what's happening. LA uses the extruder jerk value to limit the max. possible extruder speed. If this speed would be exceeded due to a high K value or high acceleration setting, the print acceleration will be reduced to stay withing the given ejerk value. During each acceleration / deceleration phase, the extruder speed is offseted by the calculated LA speed.

In your video, it looks like the extruder is running smooth but the noise might be missed steps. So the first thing to check would be if your extruder can actualy handle an extruder jerk setting of 5 (without LA). I know this isn't straight forward, feel free to use creative ways to do that ;)

Even if you get it to work, be prepared that you would need a quite high ejerk setting to use K values as they are on such a bowden setup without slowing down the acceleration. And even if you can do that, most reports I have so far are showing that the print results are bad anyway due to other bowden effects. 
Small example: When you print the calibration pattern, the line width of the fast section should recover to it's nominal width after some cm. In the stated reports, the line width just stays thin until the slow section starts again. So the pressure which is built up by the extruder isn't reaching the nozzle at all.

I tend to call pressure compensation not working for bowden printers and likely I will add a note regarding that in the documentation if there is no real positive feedback in the near future.
There is no way to compensate for pressure effects in software as long as the bowden tube is eating up all the changes the extruder does.

thinkyhead, 2018-04-03:
@Sebastianv650 — Sounds like these would be good notes to add to the Linear Advance page "troubleshooting" section.

Artem-B, 2018-04-03:
As I mentioned, the print results in my case are actually pretty good, though I'm not sure yet whether it's due to LIN_ADVANCE or to the difference in hotend movements. On the test cube I normally see a very minor decaying wave pattern along Y axis, but it is gone when LIN_ADVANCE is on. I also see moderate improvement along the edges of X and Y letters on the cube, so I'm not ready to give up on LIN_ADVANCE yet. :-) I'll tinker a bit with jerk/acceleration/max extruder speed and see whether I can get it to be silent.

Artem-B, 2018-04-05:
Tinkering with E jerk helped a little, when I cranked the jerk down to almost zero. Alas, X/Y also slowed down to a crawl, so it was not very useful. I did verify that my extruder can handle pretty high jerk, acceleration and speed at least when it does not change direction. Apparently it's the direction change that seems to mess with the driver. I may try running it exclusively in spreadcycle mode, though it does come with it's own annoying high-pitch whine. 

> What extruder do you have on your CR10?
It's stock. Hmm. Could that be that it's not the missing steps, but the *filament* skipping on the cogwheel? I should try cranking up the pressure and see if that helps.

Sebastianv650, 2018-04-05:
> I did verify that my extruder can handle pretty high jerk, acceleration and speed at least when it does not change direction. Apparently it's the direction change that seems to mess with the driver.

How do you test jerk without changing direction?

Itox001, 2018-04-17:
HUGE SUCCESS! I just got around to trying linear advance 1.5 on my heavily modded tarantula. Got around 400 mm of capricorn bowden, no geared extruder. K=0.65 looked good with the pattern generator. Just printed a test cube and it looks pretty good! 

I used to print everything at the same speed (infill, perimeters, external perimeters, etc) because I noticed that after printing fast inner perimeters, the start of the slow external perimeter would be overextruded because of residual pressure and leave a protuded line, which looked like banding depending on seam placement. After linear advance, this effect is almost gone. Attached a pic of a cube with 35 mm/s inner 20 mm/s outer perimeters, without and with linear advance enabled, respectively. All other settings are the same.
[image]

I'm not sure what the line above the X with linear advance enabled is, but at least the problem I was solving is gone.

Con is, print took 27% longer due to acceleration reduction. Current Ejerk is at 5, but I believe it can be higher, since I haven't heard a single click yet. Advice on how to calibrate Ejerk is appreciated.

Sebastianv650, 2018-04-17:
Great to hear it's working for you! Especialy as it shows that LA is able to work with bowden systems at least to some degree.

> I'm not sure what the line above the X with linear advance enabled is, but at least the problem I was solving is gone.

I would guess it has something todo with the bridging above the "X" tip, where the slicer will most likely increase the flow to some bridging flow, on the other hand the backpressure from last layer is no longer there for this short segment. One or a combination of this two effects might lead to the slight over or underextrusion you get between the tip of the X.

> Advice on how to calibrate Ejerk is appreciated.

I don't know of a "perfect" procedure to adjust jerk, and E axis ist most likely the most difficult one. Usualy, a skipped E step should be quite visible as a stepper is allways skipping a full step so it leads to a quite hughe underextrusion.
Maybe this idea is useful: Create a gcode which prints a single wall cube without any infill. You might strech it along one direction, for example it could be 60x10mm. Along the long side which is facing to you on the print bed, devide the G1 moves in 3 segments:
- A slow section, maybe 20mm/s for 15mm followed by
- a fast section, maybe 40mm/s 30mm followed by
- another 20mm/s 15mm long section.

Using your prefered K factor, increase the E jerk after a few mm on Z, like ejerk = 5 for Z=0 to 3mm, ejerk 10 for 3-6mm. Most likely on the slow-fast transition you will hear and see a skipped jerk when it happens. Turn back the jerk to a value you feel save with and do a test print.

Itox001, 2018-04-18:
Spoke too soon probably. After prinintg that cube I printed benchy and it had two heavily undrextruded layers. Then printed it again to check if it was a random clog or something but the lines are in the same exact spot. Then printed the same gcode but with linear advance off and the lines were gone. So linear advance causing weird underextrusion? Pic attached, left if k=0, both on the right are k=0.65:
[image]

VanessaE, 2018-04-18:
@Itox001   I have similar problems.  I have a benchy here that looks to have the same extrusion problem, in the same place, just a little less pronounced than yours:

[image]

(lighting is deliberately positioned to accentuate the shadows between layers, making it look quite bad, but in reality it actually looks decent, best one I've ever managed to print; don't mind the dark spots left of center, that's just ink on the print, and crud on my camera lens)

If you open the image in another tab and zoom to 100%, you can see similar problems around the starboard doorway, especially to the left in the image (those are the result of Slic3r not being consistent about where it places seams, besides my being unable to tune-out the pressure difference).

Here's my latest attempt at the calibration pattern, after discovering and fixing a problem with my hotend:

[image]

The lines for K=0.11 to 0.13 (cropped to highlight the transition points):
[image]

The higher values (about K=0.29 to 0.39, cropped to highlight transitions)
[image]

(Please excuse the "bent" look, I didn't take time to apply any lens distortion correction)

As you can see, the K value that results in perfect slow->fast transitions (K=0.12), is nowhere near correct for fast->slow transitions, slow being 10 mm/s and fast being 100 mm/s, with 500 mm/s² acceleration.  All of the fast->slow transitions have blobs, up to about K=0.35, but as you can see, well before that value, the ends of the fast segments begin to starve, while at the same time, the starts of the adjacent slow segments have blobs.

As you can also see, the distance it takes for pressure to actually build up or bleed off just doesn't match what the math is trying to do.  I think there needs to be a way to make the pressure corrections happen faster, independent of the user's normal acceleration settings (what looks good for normal purposes such as ringing may be far too slow for pressure management).

@thinkyhead is this possibly related to that jerk calculation problem discussed elsewhere?

<details><summary>The gcode that produced that test pattern</summary>
```gcode
; ### Marlin K-Factor Calibration Pattern ###
; -------------------------------------------
;
; Created: Fri Apr 06 2018 08:34:55 GMT-0400 (EDT)
;
; Settings Printer:
; Filament Diameter = 1.72 mm
; Nozzle Diameter = 0.4 mm
; Nozzle Temperature = 200 °C
; Bed Temperature = 70 °C
; Retraction Distance = 2.5 mm
; Layer Height = 0.2 mm
; Z-axis Offset = 0 mm
;
; Settings Print Bed:
; Bed Shape = Rect
; Bed Size X = 200 mm
; Bed Size Y = 200 mm
; Origin Bed Center = false
;
; Settings Speed:
; Slow Printing Speed = 600 mm/min
; Fast Printing Speed = 6000 mm/min
; Movement Speed = 7200 mm/min
; Retract Speed = 1800 mm/min
; Printing Acceleration = 500 mm/s^2
; Jerk X-axis =  firmware default
; Jerk Y-axis =  firmware default
; Jerk Z-axis =  firmware default
; Jerk Extruder =  firmware default
;
; Settings Pattern:
; Linear Advance Version = 1.5
; Starting Value Factor = 0
; Ending Value Factor = 0.4
; Factor Stepping = 0.01
; Test Line Spacing = 4 mm
; Test Line Length Slow = 50 mm
; Test Line Length Fast = 50 mm
; Print Pattern = Standard
; Print Frame = true
; Number Lines = true
[... 968 lines cut ...]
G1 X185 Y159.5 F7200 ; move to start
G1 X187 Y159.5 E0.0826 F600 ; 8
G1 X187 Y157.5 E0.0826 F600 ; 8
G1 X185 Y157.5 E0.0826 F600 ; 8
G1 X185 Y159.5 E0.0826 F600 ; 8
G1 X185 Y161.5 E0.0826 F600 ; 8
G1 X187 Y161.5 E0.0826 F600 ; 8
G1 X187 Y159.5 E0.0826 F600 ; 8
G10 ; retract
G1 Z0.3 F600 ; zHop
;
; finish
;
M104 S0 ; turn off hotend
M140 S0 ; turn off bed
G1 Z30 X200 Y200 F7200 ; move away from the print
M84 ; disable motors
M502 ; resets parameters from ROM
M501 ; resets parameters from EEPROM
;
```
</details>

` `
> (All items printed using my usual MK4 direct-feed geared-drive extruder, pushing Atomic bright white PLA into a genuine J-head v8 at 200°C, with 0.4 mm nozzle; 0.2 mm layers; calibration pattern is standard mode, all lines printed left-to-right, smallest K values at the bottom/front of the plate).

VanessaE, 2018-04-19:
I wonder if the rate at which the high->low correction is made is really the issue?  I mean, if you look at those lines, it's almost as though the correction is applied properly, but then is immediately canceled out *after* the fast->slow junction?

Assuming for the moment that it isn't just some characteristic of the hotend, what else could cause the blobbing seen here?

Itox001, 2018-04-19:
Okay, so here is the benchy. It seems to me like it happens at layers 30 and 39 approximately. At 0.2 layer height, that would make it at 6 mm and 7.8 mm. I hope you can find something useful!
[image]

Gcode attached.
[benchyGcodeLinAdv.zip](https://github.com/MarlinFirmware/Marlin/files/1930109/benchyGcodeLinAdv.zip)

thinkyhead, 2018-04-20:
> is this possibly related to that jerk calculation problem discussed elsewhere?

That issue was causing pauses in the lines, so that should be considered if they're supposed to be continuous.

(Another issue was also causing pauses, but that was specifically caused by using `AUTO_POWER_CONTROL` with Trinamic steppers.)

Itox001, 2018-04-20:
Yes, actually layer 30 is when the bottom layer of the box behind the cabin is first laid down, and layer 40 is when the bottom layer of the floor around the box is first laid down. So, that plays a part somehow.

Sebastianv650, 2018-04-21:
Exactly, on the other layers perimeter starts after infill (55mm/s). At the two issue layers, the perimeter starts after "solid infill" with 35mm/s which is the same speed as the perimeter. So somehow the pressure isn't perfect in this situation and it also only slowly recovers.

> I mean, if you look at those lines, it's almost as though the correction is applied properly, but then is immediately canceled out after the fast->slow junction?

I would expect the blobs happen at the start of deceleration, not at its end (=the point where the slow section starts). But I can't say that fure sure with the picture.

> Assuming for the moment that it isn't just some characteristic of the hotend, what else could cause the blobbing seen here?

I can't think of any other reasons. All the line width = pressure differences we see in our prints are due to effects inside the bowden (or at elast the short section between hobbed bolt and heat brake) and in the molten section between heat brake and nozzle. The molten section is nearly out of control for the FW and given as a manufacturing quality thing from the hot end. I'm wondering why there is not more research on this side from, for example, e3d side. I guess there is quite some potential by using more advanced inside nozzle geometries, surface roughnesses or even platings.

VanessaE, 2018-04-23:
I think this issue needs to be re-opened, as I also have this rattling problem (I've only just noticed it with my bot).

ManuelMcLure, 2018-04-24:
I saw the "waterline" line appear on my Benchys (even worse than what I see in your pictures) when I switched from Cura to Slic3r PE, although the rest of the Benchy looked much better than the Cura-sliced one. Watching carefully I found that Slic3r was turning off the part cooling fan completely when it started printing the "deck" level and that was causing that layer to ooze over the edge of the layer underneath. Once I told Slic3r to never go under 50% fan (except for the first layer) the "waterline" disappeared. So this may not be LA related but rather cooling related.

VanessaE, 2018-04-26:
Something of note:  I just realized that extruder rattle I mentioned is quite negligible during normal linear-advance-affected print moves, but once the printer gets to some part of the model with lots of retractions between very short extrude moves, such as short gap fill/crowning lines followed by retract-travel-unretract followed by more gap fill somewhere else (and repeat a few times), the retractions get noisy, like the extruder's being jammed up to full retract speed (usually between 20 and 40 mm/s) without jerk (5), whether with firmware retract or normal slicer-initiated retract.

@thinkyhead does that give you something to go on?

(moved from #10446, sorry for the confusion :stuck_out_tongue:  )

VanessaE, 2018-04-27:
I think I've spotted something.  Set E jerk and accel really low, say 1 and 50 respectively, retraction length as high as your hotend can safely tolerate (just so you have time to watch it), and print something with lots of retracts (such as https://www.thingiverse.com/thing:1363023).  Print it at 30 mm/s, 1000 mm/s X/Y accel, with a retraction speed of 40 mm/s.

Note your extruder movement, particularly when the gap fill bits are bring printed around the three holes and the "ctrlV" in the part.  If you watch close, you can see this speed pattern:  the extruder is going forward for normal print moves, runs backward at jerk speed to start a retract, accelerates to and runs at retract speed, then stops instantly, without deceleration, when it gets to the end of the 1 mm retract move.

Here's a short clip that should illustrate it:  https://www.youtube.com/watch?v=g6O-PD106zI&feature=youtu.be (maybe play it at quarter-speed)

(I'm experimenting with KISSlicer's PreloadVE feature, with linear advance set to 0, so that slower movement at the beginning of a retract might be the PVE feature, not that it matters.)

Sebastianv650, 2018-04-29:
@VanessaE  I'm very sure what you have described here regarding retractions without acceleration is due to the faulty jerk handling we recognised some time ago. I was reading thinkyhead started porting the GRBL way of handling it, I guess this might be solved when he finished that.

@comps I think you want to check your acceleration if it's realy set to 10000mm/s². I'm also using a geared extruder with printed gears (without glue for the screw), never came loose up to now.

@Lenbok I'm aware of that, I just never noticed the need for such a feature up to now. Some kind of handling for deceleration is on my list, I'm just short on time at the moment. But it will come.

VanessaE, 2018-04-29:
@Sebastianv650 yeah, thought so.  But you know me, I forget stuff, and I have a bad habit of being kinda verbose. :smiley:

boelle, 2019-02-20:
@Artem-B still having issues when using latest bugfix 2.0?

Artem-B, 2019-02-20:
@boelle  I can no longer tell as I've switched to Klipper.

boelle, 2019-02-20:
then maybe click the green close button below?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

Snagi0, 2019-09-03:
I have the same problem with current bugfix 2.0.x. I think this should be reopened. Here is my comment on the other thread. If you need more info let me know. Btw I am using 24V to drive the steppers.
> You can also try to enable junction deviation, instead of using jerk settings. That also seemed to have a positive effect on my setup. As stated earlier, enable SQUARE_WAVE_STEPPING and set MINIMUM_STEPPER_PULSE to 1 or 2.
> 
> I have a TMC2208 with UART as an extruder on bowden system. I have above settings enabled and running only in STEALTHCHOP. There is no problems if I have linear advance disabled. If I enable linear advance, I can see a clear benefit on print quality, but the extruder stepper starts to make rattling noises when it compensates for the pressure. There is no skipping or any mechanical problems. I can see that the extruder is running in both directions very rapidly during the rampup or rampdown phase when controlling the pressure. Those quick jerk jumps should be integrated to the normal extruder movements instead of handled separately. There has been multiple bug reports with TMC drivers and linear advance, but all of them has been closed without fixing.
> 
> To sum it up: **TMC2208 with UART control as extruder on bowden system running only stealthchop. Works very well and silent without linear advance. Prints even better with linear advance, but there is terrible noise caused by the control scheme used with linear advance.** This rattle mechanical vibrations prevent the use of linear advance for me.
> 
> We really **need linear advance v.2** to correct this problem (and I guess s-curve handling with lin adv also?).
> I'm using **bugfix2.0.x (28.8.2019)**.
> 
> Here is an example of the other bugs. Artem-B had the same problem and scoped his step and dir channels where you can see rapid direction changes in 10ms range. There is no point on having that kind of control, instead the main movement direction should be slowed down.
> [#10272 (comment)](https://github.com/MarlinFirmware/Marlin/issues/10272#issue-310340937)

Sonillion, 2019-09-22:
Hello, I can confirm that I also have same problem with TMC 2209 (board SKR mini E3) on my ender 3.

Snagi0, 2019-09-24:
A little update. 

Earlier I tried to affect the rattling with lin.adv. enabled using different and very extreme stepper settings. With some long delays there was a clear difference (less or no rattle), but at that point the settings were otherwise unusable and had major printing issues. So it seems that there is no "magic" stepper setting combination to fix this issue.

I also played with different jerk and junction deviation settings, but those didn't seem to have much effect. 

Right now I have reverted to quite low junction deviation value and recommended stepper settings with MINIMUM_STEPPER_PULSE set to 1 (also SQUARE_WAVE_STEPPING & ADAPTIVE_STEP_SMOOTHING etc enabled).

However I tried to increase the **microstepping from 16 with interpolation to direct 256**, and that **helped a bit with the rattle**. When the printer is doing one axis extrudes the problem is basically gone, but on circular extrudes the rattle is still there. Maybe a bit decreased thou.

If I'd need to guess, I think that the control is trying to move smaller moves than one microstep and then realizes that we need to back up a bit etc etc causing the oscillation (rattle). Or something similar. Smaller moves with higher microsteps would then help a bit. However there seems to be no any filtering or smoothing to the control (which would probably help as there is no benefit or need to do so fast moves). This might also be the cause why there is no problems with some systems (if the drivers can't keep up or MCU is not fast enough to cause problems).  

**It seems that nobody is working on this, which is a shame. If you are, please let me know. To me this is also a bit weird, as the problem is a major issue on core Marlin function and the root cause seems to be directly in the function implementation.** There may not be yet very many affected users as the overall HW used by the community may be a bit on the older side, but this is fast changing.

Btw, I may not have mentioned, I am using LPC1768 32bit ReArm board.

Sebastianv650, 2019-09-25:
@Snagi0 

> It seems that nobody is working on this, which is a shame. If you are, please let me know. To me this is also a bit weird, as the problem is a major issue on core Marlin function and the root cause [...]

Marlin only sees progress if people like you are seeing a problem which isn't worked on - and start solving them on their own. That's exactly how I came to develop LA in the first place. My hope was that a more skilled person would take over the development, but that rarely happens ;)
I can't support LA in the way you likely want until my end of life, it needs a lot of time especially as Marlin itself is changing every day. I did LA as I found a problem and needed a solution, not because I was in search for a new life time hobby. Sadly, the day only has 24h. Even worse is the fact that I can't reproduce the problem as my printer is still the same as when I started the first lines of code, which means a classic 8 bit board with a direct drive extruder.

So in short: I'm sorry, but especially with 32bit related problems I can't help.

Snagi0, 2019-09-25:
Thank you for the response, info and comments. I totally understand. 

I also have many other projects of my own, those are the reason I have the 3D printer in the first place. Also I think that my coding might not ether be quite at that level, to do the necessary work, at least without a major time investment, that I really cannot afford. 

However it seems that this thing might be getting some recognition slowly. I hope that someone picks up the work. The main reason in my view is that there has not been a clear enough signal that updates and fixing is really needed. Maybe users with newer HW and all bells and whistles enabled in Marlin are still a bit rare. The fact that some people have reported the same thing but all comments/bugs are scattered in multiple places and maybe closed due to lack of interest one by one is not helping. And some of the effected have moved to other firmwares etc.

Would it be possible to get some kind of overview/info/details or something about the LA implementation that might help someone to pick up the torch?

Is there anything else that could be done to help with improvements?

Here is pics of 1 layer height (0,10 mm), 3 perimeters (0,36 mm line width) brim that has a visual effect of the back and forth extruder movement during the curved lines. I'm not quite sure how typical this example is, as my settings are not 100% tuned yet, but I think that the effect is at least mainly (if not completely) caused by LA.

[image]
[image]

boelle, 2019-10-24:
@Snagi0 still an issue?

Snagi0, 2019-10-24:
Has there been any commits relating to this? I have not seen any and I'm still using the same build from a month ago.

So the issue is still there. For me, right now, the jerkyness of the filament output is the main problem and has a negative effect on some aspects of the print, especially on bridging.

boelle, 2019-10-24:
> Has there been any commits relating to this?

have you checked? 

> So the issue is still there.

not for me and i do use lin adv

Snagi0, 2019-10-24:
I tried to look through them.

What are your conrol board/MCU and extruder stepper driver models?

Snagi0, 2019-10-24:
I bet that the reason is the drv8825.
If you would use the tmc2100 driver on the extruder you probably would see it.

ManuelMcLure, 2019-10-24:
I sometimes get extruder rattling with LA and gyroid infill at 0.2mm layer height. Gyroid infill at 0.1mm layer height seems fine. Only my X and Y are TMC drivers (TMC2209) while my Z and extruder are A4988.
After I've finished some long prints that are in progress I might try turning off S-curve acceleration - the rattling might have started when I enabled that option.

ManuelMcLure, 2019-10-29:
My tests showed that turning off S-curve acceleration made no difference to extruder rattling. However, switching to CLASSIC_JERK reduced the rattling a lot. My Junction Deviation value is set to 0.10 using @Roxy-3D 's formula, but I also saw rattling with the default JD value of 0.02.

ManuelMcLure, 2019-10-29:
I created a test print to demonstrate extruder rattling. 
[ExtruderRattlingTest.zip](https://github.com/MarlinFirmware/Marlin/files/3781736/ExtruderRattlingTest.zip)
The uploaded file includes my configuration files (based on the Anet A8 files modified for a BMG extruder) as well as the STL and gcode. The gcode was generated by PrusaSlicer - I set spiral mode with only one bottom layer and 60mm/s speed for outer perimeters. Once the first layer is complete, you can hear the rattling whenever the printer is printing the curves at the end of the arms of the test pattern. The configuration leaves the LA k-factor at 0 but my start gcode for the printer sets the LA k-factor to 0.07 which was tuned using the k-factor test pattern.
I tried reducing JD from 0.10 to 0.02 with no effect on the rattling, and I also reduced E max acceleration to 3000 (from 10000) again with no effect.
Just enabling ``CLASSIC_JERK`` stops the extruder rattling.
I noticed that the code uses the following to calculate E jerk if both LA and JD are enabled:
```
e_jerk = SQRT(SQRT(0.5) * junction_deviation_mm * (e_ max_acceleration) * RECIPROCAL(1.0 - SQRT(0.5)))
```
By my calculation, this results in an E jerk of 49 for an E max acceleration of 10000mm/s^2 and a JD or 0.10. That's an order of magnitude larger than the default E jerk of 5. I don't know if that is relevant.
My JD setting of 0.10 is based on the formula in #15509 but as I mentioned above I have tried reducing the value and still get extruder rattling.

ManuelMcLure, 2019-10-31:
Using JD of 0.02 and E acceleration of 3000 made things a lot worse. The extruder was making buzzing noises even when stopped, and the rattle on the corners was louder.

ManuelMcLure, 2019-10-31:
Here's my results for various combinations of JD and E acceleration:

|JD|Max E Accell|Result|
|-|-|-|
|0.02|3000|Very bad - rattling everywhere|
|0.10|3000|Rattling noticeable at ends of arms|
|0.02|10000|Almost as bad as JD=0.02, Emax=3000|
|0.10|10000|Rattling noticeable at ends of arms - least noticeable but still rattling|

ManuelMcLure, 2019-10-31:
I turned on LA_DEBUG (with JD=0.10, Emax=10000) and see a lot of the following:
```
...
More than 2 steps per eISR loop executed.
More than 2 steps per eISR loop executed.
...
```
So this may be relevant.

ManuelMcLure, 2019-10-31:
With ``CLASSIC_JERK`` enabled I don't get the rattling, but I'm still seeing the ``More than 2 steps per eISR loop executed.`` messages

ManuelMcLure, 2019-10-31:
One thing I'm noticing with JD enabled is that at the end of a long diagonal move on bottom rectilinear infill when it changes direction to make the next line it seems like just before it changes direction it actually speeds up. So it starts the line, reaches a steady speed, then speeds up more before changing direction. This doesn't seem right.

boelle, 2019-11-24:
closing, the discussion can continue on discord

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

or here even if its closed

closed 2019-11-24 by boelle

yeowjoon99, 2020-01-05:
It doesn't only occur on TMC2208's though. My extruder uses an LV8729 and I have the exact same issue as @Snagi0.

Straight lines print perfectly fine but curves are EXTREMELY jerky on the extruder.

hazedav, 2020-02-23:
> Hello, I can confirm that I also have same problem with TMC 2209 (board SKR mini E3) on my ender 3.

I also have LIN_ADVANCE extruder jerky/jittering with TMC 2209 (board SKR mini E3) on my ender 3.  Marlin 2.0-bugfix (2.0.3).  I notice this issue is closed, but it still seems like a thing.

Please advise.