MarlinFirmware/Marlin #10273: full graphic smart controller A8

manotroll, 2018-04-01:
what parameter should I use to use this LCD on the A8 motherboard?

thinkyhead, 2018-04-02:
`REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

manotroll, 2018-04-02:
I'll test and report if it worked.

yo8aiv, 2018-05-30:
for me it didn't work. - only beeper works and seems i can select something when i press the encoder but no image. 
i have tryied everything. just white screen no image
is there a tutorial to test separatley with arduino this LCD ? 
thanks.

thinkyhead, 2018-05-30:
Is the contrast adjusted correctly?

benlye, 2018-05-31:
The `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` is not directly compatible with the Anet board - you need to have some kind of adapter to convert the wiring.  Do you have one?  The details and a couple of options are in the Anet pins file.

yo8aiv, 2018-05-31:
thanks for answering. yes  i have done the wiring accordingly to anet pins file.  tested the 2 variants for reprap in there with no sucess.  
i have tryied a least 5 times in a 3 days triple checking the wiring .. last 4 times i used arduino dupont cables for easy identification of pins.
here i posted a video for the seller showing that nothing happens.  https://youtu.be/KBgbZYa41oA

the only place i found same lcd ppl made it work is here https://reprap.org/forum/read.php?4,492402  
but isn't very clear.

manotroll, 2018-09-16:
> 
> 
> I'll test and report if it worked.

even changing this parameter is not working

thinkyhead, 2018-09-17:
If someone can use `PINS_DEBUGGING` to figure out the pins on the EXT plugs, then it's theoretically possible to produce a pins file for the Anet A8 with pin assignments that line up with the standard connectors like those used for the RRDFGSC.

benlye, 2018-09-17:
> If someone can use `PINS_DEBUGGING` to figure out the pins on the EXT plugs, then it's theoretically possible to produce a pins file for the Anet A8 with pin assignments that line up with the standard connectors like those used for the RRDFGSC.

The Anet A8 pins are already well known, and are documented in the [pins file](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_ANET_10.h#L218).

It's not possible to attach a non-Anet display to the Anet board without a hardware adapter because the 5V and GND pins are in different positions and those cannot be reassigned in the config.  You can either [hack a pair of 10-pin cables](https://www.thingiverse.com/thing:2103748) or get a [PCB adapter ](https://aisler.net/benlye/anet-lcd-adapter/pcb)made.

Bob-the-Kuhn, 2018-09-18:
A variation on the PCB adapter would be to mount the four connectors to a perf board and then solder 30 AWG (wire wrap) wires as needed.

closed 2019-07-17 by shitcreek