MarlinFirmware/Marlin #10277: bugfix v1.1.x dual x carriage printer no way to set offsets between T0 & T1 XYZ

bruce356, 2018-04-02:
### Description
As far as I can find no allowance has been made for the setting of Z offset on dual-X carriage printers between T0 & T1 and the one available setting for X & Y does not work.
// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// #define HOTEND_OFFSET_X {0.0, -0.24} // (in mm) for each extruder, offset of the hotend on the X axis
// #define HOTEND_OFFSET_Y {0.0, 0.18}  // (in mm) for each extruder, offset of the hotend on the Y axis

### Steps to Reproduce
Test it in a Dual X carriage printer running Marlin firmware (by the way as far as I know this option have never worked at least not on the last year).
-->
**Expected behavior:** [What you expect to happen]
To allow offsets on XYZ between T0 and T1
**Actual behavior:** [What actually happens]
Z offset not available and XY offsets do not work.

Thanks and regards - bruce

bruce356, 2018-04-04:
@thinkyhead
Hi, If I enter offsets through host with M218 T1 (X1.5,Y2,Z0.1) the offsets seem to be functioning normally (I did not actually try printing anything, just going by the numbers on LCD).

If these offsets are saved to EEPROM and the printer is reset or restarted (using saved eeprom offsets) and a G28 initiated,  T1 X position is regestering completley wrongly, with above offsets, position for T1 shows as 1.5mm instead of 291mm.

If (X-1.5,Y2,Z0.1) are used (with same process as above) then T1 X position shows correctly as 291mm.

These offset settings if available for adjustment and checking through the GLCD instead of only via host software, would make life a lot easier.

Thanks and regards - bruce

LOG:-
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x704cab90, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo: External Reset
Recv: Marlin bugfix-1.1.x
Recv: 
Recv: echo: Last Updated: 2018-01-20 | Author: (none, default config)
Recv: echo:Compiled: Apr  4 2018
Recv: echo: Free Memory: 2397  PlannerBufferBytes: 1408
Recv: echo:V54 stored settings retrieved (1027 bytes; crc 61800)
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 T1 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y160.00 Z3200.00 E100.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E30.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1100 Y1100 Z100 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1100.00 R1100.00 T1500.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Hotend offsets:
Recv: echo:  M218 T1 X1.50 Y2.00 Z0.10
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H180 B65 F0
Recv: echo:  M145 S1 H240 B90 F0
Recv: echo:PID settings:
Recv: echo:  M301 P18.82 I1.47 D60.12
Recv: echo:  M304 P320.07 I62.62 D409.01
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.77
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.06
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Create-2-RUMBA EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
[... 98 lines cut ...]
Recv: ok
[...]
Send: T1
Recv: echo:T1
Recv: >>> gcode_T(1)
Recv:   current_position=(-41.00, 195.00, 5.77) : BEFORE
Recv: Dual X Carriage Mode DXC_FULL_CONTROL_MODE
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 291.00
Recv:   current_position=(1.50, 197.00, 5.87) : New Extruder
Recv: Active extruder parked: no
Recv:   current_position=(1.50, 197.00, 5.87) : New extruder (parked)
Recv:   current_position=(1.50, 197.00, 5.87) : Sync After Toolchange
Recv:   current_position=(1.50, 197.00, 5.87) : sync_plan_position
Recv: echo:Active Extruder: 1
Recv:   current_position=(1.50, 197.00, 5.87) : AFTER (NOTE! X 1.50 on T1 is WRONG it should be 291mm)
Recv: <<< gcode_T
Recv: ok
[...]
Changing monitoring state from 'Operational' to 'Offline'
Connection closed, closing down monitor

bruce356, 2018-04-06:
Hi wondering if anyone is looking at this - thanks

thinkyhead, 2018-04-08:
Busy week. Looking into it now.

thinkyhead, 2018-04-08:
For Dual X Carriage your `HOTEND_OFFSET_X` setting of `{0, 1.5}` is incorrect. This value should be set to the X position of the second X carriage when it's homed. If your second carriage is positioned at X=291 when homed, then `HOTEND_OFFSET_X` should be set to `{0, 291}`.

bruce356, 2018-04-11:
Hi thinkyhead, many thanks for your assistance, I obviously assumed wrongly, that with a dual X carriage  the X offset had to be the difference between the two hot ends when they were both located at say X125, one at a time, and the slight difference in location due to mechanical variables. 

Its all working as it should be for XYZ so this can be closed.
Regards - bruce

closed 2018-04-12 by thinkyhead

thinkyhead, 2018-04-12:
> it would be nice to be able to adjust these parameters through GLCD

We are trying to keep the LCD menus from exploding with every possible setting, and this is a value that doesn't typically need to be changed often enough to merit a menu entry. A menu item has to either be there all the time, taking up PROGMEM on boards that may have very little, or requires yet another obscure configuration option to enable/disable.