MarlinFirmware/Marlin #10280: Adjust fan speed via LCD/knob

krisrok, 2018-04-02:
I browsed through code and issues and didn't find a way to adjust the fan speed in terms of a LCD menu option.

Is this currently possible? Does it conflict with the auto fan behaviour?

I want to be able to fine tune the fan speed during a print while printing from SD card.

krisrok, 2018-04-02:
Thanks for pointing me in the right direction. 

The problem is my printer has only one fan. I want it to auto-cool the extruder via E0_AUTO_FAN_PIN but i also want to be able to override via LCD menu.

This seems to be unsupported as of now: I cannot use the same pin for E0_AUTO_FAN_PIN and for FAN_PIN. The sanity check throws an error: "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

Maybe we should allow for a multiplier in this situation? I'll have to look into it some more.

Ianmcmill, 2019-05-29:
An addition to this request:
It would be very convenient to have `#define EXTRUDER_AUTO_FAN_SPEED` as a setting in the EEPROM.

Repetier Firmware has this feature.

borland1, 2022-11-22:
I also have a **single** hot-end **fan** printer and would like this **FR** implemented.

Suggest implementation to have the **Auto Fan** speed set at minimum level in firmware for sufficient heat sink cooling, but with ability increase fan speed for additional part cooling via the LCD Menu.