MarlinFirmware/Marlin #10283: Ambient temperature won't let to start print

xyyzyyx, 2018-04-02:
I start print at let say 0 degrees ambient temp the Min temp issue occures a cannot do anything besides hard reset (mostly min temp bed). I have to heat bed up to about 5 degrees in differend way for example by touching hand to the bed and then I can start print.
I belive an ambient temp sensor with temp comparition would be useful together with "testing" heatup  for few second to se if temp changes and if not then before the "min temp error".
Is it a simple way to solve it?

Itox001, 2018-04-03:
You can change the mintemp threshold in the config file, I believe. Setting it to something like -5 degrees should do the trick.

xyyzyyx, 2018-04-03:
Sure I can change it and already did, but it doesn't solve the problem. A lot of temp sensors are unstable at about 0 degrees (temp jumps few degrees up an down), so it won't start. There should be a possibility to strat a testing heatup with comparition to ambient temp sensor. I belive it should work simmilar to this - when temp of the bed changes (goes up) in comparition to ambient temp, heating goes further, but if not it stops and gives min temp error.

closed 2019-07-11 by shitcreek