MarlinFirmware/Marlin #10284: Q: Best way to set pins in pins_XXX.h if same pin can have different functions

alexxy, 2018-04-02:
Hi!

What is the best way to set pins_XXX.h if same pin can have diffrent functions depending on current config.

Example 1:
MKS Gen 1.4 doesnt have Z_PROBE_PIN but i can set it to Z_MAX if i use only normal Z_MIN endstop for G28
Example 2:
Azteeg X5 GT board can use same pin for X_CS_PIN or X_ENABLE_PIN (same for Y,Z,E0, E1) depending on driver type

landodragon141, 2018-04-02:
Do you mean repurpose a pin you aren't using for something else?

thinkyhead, 2018-04-03:
Most pins files assign `Z_MIN_PROBE_PIN` to the `Z_MAX_PIN` by default, but this can be overridden in `Configuration.h`. If the board you're using doesn't supply the pin or set it to the pin you prefer, simply add a define for `Z_MIN_PROBE_PIN` anywhere in your `Configuration.h` file.

thinkyhead, 2018-04-03:
Since the `pins_AZTEEG_X5_GT.h` file doesn't define any of the stepper CS pins, those will also need to be defined in `Configuration.h`.

closed 2018-04-03 by thinkyhead

alexxy, 2018-04-03:
@thinkyhead how to define them in Configuration.h? And same question about Z_MIN_PROBE_PIN

thinkyhead, 2018-04-10:
Just define them. For example:
```cpp
#define X_CS_PIN P1_22
#define Z_MIN_PROBE_PIN P1_29
```