MarlinFirmware/Marlin #10286: TMC 2130 don't work with higher baud rates

craitor, 2018-04-03:


Bob-the-Kuhn, 2018-04-03:
We'll need more info to be able to help.

What baud rate are you referring to?  host to controller, TMC2130 to controller

Where are you setting this baud rate?

What controller and host are you using?

What version of Marlin are you using?

We'll need the details on why you think it's not working.

We'll need to know how to re-create the problem.

Just today I got a five TMC2130 system running on a 2560/RAMPS controller.  The controller is running bugfix-2.0.x and communicating to Repetier Host at 250,000.  My logic analyzer says the SPI clock for the TMC2130 accesses is running at 2,000,000.

My system hasn't actually moved any steppers yet so I can't be 100% sure it's working.

Right now I'm looking at

Bob-the-Kuhn, 2018-04-05:
We'll be glad to help.  

Please ZIP & post your Configuration.h, Configuration_adv.h & pins_YOUR_BOARD.h files. 

We'll need details on the problem.

We'll need to know the steps to reproduce the problem.

craitor, 2018-04-12:
Sorry I got tangled up in a few other things, hence the late reply.

So I retested it this time without the bluethooth and I couldn't get it to work either.
I am using the RAMPS 1.4 together with the repetier software. I have the most recent version of Marlin installed. The TMC2130 are from Watterrot.

I ran the M122 command in Repetier and the stepper motors are constantly disabled. Here is the output I obtained from the M122 command:

00:33:24.966 : X	Y	Z
00:33:24.966 : Enabled		false	false	false
00:33:24.966 : Set current	800	800	800
00:33:24.970 : RMS current	1436	1436	1436
00:33:24.970 : MAX current	2025	2025	2025
00:33:24.970 : Run current	25/31	25/31	25/31
00:33:24.974 : Hold current	12/31	12/31	12/31
00:33:24.974 : CS actual		0/31	0/31	0/31
00:33:24.974 : PWM scale	0	0	0
00:33:24.974 : vsense		0=.325	0=.325	0=.325
00:33:24.978 : stealthChop	false	false	false
00:33:24.978 : msteps		256	256	256
00:33:24.978 : tstep		0	0	0
00:33:24.978 : pwm
00:33:24.978 : threshold		0	0	0
00:33:24.982 : [mm/s]		-	-	-
00:33:24.982 : OT prewarn	false	false	false
00:33:24.982 : OT prewarn has
00:33:24.982 : been triggered	false	false	false
00:33:24.983 : off time		0	0	0
00:33:24.987 : blank time	16	16	16
00:33:24.987 : hysterisis
00:33:24.987 : -end		-3	-3	-3
00:33:24.987 : -start		1	1	1
00:33:24.987 : Stallguard thrs	0	0	0
00:33:24.991 : DRVSTATUS	X	Y	Z
00:33:24.991 : stallguard
00:33:24.991 : sg_result		0	0	0
00:33:24.991 : fsactive
00:33:24.991 : stst
00:33:24.991 : olb
00:33:24.991 : ola
00:33:24.991 : s2gb
00:33:24.995 : s2ga
00:33:24.995 : otpw
00:33:24.995 : ot
00:33:24.995 : Driver registers:
00:33:24.995 : X = 0x00:00:00:00
00:33:24.995 : Y = 0x00:00:00:00
00:33:24.998 : Z = 0x00:00:00:00

I attached the requested  [files.](url
[code.zip](https://github.com/MarlinFirmware/Marlin/files/1901828/code.zip)

)

thinkyhead, 2018-04-13:
Duplicate of #10382

closed 2018-04-13 by thinkyhead