MarlinFirmware/Marlin #10288: Issues switching to UBL due to Bi Linear issues.

D2junkie, 2018-04-03:
Hello, I've recently tried switching to UBL as I'm having issues with bi linear leveling being too low on one side and too high on the other. When I try to G26 and get the mesh print it will do a small portion of the mesh then the z just starts raising up and doesn't stop without my intervention. Help with either issue would be amazing, the bi linear issues only affect my larger prints but as I'm trying to print the pieces for a rather large prop right now I'm at a stand still until I can figure one of these out.

D2junkie, 2018-04-03:
I am just using release version 1.1.x, when I got my printer muy friend helped me with the setup and just uploaded a copy of his firmware and we made the necessary adjustments. Which bugfix version do I need for the UBL fix or do either work?

Roxy-3D, 2018-04-03:
Both bugfix versions have the same code.    Bugfix-2.0.0 has a new file layout, but they should produce functionally identical firmware images.

D2junkie, 2018-04-04:
I flashed to bugfix 1.1.x and now I cannot connect to octoprint. Any ideas, the bltouch flashes and does the self test but then it never completes the connection and eventually just times out.

thinkyhead, 2018-04-08:
The issue you describe is almost always caused by a slight twist in the X gantry leading to the probe changing its orientation as it moves across the bed. You can verify this by taping a long skewer or straw to the probe, and if the angle of the carriage is changing it will be magnified. Unfortunately we cannot correct for this in software. You have to adjust the carriage and/or X gantry so that the twisting is eliminated.

Roxy-3D, 2018-04-08:
> You have to adjust the carriage and/or X gantry so that the twist is eliminated and the probe remains perpendicular to the bed.

You really should try to fix any gantry twist you have.    But if you can live with some distortion in your printed parts, you can do a G26 and edit your mesh points so you get a good first layer.   You will still have some distortion in your printed parts, but with a valid mesh for the Z-Height, you can leave the gantry twisted.

D2junkie, 2018-04-08:
Thank you for the responses, I had investigated the twisted gantry in my attempts to fix this issue. I had confirmed they were square and level (hadn't used the skewer idea) and have now tried the skewer idea to come to the same conclusion that from what I can tell there is no noticeable tilt. However the bugfix has allowed me to get get UBL working and I have fine tuned a 10x10 mesh to be getting some of the best first layers I've seen in a while.

Roxy-3D, 2018-04-08:
And...   Turning on the Double Click for Z-BabyStepping really helps too.    I use it to fine tune the first layer as the skirt is being printed.

D2junkie, 2018-04-08:
Yea I have babystepping turned on, it's a great tool.

closed 2018-04-09 by thinkyhead