MarlinFirmware/Marlin #10289: [2.0] MKS Gen 1.4 TMC2208 - MONITOR_DRIVER_STATUS only last Driver Enabled

Rexima, 2018-04-03:
I followed this german guide (http://3d-druckerei.blogspot.de/2018/03/blog-post.html) to make integrate my TMC drivers into Marlin.
And it looks like only the last driver "E0" is activated.

> Send: M122
> Recv: 		X	Y	Z	Z2	E0
> Recv: Enabled		false	false	false	false	true
> Recv: Set current	800	800	800	800	800
> Recv: RMS current	1436	1436	1436	795	795
> Recv: MAX current	2025	2025	2025	1121	1121
> Recv: Run current	25/31	25/31	25/31	25/31	25/31
> Recv: Hold current	12/31	12/31	12/31	12/31	12/31
> Recv: CS actual		0/31	0/31	0/31	0/31	12/31
> Recv: PWM scale	0	0	0	0	0
> Recv: vsense		0=.325	0=.325	0=.325	1=.18	1=.18
> Recv: stealthChop	false	false	false	false	true
> Recv: msteps		256	256	256	16	16
> Recv: tstep		0	0	0	0	1048575
> Recv: pwm
> Recv: threshold		82	82	65	65	24
> Recv: [mm/s]		1928.35	1928.35	486.54	30.41	329.43
> Recv: OT prewarn	false	false	false	false	false
> Recv: OT prewarn has
> Recv: been triggered	false	false	false	false	false
> Recv: off time		0	0	0	5	5
> Recv: blank time	16	16	16	24	24
> Recv: hysteresis
> Recv: -end		-3	-3	-3	2	2
> Recv: -start		1	1	1	3	3
> Recv: Stallguard thrs
> Recv: DRVSTATUS	X	Y	Z	Z2	E0
> Recv: stst						X
> Recv: olb
> Recv: ola
> Recv: s2gb
> Recv: s2ga
> Recv: otpw
> Recv: ot
> Recv: 157C
> Recv: 150C
> Recv: 143C
> Recv: 120C
> Recv: s2vsa
> Recv: s2vsb
> Recv: Driver registers:	X = 0x00:00:00:00
> Recv: 	Y = 0x00:00:00:00
> Recv: 	Z = 0x00:00:00:00
> Recv: 	Z2 = 0x00:00:00:00
> Recv: 	E0 = 0xC0:0C:00:00
> Recv: 
> Recv: 
> Recv: ok

teemuatlut, 2018-04-03:
This is a limitation in the software serial library where only the last communication attempt works. Your driver should work fine but the reading capabilities are limited for now.
So it's a known problem with some ideas for a solution for now.

Rexima, 2018-04-03:
Okay thank you very much 👍

Studio-dyne, 2018-04-06:
hi , i have an issue , with the same config ... Can you help me , and take a look please 
[#10331](https://github.com/MarlinFirmware/Marlin/issues/10331)

swapsCAPS, 2018-05-12:
@Rexima Did you use the pinout as described in that guide? I thought we could not use that particular config as some of those pins don't support Rx

boelle, 2019-02-20:
@Rexima problem solved? then maybe hit the close botton below

closed 2019-02-20 by Rexima