MarlinFirmware/Marlin #10290: M110 for set line number how to get  the current line printed?

amrithmmh, 2018-04-03:
so M110 is used to set the line number to resume print job but how do you get the line number which is being printed ?

amrithmmh, 2018-04-11:
so @thinkyhead how is the pause and resume in octoprint implemented if line being printed cannot be known?

thinkyhead, 2018-04-11:
> so @thinkyhead how is the pause and resume in octoprint implemented if line being printed cannot be known?

OctoPrint is a host. It knows what lines it has sent. When it pauses, it simply stops sending lines.

boelle, 2019-02-20:
@amrithmmh 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

amrithmmh, 2019-02-25:
@boelle its very old comment 
but thanks for the info

boelle, 2019-02-25:
yeah and i think you have solved the problem right?

amrithmmh, 2019-03-02:
@boelle not solved my problem but understood that there is  a queue

closed 2019-03-02 by amrithmmh