MarlinFirmware/Marlin #10294: [possibly dangerous] DIAG1 pin in TMC2130 should not be set active high by default

marcio-ao, 2018-04-03:
The [following line](https://github.com/MarlinFirmware/Marlin/blob/d95ed46c4ce848ec748c8d5d5ecacb33740fe76b/Marlin/stepper_indirection.cpp#L185) in Marlin is potentially dangerous:

`st.diag1_active_high(1); // For sensorless homing`

The reason is that some boards (for example, the EinsyRambo from Ultimachine), ties the two pins together with a pull-up resistor. This is one of the possibilities in the data sheet, so it should be expected that more boards will choose this approach.

The default value of zero is safe, but setting diag1_active_high to 1 could cause the two diag pins to short out on such boards and I do not know whether the TMC drivers can survive it. This potentially unsafe condition should not be a default in Marlin.

I would think this should either be a configuration option, or the selection ought to be made via a variable in the in the pins_XXX.h file for the boards, as the choice is board specific.

teemuatlut, 2018-04-04:
We can drop the config line. As far as sensorless homing is concerned, the only difference is needing to flip the end stop logic.

marcio-ao, 2018-04-04:
How about adding a constant to the pins_XXX.h file for the board, such as X_DIAG1_PIN_ACTIVE_HIGH? That way once the user selected their board, the st.diag1_active_high(1) flag would be set correctly for their board.

The other advantage of doing this is that sanitycheck.h could be updated to help the user configure X_[MIN/MAX]_ENDSTOP_INVERTING correctly. i.e. It could warn the user to enable X_[MIN/MAX]_ENDSTOP_INVERTING whenever DISABLED(X_DIAG1_ACTIVE_HIGH) and ENABLED(SENSORLESS_HOMING).

teemuatlut, 2018-04-04:
If `diag1_active_high(0)` is the safer option for _some_ and the rest don't care, we should change the behavior.
The thing I'm missing though is why changing the polarity would prevent a short condition?

marcio-ao, 2018-04-04:
@teemuatlut : It's more than polarity. On page 25 of the datasheet, "5.1 General Configuration Registers", it defines diag1_pushpull as such:

0. DIAG1 is open collector output (active low)
1. Enable DIAG1 push pull output (active high)

The key here is that when st.diag1_active_high(1) is set, the driver both actively pushes and pulls, so this would lead to a short. But when st.diag1_active_high(0), it is a open-collector and only pulls (and thus you need an external pull-up resistor).

marcio-ao, 2018-04-04:
> If diag1_active_high(0) is the safer option for some and the rest don't care, we should change the behavior.

Actually, the choice isn't arbitrary. In order for diag1_active_high(0) to work, the board needs a pull-up resistor (or the internal pull-up resistor must be enabled in the MCU input). In order for diag1_active_high(1) to be safe, then the board *must not* interconnect those two pins.

So really the only option that would be safe for all boards would be to "diag1_active_high(0)", but then it would also be necessary to know whether the board had an pull-up resistor, or whether one would need to be set on the MCU input.

marcio-ao, 2018-04-04:
By default ENDSTOPPULLUPS is enabled in "Configuration.h". If we remove `st.diag1_active_high(1);` then SENSORLESS_HOMING can be made to work by setting [AXIS]_MIN_ENDSTOP_INVERTING on the appropriate axis. This is something "sanitycheck.h" could help the user with.

With these new defaults, there is a possibility that on some boards the pull-up resistor would be doubled-up (one on the MCU via the default ENDSTOPPULLUPS and another on the board itself), but this unlikely to damage the driver and certainly is safer than the current default, which causes a short on both boards.

marcio-ao, 2018-04-05:
@teemuatlut : Made a PR with a suggested correction.

closed 2018-04-08 by thinkyhead

wbarber69, 2018-07-03:
Okay. That explains why I don’t remember having to set endstoppullups.  But that sanity check does exist in my currently installed working bugfix from April.  But it doesn’t explain why with the most recent I no longer have motion on x and y.  But I also have to point out that the firmware tried to require me to flip the z endstop as well. But my current configuration does not support this type of endstop.  Not that I couldn’t flip the polarity of the switch but I got around this by commenting out the z sensitivity line to get around sanity check. But it still requires me to set from terminal on the running firmware the sensitivity of z to greater than 0.  So it’s kinda broken there.  I know that I broke it by commenting out a sub setting. But it did allow my z axis to move finally.  And that setting is problematic on both my April build and this one.   So the issue is that sensorless homing requires endstop pull-ups.  And for those pull-ups to be set to true.  Even though I need to set z to false.  So there needs to be a way to determine which drivers you have sh set up on.

thinkyhead, 2018-07-04:
> I also have to point out that the firmware tried to require me to flip the z endstop as well.

Comment out `Z_HOMING_SENSITIVITY` to specify that Z is using a normal endstop.

wbarber69, 2018-07-04:
Yes that’s what I did.  But it then required me to still enter a positive sensitivity number.  I just randomly entered 14 and then I was able to use the z axis.  For some reason it wouldn’t move with a sensitivity value of 0 and yes the z homing sensitivity value is still persistent even after commenting it out.  Also, like I said on latest bugfix I cannot move the x or y axis at all.

thinkyhead, 2018-07-04:
Submitted #11196 / #11197 to skip setting SGT values for axes without sensorless homing.

thinkyhead, 2018-07-04:
> like I said on latest bugfix I cannot move the x or y axis at all

I'm sorry we can't do much about that without more data. Your issue is so far unique.

wbarber69, 2018-07-04:
I realize that, and I’m. I’m not sure how to proceed, in order to gather the data needed to squash this bug

thinkyhead, 2018-07-07:
Observe the output of `MONITOR_DRIVER_STATUS` to see if anything strange appears.

wbarber69, 2018-07-07:
I would have to have that but turned on.  I don’t think I do.  I have some prints to get done first then I’ll give it a go

dremeier, 2019-03-14:
I would like to have the diag1 as aktiv high (pushpull). Where and how i had to add 'st.diag1_active_high'?

wbarber69, 2019-03-14:
2130 was changed from active high for safety concerns if you want to change it ask the guy that wrote the libraries