MarlinFirmware/Marlin #10295: Stop a print in progress

AhmadGhzawi, 2018-04-03:
Hello there,
I'm currently interfacing my printer using MATLAB. I send commands to the printer by the means of gcode.
i.e: G28 auto home. And it is working just fine. But, when i try to stop a print, the printer stops for a bit and then continue printing.

I used this sequence of code, and after that, it continues to print.
M410
M0 [S1]
M104 S0
M140 S0
G28');
G92 E0
G1 E-5  F300
G28 E0
M84
M117 Print Aborted

So, what gcode is used to stop a print in progress?!

Roxy-3D, 2018-04-04:
Probably the place to start is look at the menu structure in ultralcd.cpp.

thinkyhead, 2018-04-08:
`M0` or `M1` (with no `S` parameter) will stop a print until you press the controller button. Note that if this method is used, the heaters will remain on indefinitely.

thinkyhead, 2018-04-08:
`M410` doesn't kill the print job, it just kills the current move.

`M0` or `M1` (with no `S` parameter) will stop a print until you press the controller button or (with EMERGENCY_PARSER) send an `M108` command from the host. Note that if this method is used, the heaters will remain on indefinitely.

There is no G-code to stop an SDCard print in-progress.

Note that `G28 E0` doesn't do anything different from `G28`.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-06-22 by AhmadGhzawi

InsanityAutomation, 2019-06-22:
For reference to not leave this closed without resolution, M524 was added which provides the requested function.