MarlinFirmware/Marlin #10297: keeping marlin up to date

yellobello, 2018-04-03:
Hi,

There is a question that is bugging me for quite some time now and due to recent re-occurance of this problem, I finally ask you guys :)
I just would like to know how your workflow is when you update marlin, since I have the suspicion that mine is a bit... cumbersome I guess.

I am no dev, so I did not have the idea yet to use git for updating, but what I always do is
download a new release from github, extract it to a new folder, replace configuration.h, configuration_adv.h and my pins file with my versions of the files and then use my Atom/Platformio enviroment to compile.
When there has been a long time since my last update I go ahead and do a cumbersome diff between my configs and the latest config file, since it could have well been that I missed some new entries... anyways I have the suspicion that there is an easier way to keep up to date (I am testing the 2.0.0 branch currently...)

How do you guys do it and ensure you don´t miss out any new features/changes...?

best regards

teemuatlut, 2018-04-04:
```
git stash
git fetch origin
git rebase origin/bugfix-2.0.x
git stash pop
```
You can also commit your config with `git commit -a -m "My config"` and skip the `stash` commands.
The `fetch rebase` could possibly be replaced with `git pull`.

Roxy-3D, 2018-04-04:
Using a Visual Diff program makes it easy to cross your working settings over to a newer version of the Configuration.h files.   You might want to check out NotePad++ with the visual diff plug in enabled.

ImplementOfWar, 2018-04-06:
What would be awesome is if thinkyhead the master of the universe would create a windows tool that allowed us to run it while connected to our printers over USB,  check the version on the board,  and then download and then compile,  and automatically port over the configuration from previous version to the newer version,  and then re-upload.  Sort of like an "auto update" system.

Thinkyhead could also become like Microsoft and start updates in the middle of prints,  and then blame people for not unchecking the setting for "always update immediately when new version is found".

Roxy-3D, 2018-04-06:
> What would be awesome is if thinkyhead the master of the universe would create a windows tool that allowed us to 

No!  Software takes so much time to develop.    His changes are needed here more than we need Yet Another Git tool.

ImplementOfWar, 2018-04-06:
@Roxy-3D You're right!  Instead I nominate you to create this yet another git tool!  :)

thinkyhead, 2018-04-08:
@ImplementOfWar — Please keep the jibber-jabber and nicknaming to a minimum. We would like to help @yellobello with the issue at hand, and surely you know that every reply to an issue goes out in an email to everyone in the conversation. Impertinent comments, if they are useful, should be posted in a new issue.

AxelVoitier, 2018-05-01:
I use sed scripts to rewrite from scratch the fresh config files (including their recent modifications) copied from the examples (and thanks a lot for keeping them up-to-date by the way!). To track the changes, I either look at the github commit history, or I do a git diff on the config files between my current version and the latest one.

Good thing also is that all my changes (and only my changes set) resides in one or two condensed files. It makes it easy to follow how my config differs from the default ones. 

Only drawback is that, at first, sed scripts are not super readable...

I published them here if you are interested: https://github.com/AxelVoitier/marlin-config

I also tried to do a python script that would connect to the printer and list all the difference between M503 and the config files. But it has some connections issue... So I didn't published that one (and it turns out I need it less now, so...)

thinkyhead, 2018-05-02:
@AxelVoitier — Thanks! Such things are always of interest. Have you looked at [Marlin-config](https://github.com/akaJes/marlin-config/releases)? It's under constant development, and if it doesn't have the ability now to migrate older configs to the latest Marlin, I'm sure that capability will be added at some point.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead