MarlinFirmware/Marlin #10300: Hot End Fan on RAMPS 1.4 not working after upgrade to 1.1.8

PicoPi, 2018-04-04:
My hot end fan was working and coming on after 50ºC and going out bellow that, until I upgraded to 1.1.8.
Now it doesn't work anymore. I can turn on the fan with M106 S255, so I know the mosfets are fine.
Checked the Config.h and config_adv.h and Pin_RAMPS.h, I can't find the problems.
Why is the fan and other stuff like SD Card not enabled by default for RAMPS 1.4 Boards? 
How could you print without having a hot-end fan, no matter type of plastic or machine type?
Part cooling fan I get, but extruder fan?

kAdonis, 2018-04-04:
>  I can turn on the fan with M106 S255
Interesting
M106 S255 should only turn on your part cooling fan.
please post your configs and your Pins_RAMPS.h. How are your Fans wired?
I agree with you, it should not be hidden in config_adv.h, but it can´t be default, because it needs additional Hardware (mosfet)

PicoPi, 2018-04-05:
Hi,
I have converted my Anet A6 to RAMPS 1.4+ Arduino Mega2560 but cant get the extruder fan working!
It works with M106 S255 command, so I know it is properly connected and the MOSFET works, but it will not come on automatically when the hot-end passes 50ºC !!
I am running the latest Marlin (1.1.8) here are some relevant lines in
****************
configuration.h
****************
#define MOTHERBOARD BOARD_RAMPS_14_EFB
#define PREHEAT_1_TEMP_HOTEND 190
#define PREHEAT_1_TEMP_BED     60
#define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255
****************
configuration_adv.h
*****************
#define E0_AUTO_FAN_PIN -1
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed
-----> I have tried setting E0_AUTO_FAN_PIN to 11 with no results
If I use the servo port pin 11 for E0_AUTO_FAN_PIN, it will turn on a fan that is connected to an external MOSFET like TinyFan board when the temp reaches 50ºC !!!
I have tried assigning it to 9, but get an error
#error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
*********************
RAMPS.h
*********************
// Heaters / Fans
//
#ifndef RAMPS_D8_PIN
  #define RAMPS_D8_PIN   8
#endif
#ifndef RAMPS_D9_PIN
  #define RAMPS_D9_PIN   9
#endif
#ifndef RAMPS_D10_PIN
  #define RAMPS_D10_PIN 10
#endif

#define HEATER_0_PIN     RAMPS_D10_PIN

#if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
  #define FAN_PIN        RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN

I am at the end of my whits. I need help. I don't know why D9 port (fan connector on the RAMPS board) doesn't turn on the fan automatically, but I can turn it on with the M106 S255 command!
>> HELP <<

PicoPi, 2018-04-05:
I have my HotEnd fan connected to D9 as I did on my Prusa.
My part cooling fan is connected to an external MOSFET board I made (like "The MightyFan" http://wiki.aus3d.com.au/MightyFan)
Why would I want to connect the part cooling fan to D9?
I always need an extruder fan. I have no part cooling fan on my Prusa MK2 clone and this setup works fine in 1.1.0 version.

PicoPi, 2018-04-05:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1881000/Configs.zip)

kAdonis, 2018-04-05:
I have my Extruder Auto Fan also on D9

In Configuration_adv.h I have:
`#define E0_AUTO_FAN_PIN  RAMPS_D9_PIN`

and in pins_Ramps.h I changed

```
#if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
  #define FAN_PIN        5  // Nonstandard, with Mosfet
  #define HEATER_BED_PIN RAMPS_D8_PIN
```
Pin 11 should work also