MarlinFirmware/Marlin #10309: Delta Printer not lowering to a safe height after homing

jaanshgo, 2018-04-04:
Hello everyone, 

After spending many hours on forums trying to find my issue I can't fix my problem so here I am.

My problem is that I want my delta printer to lower its height after homing in order to prevent movement on the x and y axis from smashing into the top. I can't seem to find where to change this on the marlin configuration.h file. I am using the generic delta printer configuration.h file found in the marlin examples folder modified to my printer.

I have tried changing the value of:

 #define MANUAL_Z_HOME_POS DELTA_HEIGHT which also changes Z_MAX_POS with no luck.

Where exactly do you change to make the z hight lower after homing? also how do I adjust:
 #define MAX_SOFTWARE_ENDSTOP_Z and is it something I want to adjust?

Thank you very much in advance. I've been struggling for hours changing different values hoping one will work with no luck.

[Configurationdoth.txt](https://github.com/MarlinFirmware/Marlin/files/1878181/Configurationdoth.txt)

viper93458, 2018-04-05:
I believe you are looking for #define DELTA_HOME_TO_SAFE_ZONE in configuration.h

-William

jaanshgo, 2018-04-05:
You are amazing. I can't believe I didn't see that. Is there a way to adjust how much it lowers?

viper93458, 2018-04-05:
Per a note that thinkyhead left in another thread in December of 2016:

The "safe zone" height is calculated to be the highest point where you can still move X and Y in the full bed region. It's calculated from the printer's geometry.

-William

jaanshgo, 2018-04-05:
my printer has a smaller print bed so even going to the limit of the print bed it still has space where it could go up. It's not a huge deal breaker but it would be great to base it off of the print space and not the possible mechanical movement. Is there another file that you know of where I could refer to it and adjust it? It has to be somewhere but there are so many files.

jaanshgo, 2018-04-05:
Also, now that you are here, maybe you can help with this. Is there a way to make that height DELTA_HOME_TO_SAFE_ZONE into the new max Z height so that the printer wont accept any movement above unless you enable homing?

jaanshgo, 2018-04-05:
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)
#define Z_MIN_POS 0
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
#define Z_MAX_POS MANUAL_Z_HOME_POS

Based on the comment above I would assume changing the value of Z_MAX_POS to something like: 
#define Z_MAX_POS MANUAL_Z_HOME_POS - 20
would prevent the printer from moving 20 milimiters from the limit switches but it seems to have no effect. Am I missing something?

jaanshgo, 2018-04-05:
I've found the file that has most of the code. Marlin_main.cpp. It is a behemoth of code and you can change so many things. Most things I am scared of changing with fears of breaking something tho.

closed 2018-04-05 by jaanshgo