MarlinFirmware/Marlin #10310: [bugfix 1.1.x] USB communication loss mid-print, everytime

klazzera, 2018-04-05:
### Description

I'm not really sure if this is a bug due to the marlin firmware, the ramps board, the arduino board or even the power supply. I have this problem for over two weeks now and checked many online sources, had no luck fixing. I can't pinpoint the problem.

I use ramps 1.4, arduino mega 2560(with 16u2 usb chip) and marlix bugfix 1.1.x

In the middle of the print, my usb connection is lost, when triyng to reconnect, i get "error opening selected port" message.

- The port number is correct.
- Doesnt matter if i'm printing from SD or over USB.
- I'm able to reconnect only after unplugging/plugging the usb cable, but connection gets lost again shortly after starting a print.
- If printing from SD, only the usb connection to the PC is lost, for my case, on Simplify 3D it still shows as it is connected to the board but arduino board doesnt respond over serial. Printing continues without problem but i can't command the printer through USB serial. I click disconnect then try to reconnect, i get the "error opening selected port" message.
- When printing from USB, printing stops and Arduino board stops responding over serial.
- I'm using a  regular 12V power supply at least 300% above the 3D printers power consumption, board is powered thorugh both the 12V and the USB
- If i keep the USB plugged in to the PC, do not open the serial port and start printing from SD, i'm able to connect over USB serial even if it's the 8th hour of the print, but of course board gets resetted due to the initialization of the serial connection, but at least i'm able to connect. But problem happens again if i start another print while still connected over serial this way.
- Tried 3 different cables, 3 different arduino boards and ramps 1.4 boards of the same batch, nothing changed
- My usb cables are 2 meters long
-Tried to change the baudrate to 115200, problem persists

### Steps to Reproduce

1. Connect to the machine through serial from Simplify 3D
2. Start a print either from SD or over USB
3. USB communication stalls
4. I try to disconnect and reconnect, but i get "error opening selected port" message. Tried opening the serial port from Arduino IDE, gives the error "Error while setting serial port parameters: 250,000 N 8 1"

**Expected behavior:** [What you expect to happen]

I should be able to stay connected at any time

**Actual behavior:** [What actually happens]

USB communication is lost mid print until next USB cable unplug/plug

If this is not the proper place to post it, please close this issue and inform me

lrpirlet, 2018-04-05:
Humm, USB has got 2 ends, you only talk about the printer side...

Just to close a door, have a look at https://www.windowscentral.com/how-prevent-windows-10-turning-usb-devices (or google "USB power saving"), maybe your computer does power down its side.

thinkyhead, 2018-04-08:
I'm not sure of everything you could look into. While you're waiting for someone around here to answer, also be sure to ask for help in some other forums, such as:
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)

klazzera, 2018-04-10:
after more extensive testing, it turns out i have huge EMI going around my setup, extra isolation at the USB cable solved the problem. sorry for bothering here for this issue.

closed 2018-04-10 by klazzera