MarlinFirmware/Marlin #10316: Extruder fan doesn't come on automatically.

PicoPi, 2018-04-05:
Hi,
I have converted my Anet A6 to RAMPS 1.4+ Arduino Mega2560 but cant get the extruder fan working!
It works with M106 S255 command, so I know it is properly connected and the MOSFET works, but it will not come on automatically when the hot-end passes 50ºC !!
I am running the latest Marlin (1.1.8) here are some relevant lines in
****************
configuration.h
****************
#define MOTHERBOARD BOARD_RAMPS_14_EFB
#define PREHEAT_1_TEMP_HOTEND 190
#define PREHEAT_1_TEMP_BED     60
#define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255
****************
configuration_adv.h
*****************
#define E0_AUTO_FAN_PIN -1
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed
-----> I have tried setting E0_AUTO_FAN_PIN to 11 with no results
If I use the servo port pin 11 for E0_AUTO_FAN_PIN, it will turn on a fan that is connected to an external MOSFET like TinyFan board when the temp reaches 50ºC !!!
I have tried assigning it to 9, but get an error
#error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
*********************
RAMPS.h
*********************
// Heaters / Fans
//
#ifndef RAMPS_D8_PIN
  #define RAMPS_D8_PIN   8
#endif
#ifndef RAMPS_D9_PIN
  #define RAMPS_D9_PIN   9
#endif
#ifndef RAMPS_D10_PIN
  #define RAMPS_D10_PIN 10
#endif

#define HEATER_0_PIN     RAMPS_D10_PIN

#if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
  #define FAN_PIN        RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN

I am at the end of my whits. I need help. I don't know why D9 port (fan connector on the RAMPS board) doesn't turn on the fan automatically, but I can turn it on with the M106 S255 command!
>> HELP <<

th33xitus, 2018-04-05:
I think you are missunderstanding something there.
Your Extruder Fan (the fan which cools the Extruder, NOT the part-cooling-fan) isn't (or shouldn't be) connected to D9.
D9 is for the part-cooling-fan which will get controlled by your g-code. That's why you can activate it via g-code command.
Aren't you using any part-cooling fan or why do you want to pick D9 for your extruder?

The error message you get `#error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."` actually tells you what you did wrong. I am not quite sure if you can just redefine D9 to another pin like `#define RAMPS_D9_PIN 11` for example and just do `#define E0_AUTO_FAN_PIN 9` then.
Give it a try, maybe it works, maybe not. I would suggest you to take a diffrent aproach on that.

kAdonis, 2018-04-05:
why do you post this twice?
#10300

PicoPi, 2018-04-05:
I tried it and it works. Thank you. 
Some one has changed this from the default it used to be!
My Prusa MK2 clone works fine without this code mod running an earlier version. I think it's 1.1.0?
Which brings me to another point. Why is the version number not in Marlin.ino? Can't tell which version of code I am looking at!
I have no part cooling fan on my Prusa and my extruder fan is connected to D9 and automatically comes on after 50ºC and turns off bellow that!
EVERY printer needs an extruder fan, but not every material used for printing needs cooling i.e. ABS
I spent hours trying to debug this, when all along it was the dumb code. Whomever made this change, should stop working on the code immediately. His logic is flawed.

closed 2018-04-05 by PicoPi

th33xitus, 2018-04-05:
> Whomever made this change, should stop working on the code immediately. His logic is flawed.

Pretty harsh words for just missing to compare the new configurations to your old ones or going through them step by step if you don't have them anymore. I always keep at least one last set of working configuration-files on my pc, just to be able to compare them with the new files and look through them if something was changed. I don't think you can make anyone else responsible for this other than yourself.

PicoPi, 2018-04-05:
And if you have not used Marlin before? 
You would be pissed off too if you spent 4 hours and didn't have the previous files where in a laptop with a crashed HD.

PicoPi, 2018-04-05:
I did use Marlin before. Many people are new and haven't any experience to troubleshoot something this illogical. 
Didn't have the old config files to compare. Do you understand now?
If I had access to the old Marlin, I wouldn't be downloading it again and starting from scratch!
Who puts a parts fan and not an extruder fan? What is the purpose of 50ºC threshold if the extruder fan has to be hard wired to 12V? 
Defend this all you want. Its still dumb. 
I brought it to your attention and my job is done. What you do or don't do with it, is not my concern.

thinkyhead, 2018-04-08:
> Why is the version number not in Marlin.ino? Can't tell which version of code I am looking at!

Good idea. We'll have to add that. For now it can be found in `Version.h`

thinkyhead, 2018-04-08:
> Who puts a parts fan and not an extruder fan?

Actually, most printers have extruder fans that are always on, connected straight to a 12V plug with no digital control. The "AUTO" fans are optional because they are not as common and may be connected to any spare MOSFET.

thinkyhead, 2018-04-08:
To sum up, the intended way to do configuration for the `En_AUTO_FAN_PIN` settings is to just comment out the ones that you want to enable, and then they will be set to the default pin defined in your board's `pins_MYBOARD.h` file. For example, to use the default pin for `E0_AUTO_FAN_PIN`:
```cpp
//#define E0_AUTO_FAN_PIN -1
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
```

PicoPi, 2018-04-08:
This issue is closed you idiot.

thinkyhead, 2018-04-08:
@PicoPi — I don't know what you expect to achieve insulting the project curators.

thinkyhead, 2018-04-08:
@PicoPi — I posted extra information to clarify all the options for your benefit and for the benefit of others who may come across this issue in future.

I don't know what you expect to achieve by insulting the project curators.

th33xitus, 2018-04-08:
@thinkyhead Never mind mate, keep up your good work. Some people are just limited in their thinking.

I don't know if it is possible to ban people here, but this person would be a perfect candidate imho to refuse him total access to the bug-section of your git.

PicoPi, 2018-04-08:
You are both idiots and noob programmers. It shows in your logic and your code.
Ban me? you are the one that keep sending me emails about a closed case.
Stop doing that. How can I ban you from doing that? nvm. just piss off.

th33xitus, 2018-04-08:
Wow, you are actually THAT limited that you can't even find or press the "unsubscribe" button? :D 
Thanks dude, you just made my day.

You are a shame for the 3D-Printing-Community.

PicoPi, 2018-04-08:
No stupid. If I unsub, I won't get any notifications from github, not just this thread.
I did better, and muted this thread. Noob.

thinkyhead, 2018-04-08:
The button is near the top of this page:
[image]

@PicoPi — We get some amazingly odd and immature characters around here sometimes. And yes, when they act particularly awful we block them. Have a fabulous day!

kodaman2, 2019-01-03:
I was doing some research for the autofan options and I came across this issue gem lmao. Give @PicoPi a break he's gotta be at least 12yo with that kind of attitude and calling software devs noobs and flawed logic. I wish github gave a jac**ss of the day trophy.

The code is fine, I'll give you a small hint before you go onto your next github issue name calling adventure. There is part of the code which is the control logic, and another part is the configuration. The configuration part is your responsibility!

jorjonides, 2020-05-01:
Thanks for the info guys, really helped!
I think you guys are doing a really good job with this firmware, keep it up! @thinkyhead 
Nevermind the haters in the world! :)