MarlinFirmware/Marlin #10327: Error "'DOGLCD_CS' was not declared in this scope" with Exit Status 1

Geric2004, 2018-04-06:
### Description
I keep getting this error. I'm pretty new to Marlin so please forgive me if I made some mistakes in this post.

**Expected behavior:** 
Compile and Upload to my Ramps Arduino Board Sandwich.

**Actual behavior:** 
Fails at compiling and gives this Error
[Marlin 1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/1882962/Marlin.1.1.x.zip)

Geric2004, 2018-04-06:
Thanks! I was having a rough time figuring it out.

Bob-the-Kuhn, 2018-04-06:
Great!

Can I close the issue?

Geric2004, 2018-04-06:
One last question, I tried it and there is no display on the Full Smart Graphic controller. (I have it connected to a laptop only since I haven't migrated all the electronics from my old board. The Display is the only thing connected)

Bob-the-Kuhn, 2018-04-06:
Make sure your config file matches your LCD.  Lots of similar names:
- REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
- REPRAP_DISCOUNT_SMART_CONTROLLER
- ELB_FULL_GRAPHIC_CONTROLLER
- REPRAPWORLD_GRAPHICAL_LCD

If one end is a MKS product and the other isn't then chances are good that the connectors on the 10 pin cables are rotated 180 degrees from each other.  If that's the case then you'll need to either modify one end (pry the housings up, rotate and push them back down), make custom cables with one end reversed or shave the keying plugs off one end of the cables and plug it in backwards.

Geric2004, 2018-04-06:
I have a Chinese Ramps Kit (That came with the Display, Ramps Shield and Arduino Clone) And I checked the name printed on the LCD it self it says RepRap Discount Full Graphic Smart Controller. I still can't find a fix for it.

Geric2004, 2018-04-07:
Thank you so much! It finally worked. You may close this issue now

Bob-the-Kuhn, 2018-04-07:
FANTASTIC!

closed 2018-04-07 by Bob-the-Kuhn