MarlinFirmware/Marlin #10330: Progress bar never reaches 100%

marcio-ao, 2018-04-06:
We're finding that the progress bar stays stuck at 99% in most SD card prints.

I'm not exactly sure how to go about fixing this. I suspect that what is happening is that "ultralcd.cpp" only updates the progress when [IS_SD_PRINTING is true](https://github.com/MarlinFirmware/Marlin/blob/ac3fab7a9680a880fd1b113be7e758c6dabb2a57/Marlin/ultralcd.cpp#L627):

```
    // Set current percentage from SD when actively printing
    #if ENABLED(LCD_SET_PROGRESS_MANUALLY)
      if (IS_SD_PRINTING)
        progress_bar_percent = card.percentDone();
    #endif
```

But once a print finished, that flag is probably set to false before the ultralcd has a chance to refresh. Perhaps removing the if statement would fix the problem? Would there be any concerns in calling card.percentDone() when not printing?

Does anyone have any better suggestions to cause the progress bar to reach 100%?

marcio-ao, 2018-04-06:
Actually, disabling that if does not seems to fix the issue.

marcio-ao, 2018-04-06:
So far the only way I can get the progress bar to show 100% is by adding the following to the end GCODE in the slicer:

```
M73 P100 ; Set progress bar to 100%
G4 P100 ; Wait for the LCD to refresh
```

Obviously, this is not ideal.

thinkyhead, 2018-04-08:
At the end of a print the progress bar should be cleared to zero, so that is what is now done. If it is getting set at some other point after being reset to 0, please let us know!

closed 2018-04-08 by thinkyhead

marcio-ao, 2018-04-09:
> At the end of a print the progress bar should be cleared to zero, so that is what is now done. If it is getting set at some other point after being reset to 0, please let us know!

I guess whether it is 0 or 100 doesn't matter too much. But getting stuck at 99% was a problem. Prior to this PR, it the value would never have gotten updated past 99% when LCD_SET_PROGRESS_MANUALLY was enabled.

commit 211544b5 (2018-04-07): Reset progress bar when print ends

Fix #10330
Based on #10333 by @marcio-ao
commit 2efc072f (2018-04-07): Clear progress bar on completion (#10333)

Fix #10330