MarlinFirmware/Marlin #10331: [2.0] MKS Gen 1.4 TMC2208 -> Linear adv stop extrusion

Studio-dyne, 2018-04-06:
Hi ...
Extrusion stops on the first nervous wipe or sometimes stop immediatly , I have all tried , and use very low feedrate , acc , velocity , and anything to slow down , i have lowered and upped PULSE . And just now , i have set linear adv to ' ZERO' , and extrusion never stop.

I have tried , to crash extruder by Gcode command , with all brutality as possible , but always run , the only things that make stop extruders is LINEAR advance more than zero ....

Someone have an idea , or something to make to give you more information on my issue  ?

_**I have seen another issue , and just after a G28 feedrate not applied on the first move ... and is equal to max ... an extrusion just after G28 don't respect the feedrate param send with ' move to destination'**_ 

thks for help

Nume1977, 2018-04-06:
TMC2208 do not work with linear advance or 32bit firmware unless they are serial connected or OTP'ed (One Time Programmed) to use SpreadCylce. 
**It is a hardware feature because of the dynamic Stealthchop2 / Spreadcycle. And no firmware solution has yet been found**
For more information read the open bug report on the klipper firmware github page.

melvinisken, 2018-04-06:
Unfortunately, I'm not sure if I understood your text correctly (maybe write the text in native language and try google translate?) but I guess you ran into an issue specific to the 2208 drivers. Read here: https://github.com/KevinOConnor/klipper/issues/196 (at the end of the long thread). There are people having the same issue with rapid movements causing the 2208 to lock up. It's still under investigation. 
Personally, I tested Linear Advance 1.5 with Marlin 1.1.x-bugfix and the same setup (MKS 1.4, TMC2208 and full UART mode) without problems. But I only did a short test.

Studio-dyne, 2018-04-06:
Thk you very much ....
Sorry for the bad english , i will use google the next time
I go on the issue you have mentionned

👍

Studio-dyne, 2018-04-06:
About linear advance ..... Really sorry to hear that some users don't use it ..... Impossible to make a perfect square just with the file of a slicer ... Only a setting in real time with your eyes can relax the pressure on corners the more perfect as possible ....
Use it one time , and you will use it for life ' FwRetract too '

Thks again

Studio-dyne, 2018-04-06:
@Nume1977 @melvinisken 
Is it possible to keep x/y/z TMC2208 drivers and another for extrusion ? to keep the more silence as possible ?

Studio-dyne, 2018-04-06:
I'm trying it now , and seems to work fine !

melvinisken, 2018-04-06:
Yes, that should be no problem. You can use any combination of drivers. And since the extruder is normally very slow and with no vibrations, you can safely use a standard driver there without increasing the sound very much.

Nume1977, 2018-04-06:
@Studio-dyne, yes you can keep XYZ with TMC and use another driver for extruder if you want linear advance, or you can lock that TMC2208 driver into permanent Spreadcycle, it will always be more silent than other drivers!

melvinisken, 2018-04-06:
And I will have a look at Linear Advance again :-) I found it not changing a lot by then, but I might not have tested enough :-)

closed 2018-04-07 by Studio-dyne