MarlinFirmware/Marlin #10332: Compilation error: DUE software SPI is incompatible with TMC2130 hardware SPI. Enable TMC_USE_SW_SPI to fix.

LichtiMC, 2018-04-06:
Board SMART_RAMPS:
Having wired my LCD SD-Card-Reader to the sw_spi Pins as it's using sw-spi by default -> Working fine!

Wired TMC2130 drivers to hw_spi. Now while compiling I get following error:
Marlin\src\HAL\HAL_DUE../../inc/../HAL/HAL_DUE/SanityCheck_Due.h:33:6: error: #error "DUE software SPI is incompatible with TMC2130 hardware SPI. Enable TMC_USE_SW_SPI to fix."

Isn't this a contradiction? We shall enable Software SPI as DUE software SPI is incompatible with TMC2130???
How to fix this?

Bob-the-Kuhn, 2018-04-06:
If the two SPIs are using the same MOSI, MISO and SCK pins then they need both be software SPIs or both be hardware SPIs.

Looking at the logic of the sanity check it only checks if they're the same or not.

Bob-the-Kuhn, 2018-04-06:
Short term fix is to edit \Marlin\src\HAL\HAL_DUE\SanityCheck_Due.h and delete lines 31 and beyond.

I'll get it fixed to allow for the SPIs using different pins.

Bob-the-Kuhn, 2018-04-06:
Here's the fixed logic along with better error messages.  Please let me know if you have suggestions;

```
/**
 * HARDWARE VS. SOFTWARE SPI COMPATIBILITY
 * 
 *   DUE selects hardware vs. software SPI based on if one of the hardware controllable SDSS pins is in use.
 *   
 *   The hardware SPI controller doesn't allow soft SPIs to control any shared pins.
 *   
 *   When DUE soft SPI is used then the TMC2130 must use the TMC softSPI.  
 *   
 *   When DUE hard SPI is used then the TMC2130 can use either its hardware SPI or, if there are no shared 
 *   pins, its soft SPI.
 *   
 *   Usually the hardware SPI pins are only available to the LCD.  This makes the DUE hard SPI used at the same 
 *   time as the TMC2130 soft SPI the most common setup.
 */

#if ENABLED(SDSUPPORT) && ENABLED(HAVE_TMC2130) 
  #if ENABLED(TMC_USE_SW_SPI) && !ENABLED(DUE_SOFTWARE_SPI) \
      && (    (DEFINED(TMC_SW_MOSI) && TMC_SW_MOSI == MOSI_PIN) \
           || (DEFINED(TMC_SW_MISO) && TMC_SW_MISO == MISO_PIN) \
           || (DEFINED(TMC_SW_SCK)  && TMC_SW_SCK  == SCK_PIN ) \
         )  
    #error "DUE hardware SPI is required but is incompatible with TMC2130 software SPI.  Either disable TMC_USE_SW_SPI or use separate pins for the two SPIs."
  #elif ENABLED(DUE_SOFTWARE_SPI) && ! ENABLED(TMC_USE_SW_SPI)
    #error "DUE software SPI is required but is incompatible with TMC2130 hardware SPI. Enable TMC_USE_SW_SPI to fix."
  #endif
#endif
```

LichtiMC, 2018-04-06:
Now I've got:
```Marlin\src\HAL\HAL_DUE\../../inc/../HAL/HAL_DUE/SanityCheck_Due.h:55:7: error: #error "DUE software SPI is required but is incompatible with TMC2130 hardware SPI. Enable TMC_USE_SW_SPI to fix."```

LichtiMC, 2018-04-06:
Trying to understand the description above:
``DUE selects hardware vs. software SPI based on if one of the hardware controllable SDSS pins is in use.``
What is a "hardware controllable" SDSS pin? Chip select pins are always hardware controllable, they select the active chip after all...??

Bob-the-Kuhn, 2018-04-06:
> DUE Hardware-SPI-Pins are: MISO (108), MOSI (109), SCK (110)

Those pins are not used by the hardware SPI controller used by Marlin.  Those are: 
```
SCK_PIN           76
MISO_PIN          74
MOSI_PIN          75
```

Actually, I can't find those pins anywhere in Marlin.  Where did they come from?  You mentioned the SMART_RAMPS above.  It uses the system default pins above.

What LCD are you using?

Might as well ZIP & post the following files: Configuration.h ,  Configuration_adv.h & pins_YOUR_BOARD.h.  That way I can try to reproduce your issues.

Please also do a **M43 I** command (that's an i, not a 1) and include the log for it.  That'll show us what pins are actually in use.  The M43 command is available only when PINS_DEBUGGING is enabled in Configuration_adv.h.

____

> "hardware controllable" SDSS pin

By that I meant directly controllable by the hardware SPI controller with the SAM3X chip.

The hardware SPI controller within the DUE can be setup to move a block of data and set the slave select active at the beginning and set it inactive at the end - all without software involvement beyond the setup.  That could be useful in a real time operating system (RTOS).

____

One SPI for the LCD & SD card and a different SPI for the TMC2130s.  Reasonable & practical way of doing it.

Using the hardware SPI for all three is easy to do in Marlin.  The problem is getting physical access to the SPI pins for use on the TMC2130 cables.  All the controller/shield systems I know of have the hardware SPI pins only on the connector(s) dedicated to the LCD/SD-card.  That pretty much forces the TMC2130 to use it's software SPI. 

Usually the DUEs hardware SPI is used for the LCD/SD-card just because the shield/controller is usually setup/designed that way.  The main reason for that is the LCD/SD-card move a lot more data than any other users so the higher data rates on the hardware controllers mean less loading on the CPU.

Another consideration is the SD card software will use the default SPI channel which always uses the hardware SPI's pins. Going away from the default SPI channel means that the SD card's software SPI would have to be enabled.  That requires editing files other than Configuration.h & Configuration_adv.h.

LichtiMC, 2018-04-06:
>Those pins are not used by the hardware SPI controller used by Marlin

Sorry, these were physical pin numbers, I afterwards found the digital pin-numbers.

---
>By that I meant directly controllable by the hardware SPI controller with the SAM3X chip.

>The hardware SPI controller within the DUE can be setup to move a block of data and set the slave select active at the beginning and set it inactive at the end - all without software involvement beyond the setup. That could be useful in a real time operating system (RTOS).

OK, I think I understand: We can't use this "software-less" spi system because the pins 4, 10, 52 are already used by servo, heatbed, software-spi. Correct?

---
>Usually the DUEs hardware SPI is used for the LCD/SD-card just because the shield/controller is usually setup/designed that way. The main reason for that is the LCD/SD-card move a lot more data than any other users so the higher data rates on the hardware controllers mean less loading on the CPU.

I have a [Reprap Discount Full Graphics Smart Controller](http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller)
So the Display doesn't even use SPI but only the SD-Card. And SD-Card (now) uses SW-SPI Pins 50, 51, 52, 53 and Pin 49 for SD-Card-Insert-Detection.
Regarding the cpu-load it would be better to wire the SD-CARD to HW-SPI and the TMC2130 to SW-SPI, isn't it?

Bob-the-Kuhn, 2018-04-07:
I realized I'd probably forget to do it tomorrow so ...
```
#define BEEPER_PIN       66
#define BTN_ENC          67
#define LCD_PINS_ENABLE  42
#define LCD_PINS_D4      40
#define LCD_PINS_RS      44
```

LichtiMC, 2018-04-07:
The Display is working fine with the smart adapter on AUX4. I manually wired AUX3 to match the HW-SPI of the DUE. (And 49 = SD-Detect, 52 = SDSS)

With prior versions of Marlin I already used TMC2130 with HW-SPI. (Of course I had to manually breakout the HW-SPI-Pins all the while as the SMART RAMPS Board doesn't have the DUE-HW-SPI Pins.) 
So in principle the HW-SPI does work, just not directly with the shield.

Just need a working configuration for the sd-card so that it uses the HW-SPI.

Bob-the-Kuhn, 2018-04-07:
Wow - hardware mods - most unusual

The DUE hardware SPI pins are not physically connected to the SMART RAMPS board.  They are only available on a 6 pin connector near the CPU/controller chip.  Below is the best DUE pinout diagram I've found.

Unless you're willing to do more hardware mods you're stuck with the soft SPI for the SD card.

[image]

Bob-the-Kuhn, 2018-04-07:
I think running **M43 W I** at the end of EXP2 is worth while.  That'll tell you if a logical pin is connected and which one it is.  Then we'd be 100% sure that the hardware is correct.

LichtiMC, 2018-04-07:
OK, so far so good... as stated above HW-SPI with SD-Card-Reader is working fine and Full Graphics Display is also working good.
Unfortunately as soon as I enable TMC drivers, bootloop is the result:
```
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: David
echo:Compiled: Apr  8 2018
echo: Free Memory: 82776  PlannerBufferBytes: 1344
Unified Bed Leveling System v1.01 inactive.
start
echo:PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: David
echo:Compiled: Apr  8 2018
echo: Free Memory: 82776  PlannerBufferBytes: 1344
Unified Bed Leveling System v1.01 inactive.
start
echo:PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: David
echo:Compiled: Apr  8 2018
echo: Free Memory: 82776  PlannerBufferBytes: 1344
Unified Bed Leveling System v1.01 inactive.
```
Config:
```arduino
#define HAVE_TMC2130
#if ENABLED(HAVE_TMC2130)  // Choose your axes here. This is mandatory!
  #define X_IS_TMC2130
  #define Y_IS_TMC2130
#endif
#if ENABLED(HAVE_TMC2130) || ENABLED(HAVE_TMC2208)

  #define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #define X_CURRENT          350  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256

  #define Y_CURRENT          350
  #define Y_MICROSTEPS        16

  #define Z_CURRENT          800
  #define Z_MICROSTEPS        16

  #define X2_CURRENT         800
  #define X2_MICROSTEPS       16

  #define Y2_CURRENT         800
  #define Y2_MICROSTEPS       16

  #define Z2_CURRENT         800
  #define Z2_MICROSTEPS       16

  #define E0_CURRENT         800
  #define E0_MICROSTEPS       16

  #define E1_CURRENT         800
  #define E1_MICROSTEPS       16

  #define E2_CURRENT         800
  #define E2_MICROSTEPS       16

  #define E3_CURRENT         800
  #define E3_MICROSTEPS       16

  #define E4_CURRENT         800
  #define E4_MICROSTEPS       16

  /**
   * Use software SPI for TMC2130.
   * The default SW SPI pins are defined the respective pins files,
   * but you can override or define them here.
   */
  #define TMC_USE_SW_SPI
  #define TMC_SW_MOSI       51
  #define TMC_SW_MISO       50
  #define TMC_SW_SCK        53 //eigentlich 52, aber das ist für HW-SPI SD-CARD (SDSS)

  /**
   * Use Trinamic's ultra quiet stepping mode.
   * When disabled, Marlin will use spreadCycle stepping mode.
   */
  #define STEALTHCHOP

  /**
   * Monitor Trinamic TMC2130 and TMC2208 drivers for error conditions,
   * like overtemperature and short to ground. TMC2208 requires hardware serial.
   * In the case of overtemperature Marlin can decrease the driver current until error condition clears.
   * Other detected conditions can be used to stop the current print.
   * Relevant g-codes:
   * M906 - Set or get motor current in milliamps using axis codes X, Y, Z, E. Report values if no axis codes given.
   * M911 - Report stepper driver overtemperature pre-warn condition.
   * M912 - Clear stepper driver overtemperature pre-warn condition flag.
   * M122 S0/1 - Report driver parameters (Requires TMC_DEBUG)
   */
  #define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    #define STOP_ON_ERROR
  #endif

  /**
   * The driver will switch to spreadCycle when stepper speed is over HYBRID_THRESHOLD.
   * This mode allows for faster movements at the expense of higher noise levels.
   * STEALTHCHOP needs to be enabled.
   * M913 X/Y/Z/E to live tune the setting
   */
  //#define HYBRID_THRESHOLD

  #define X_HYBRID_THRESHOLD     100  // [mm/s]
  #define X2_HYBRID_THRESHOLD    100
  #define Y_HYBRID_THRESHOLD     100
  #define Y2_HYBRID_THRESHOLD    100
  #define Z_HYBRID_THRESHOLD       3
  #define Z2_HYBRID_THRESHOLD      3
  #define E0_HYBRID_THRESHOLD     30
  #define E1_HYBRID_THRESHOLD     30
  #define E2_HYBRID_THRESHOLD     30
  #define E3_HYBRID_THRESHOLD     30
  #define E4_HYBRID_THRESHOLD     30

  /**
   * Use stallGuard2 to sense an obstacle and trigger an endstop.
   * You need to place a wire from the driver's DIAG1 pin to the X/Y endstop pin.
   * X, Y, and Z homing will always be done in spreadCycle mode.
   *
   * X/Y/Z_HOMING_SENSITIVITY is used for tuning the trigger sensitivity.
   * Higher values make the system LESS sensitive.
   * Lower value make the system MORE sensitive.
   * Too low values can lead to false positives, while too high values will collide the axis without triggering.
   * It is advised to set X/Y/Z_HOME_BUMP_MM to 0.
   * M914 X/Y/Z to live tune the setting
   */
  //#define SENSORLESS_HOMING // TMC2130 only

  #if ENABLED(SENSORLESS_HOMING)
    #define X_HOMING_SENSITIVITY  8
    #define Y_HOMING_SENSITIVITY  8
    #define Z_HOMING_SENSITIVITY  8
  #endif

  /**
   * Enable M122 debugging command for TMC stepper drivers.
   * M122 S0/1 will enable continous reporting.
   */
  #define TMC_DEBUG

  /**
   * M915 Z Axis Calibration
   *
   * - Adjust Z stepper current,
   * - Drive the Z axis to its physical maximum, and
   * - Home Z to account for the lost steps.
   *
   * Use M915 Snn to specify the current.
   * Use M925 Znn to add extra Z height to Z_MAX_POS.
   */
  //#define TMC_Z_CALIBRATION
  #if ENABLED(TMC_Z_CALIBRATION)
    #define CALIBRATION_CURRENT 250
    #define CALIBRATION_EXTRA_HEIGHT 10
  #endif

  /**
   * You can set your own advanced settings by filling in predefined functions.
   * A list of available functions can be found on the library github page
   * https://github.com/teemuatlut/TMC2130Stepper
   * https://github.com/teemuatlut/TMC2208Stepper
   *
   * Example:
   * #define TMC_ADV() { \
   *   stepperX.diag0_temp_prewarn(1); \
   *   stepperY.interpolate(0); \
   * }
   */
  #define  TMC_ADV() {  }

#endif // TMC2130 || TMC2208
```

LichtiMC, 2018-04-07:
@Bob-the-Kuhn: I think the function _IS_HW_SPI has a bug:
```
In file included from Marlin\src\HAL\HAL_DUE\../../inc/../HAL/HAL_SanityCheck.h:29:0,
from Marlin\src\HAL\HAL_DUE\../../inc/MarlinConfig.h:35,
from Marlin\src\HAL\HAL_DUE\EepromEmulation_Due.cpp:35:
Marlin\src\HAL\HAL_DUE\../../inc/../HAL/HAL_DUE/SanityCheck_Due.h:46:62: error: operator "defined" requires an identifier
```

thinkyhead, 2018-04-08:
Patched `_IS_HW_SPI` so (like `PIN_EXISTS`) it can use `defined()`.

thinkyhead, 2018-04-08:
> Shouldn't pin MOSI (Slave Input) be defined as Output?

It should. Was that due to your pin reversal, or does this still need to be looked into?

On a related note, in `private_spi.h` there is this…
```cpp
SET_INPUT(MISO_PIN);
WRITE(MISO_PIN, HIGH);
```
To be fully compatible with all platforms, maybe it should be…
```cpp
SET_INPUT_PULLUP(MISO_PIN);
```

thinkyhead, 2018-04-08:
> Shouldn't pin MOSI (Slave Input) be defined as Output?

It should, and it is normally, but `M43 I` actually overrides it.

---
On a related note, in `private_spi.h` there is this…
```cpp
SET_INPUT(MISO_PIN);
WRITE(MISO_PIN, HIGH);
```
To be fully compatible with all platforms, maybe it should be…
```cpp
SET_INPUT_PULLUP(MISO_PIN);
```

LichtiMC, 2018-04-08:
I finally found the reason for the bootloops:
On the smart ramps board, apparently pin 42 is connected via Jumper "EN_RESET" to the RESET of the DUE Board. I use PIN 42 as chip select pin for the SW-SPI.
So everytime when PIN 42 switched to HIGH, the reset of the DUE Board was triggered and it rebooted.
After removing the jumper "EN_RESET" everything is working fine.

closed 2018-04-08 by LichtiMC

thinkyhead, 2018-04-10:
Good to know! We'll have to make a note about that in `pins_RAMPS_SMART.h` for future reference. I see that `KILL_PIN` is also defined as 42 when using the `AZSMZ_12864`.

LichtiMC, 2018-04-10:
#10353 ;-)

jbaiao, 2018-06-18:
Hello.
LichtiMC, can you please help me?
I have due+smart ramps + AZSMZ_12864, but when I select my lcd "#define AZSMZ_12864" it always gives me compile error.
Can you send me the marlin 2.0 already configured?
Thank you!

LichtiMC, 2018-06-18:
I have the RepRapDiscount Full Graphic Smart Controller.
It just works with the Smart Adapter on AUX4 although I had to remove the sd-card-pin-header from the adapter. I then simply used the hardware spi for running the sd-card.

MagicBugsBunny, 2018-11-01:
Hi, i try to get my TMC2130 working on a DUE. I have same Hardware as LichtiMC.
I am a Little bit confused About HW or SW SPI.
Must i Change SD-Card to HW SPI? But were can i assing the Pins 74 75 76 for HW SPI?
TMCs than use SW SPI from the SD-Card?