MarlinFirmware/Marlin #10335: FYSETC TMC2130 SPI one direction only

hideki22, 2018-04-06:
### Description

Hello,

A month ago, I buy a FYSETC TMC2130 v1.1

I use it in a ramps 1.4 and marlin 1.1.8 firware ( And 1.1.X bugfix). 
I configure all to use it in spi mode. Driver detected by firware and work, but only in 1 direction.

If i say turn left, go left. If i say turn right, go left ...

I don't know what happend, but with old driver A4988 work ok....

I need any special config to use it in marlin and spi mode?

Has the same happened to anyone? 

Thank you

### Steps to Reproduce

1. Install driver in ramps
2. Configure marlin os include in my config files
3. Say to move in direction right or left

**Expected behavior:** go right or left

**Actual behavior:** go in one direction, left left

#### Additional Information

M122 output:

< 18:41:16: start
> 18:41:16: N1 M110 *2
< 18:41:16: echo:Marlin 1.1.8
< 18:41:16: echo: Last Updated: 2017-12-25 12:00 | Author: Hideki
< 18:41:16: echo:Compiled: Apr  4 2018
< 18:41:16: echo: Free Memory: 4475  PlannerBufferBytes: 1232
< 18:41:16: echo:Hardcoded Default Settings Loaded
< 18:41:16: echo:  G21    ; Units in mm
< 18:41:16: echo:  M149 C ; Units in Celsius
< 18:41:16: echo:Filament settings: Disabled
< 18:41:16: echo:  M200 D3.00
< 18:41:16: echo:  M200 D0
< 18:41:16: echo:Steps per unit:
< 18:41:16: echo:  M92 X80.00 Y80.00 Z1600.00 E500.00
< 18:41:16: echo:Maximum feedrates (units/s):
< 18:41:16: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
< 18:41:16: echo:Maximum Acceleration (units/s2):
< 18:41:16: echo:  M201 X3000 Y3000 Z100 E10000
< 18:41:16: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
< 18:41:16: echo:  M204 P3000.00 R3000.00 T3000.00
< 18:41:16: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
< 18:41:16: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
< 18:41:16: echo:Home offset:
< 18:41:16: echo:  M206 X0.00 Y0.00 Z0.00
< 18:41:16: echo:Material heatup parameters:
< 18:41:16: echo:  M145 S0 H180 B70 F0
< 18:41:16: echo:  M145 S1 H240 B110 F0
< 18:41:16: echo:PID settings:
< 18:41:16: echo:  M301 P22.20 I1.08 D114.00
< 18:41:16: echo:Stepper driver current:
< 18:41:16: echo:  M906 X 800
< 18:41:16: echo:Sensorless homing threshold:
< 18:41:16: echo:  M914 X3
> 18:41:28: N10 M122 *51
< 18:41:28: X
< 18:41:28: Enabled		false
< 18:41:28: Set current	800
< 18:41:28: RMS current	795
< 18:41:28: MAX current	1121
< 18:41:28: Run current	25/31
< 18:41:28: Hold current	12/31
< 18:41:28: CS actual		12/31
< 18:41:28: PWM scale	45
< 18:41:28: vsense		1=.18
< 18:41:28: stealthChop	true
< 18:41:28: msteps		16
< 18:41:28: tstep		1048575
< 18:41:28: pwm
< 18:41:28: threshold		0
< 18:41:28: [mm/s]		-
< 18:41:28: OT prewarn	false
< 18:41:28: OT prewarn has
< 18:41:28: been triggered	false
< 18:41:28: off time		5
< 18:41:28: blank time	24
< 18:41:28: hysterisis
< 18:41:28: -end		2
< 18:41:28: -start		3
< 18:41:28: Stallguard thrs	3
< 18:41:28: DRVSTATUS	X
< 18:41:28: stallguard		
< 18:41:28: sg_result		0
< 18:41:28: fsactive		
< 18:41:28: stst		X
< 18:41:28: olb		
< 18:41:28: ola		
< 18:41:28: s2gb		
< 18:41:28: s2ga		
< 18:41:28: otpw		
< 18:41:28: ot		
< 18:41:28: Driver registers:
< 18:41:28: X = 0x80:0C:00:00

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1885445/config.zip)

LichtiMC, 2018-04-06:
Try and test your endstops with [M119](http://marlinfw.org/docs/gcode/M119.html).
Maybe this is the root cause...

hideki22, 2018-04-06:
Hello,

I send M119 and this is te output.

without pressing
> 22:13:35: N16 M119 *61
< 22:13:35: Reporting endstop status
< 22:13:35: x_min: TRIGGERED
< 22:13:35: y_min: TRIGGERED
< 22:13:35: z_min: TRIGGERED

press 
> 22:13:40: N22 M119 *58
< 22:13:40: Reporting endstop status
< 22:13:40: x_min: open
< 22:13:40: y_min: TRIGGERED
< 22:13:40: z_min: TRIGGERED

Thank you.

Bob-the-Kuhn, 2018-04-06:
Things to try:
- Have you tried the old driver A4988 lately?  It should behave the same.
- Issue a M121 command to disable endstops and see if that fixes the problem.  if it does then something is wrong in your endstop setup.
- Move the TMC2130 to a different socket & see what happens.
- There's only one signal difference between going left & right.  It's the DIR pin.  It's on one of the corners.  If M121 doesn't fix it then it's either a bad TMC2130 or a bad socket on your board.  Try bending the DIR pin a little and see if that fixes it.

If none of the above gives you the cause then the TMC2130 is bad.