MarlinFirmware/Marlin #10340: Endstop unexpectedly disabled when SD support is enabled

loudaslife, 2018-04-07:
### Description

When enabling `#define SDSUPPORT` on my delta geometry printer for this first time, the Y-axis endstop was disabled without warning, resulting in an incorrect homing and subsequent physical damage to my build plate as the effector crashed into it.

### Steps to Reproduce

1. Have a functional delta printer using Marlin firmware, compiled using `#define MOTHERBOARD BOARD_PRINTRBOARD` to suit the hardware used, and without SD support initially enabled.
2. Enable `#define SDSUPPORT` after having dialed in the physical settings of the printer.

**Expected behavior:** A compile time error should have occurred, indicating that the user needs to change the Y-axis endstop to a different port to avoid conflict with the pins required for the SD card.

**Actual behavior:** The Y-axis endstop was unexpectedly remapped without warning or user interaction. Physical damage resulted as the now-incorrect homed position led to the effector crashing into the build plate.

Please do not automatically change the motion behavior of the printer when changing non-motion-related features like SD or LCD support.

#### Additional Information

The offending code in `pins_PRINTRBOARD.h:`
```
#define X_STOP_PIN         47   // E3
#if ENABLED(SDSUPPORT)
  #define Y_STOP_PIN       37   // E5 - Move Ystop to Estop socket
#else
  #define Y_STOP_PIN       20   // B0 SS - Ystop in Ystop socket
#endif
#define Z_STOP_PIN         36   // E4
```

thinkyhead, 2018-04-08:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

GMagician, 2018-04-08:
I had some strange behaviours with endstop and my cartesian printer. Sometime, in very rare cases, endstops don't work.
My micro are pulled up and close to gnd so if there is some bad cable connection they should be seen as 'detected', in my endstop board I have a led that turn on when micro is pressed.
I saw this on all 3 axes in different situation, the led turn on but stepper continue to move. After some time (less then 3 seconds) it stops and terminates homing correctly.
I'm using bugfix-2.0.x but I don't know if it's the same issue (but it sound similar)
H/W Mega2560 with ramps

psavva, 2018-04-10:
@GMagician I suggest you open a new issue...

thinkyhead, 2018-04-10:
Well, you know, Marlin can't fix hardware problems. It trusts what the endstops tell it.

GMagician, 2018-04-10:
@psavva it's several months I'm using my printer. Issue happened 10 times (maybe less) don't know if such issues can be solved easily (H/W or soft cause is) I reported only to let it be known (maybe someone has more repetitive situations which cause can be easily detected)
@thinkyhead well endstop led is turned on by switch contact then it's working. Note that I had the same issue with bltouch (red led was on but no trigger). It sound strange that I have the same behavior with different inputs, different cables and different H/W device. I know everything is possible!
I read that interrupt, in the past, had some problems...I'm using them! Do you know something about? maybe has been solved or not I didn't found anything

loudaslife, 2018-10-21:
Fixed with #10676 and #10677

closed 2018-10-21 by loudaslife